raspberrypi/rpicam-apps #516: understanding which YUV420 is being used

petercwallis, 2023-06-12:
In order to understand what's going on, I am trying to write a post-process-stage to convert the previewed image to grey scale.
According to davidplowman (Feb 8, 2022) the first stride*height bytes of an image will be Y values, followed by stride/2 * height/2 U values,  then the same number of V values.  Modifying the negate _stage code, if I copy over the first stride*height values and zero the rest, I get a black and green image.  If I set them to 0xff I get a lovely cyan image. The wikipedia page on Chroma subsampling sorta helps but gets right into too much detail about everything, and the YUV entry doesn't give enough detail.  I think I need to add in the values of U and V for each pixel in some way? Or is the image viewer doing something other than YUV420?

naushir, 2023-06-12:
In order to convert to a greyscale image, I think you want to set all the sample values in the U and V planes to 128.

petercwallis, 2023-06-12:
That's it. Perfect thanks naushir.

closed 2023-06-12 by petercwallis