raspberrypi/rpicam-apps #517: Right way to use build system for new post-processing-stages

petercwallis, 2023-06-12:
I want to add -lapriltag to the linker line in the Makefile put in the build directory.  What is the correct way to do that? It would be good to mention how to do this [here](https://www.raspberrypi.com/documentation/computers/camera_software.html) under writing your own post-process-stages.  I guess I need to edit the CMake file -  I can compile and install the library independently.

naushir, 2023-06-15:
Firstly, I would suggest you use the meson build scripts for this - the CMake build system will be slowly deprecated for libcamera apps.

To get this working, you will need to edit [this](https://github.com/raspberrypi/libcamera-apps/blob/main/post_processing_stages/meson.build) meson build file, and add something like:

```meson
apriltag_dep = dependency('apriltag', required : true)
libcamera_apps_dep += apriltag_dep
libcamera_app_src += ...
```

naushir, 2023-06-20:
If this is now resolved, feel free to close the issue.

naushir, 2023-06-21:
Resolving now...

closed 2023-06-21 by naushir

petercwallis, 2023-06-23:
Naushir - sorry this took so long to get to.

I didn't try your solution in the end - no doubt it works.  I had
previously had trouble with the make system and ended up answering my own
question.

The solution to that was the line:
```
gcc -st=c++17 -o helloWorld helloWorld.cpp -lstdc++ -lcamera-base -lcamera
-lcamera_app
```
The contents of helloWorld.cpp is of course not defined here but is based
on the hello example code.

It turns out I could add post-process-stages from my working directory once
I had installed apriltag. The line in my Makefile is:
```
apriltag: apriltag_detect.o apriltag_detect_stage.o
        gcc -std=c++17 -o $@ $^ $(BASELIBS) -lpost_processing_stages
-L/usr/local/lib/arm-linux-gnueabihf/ -lapriltag
```
This is IMHO very simple and obvious for someone who can write c once one
knows what libraries, hedders, and options are required.  Could that
information be put in the documentation somewhere?.  Why we need a
byzantine make system on a raspbery-pi is beyond me.

To make the example complete, I went to put the code for helloWorld.cpp,
apriltag_detect,cpp and apriltag_detect_stage.cpp on my github account
page.  Being a dinosaur, I don't use github to write code normally.  Now it
is taking me a stupid amount of time to figure out how to create a project
on github.  I seem to have many options to do lots of things, but which one
enables me to put code on a page that others can get to?  I did think I
would do that in an hour and then respond, but the hour turned into 2
hours, then a day, and then the day-job intervened and I still haven't
figured out how to  put something on github.  I think github and gitlab are
suffering from edge-case poisoning
<https://buttondown.email/hillelwayne/archive/edge-case-poisoning/>, but
the libcamera-app code is just badly designed. I feel sorry for you and I
was not helping by not responding.  Sorry.
P