raspberrypi/rpicam-apps #521: [BUG] Flushing SDTOUT after handling user signal.

levitatingpineapple, 2023-06-18:
Hi,
I'm trying to poll images trough `stdout`:

```
libcamera-still \
        --width 128 \
        --height 96 \
        --timeout 0 \
        --verbose 0 \
        --signal \  
        --flush \  
        --output - 
```

However output is not flushed after handling `kill -USR1 $PID` signal  (or keypress), 
so the last part of the image is missing.
This means that any image decoder will always be one image behind.

Is this the expected behaviour?

While I'm aware that this might be somewhat of an edge case,
there has already been similar issues merged in the legacy stack:
https://github.com/raspberrypi/userland/issues/133 and https://github.com/raspberrypi/userland/pull/297

naushir, 2023-06-19:
Actually let me correct myself.  It seems like ``FileOutput::closeFile()`` does not flush the output!
Can you try something like this:
```diff
diff --git a/output/file_output.cpp b/output/file_output.cpp
index 4f2a624a2b1c..72bad45534ef 100644
--- a/output/file_output.cpp
+++ b/output/file_output.cpp
@@ -67,7 +67,12 @@ void FileOutput::openFile(int64_t timestamp_us)

 void FileOutput::closeFile()
 {
-       if (fp_ && fp_ != stdout)
-               fclose(fp_);
-       fp_ = nullptr;
+       if (fp_)
+       {
+               if (options_->flush)
+                       fflush(fp_);
+               if (fp_ != stdout)
+                       fclose(fp_);
+               fp_ = nullptr;
+       }
 }
```

naushir, 2023-06-22:
The above change is now merged.  Please give the latest tree a test and hopefully this should be fixed.

naushir, 2023-06-29:
Resolving as this is now merged.

closed 2023-06-29 by naushir