raspberrypi/rpicam-apps #523: [BUG] SensorTemperature is 0.0000 for many cameras

EricClaeys, 2023-06-21:
**Describe the bug**
The `SensorTemperature` metadata is 0.0000 for many cameras, including the HQ.  However, for the Module 3 it shows the temperature.

**Bug report**
SensorTemperature.txt attached

**Additional context**
none

[SensorTemperature.txt](https://github.com/raspberrypi/libcamera-apps/files/11812074/SensorTemperature.txt)

naushir, 2023-06-21:
Thank you for raising this.  Indeed the imx744 ought to work but does not.  It seems that some recent kernel rebasing meant that the register write to enable the thermal sensor got lost.  I've re-introduced the change at https://github.com/raspberrypi/linux/pull/5513.  Once this gets merged amd a new kernel build is available, the values should be correct again.

Note that as of right now, only the IMX708 and IMX477 support thermal sensor readout.  For all other sensors, the control will not be present.

pelwell, 2023-06-21:
The Pull Request with the fix has been merged. If you wait an hour or so for the autobuilds to complete you'll be able to try the fix by running `sudo rpi-update pulls/5513`.

EricClaeys, 2023-06-21:
@naushir, @pelwell, thanks for the very quick work.
When will users be able to get the update via `apt install libcamera-apps`?
Is there a way to automatically determine if a camera supports sensor temp?  I understand that only 2 cameras support it today but I'd like the Allsky program to be able to do _something_ to know if a camera supports it.  Perhaps not include SensorTemperature in the meta data?  Or set the temp to -9999 or some other impossible value?

naushir, 2023-06-21:
> @naushir, @pelwell, thanks for the very quick work. When will users be able to get the update via `apt install libcamera-apps`?

This will be a kernel update, not libcamera apps.  The changes will be included into apt once we tag the next stable release of the kernel.  Unfortunately I don't have visibility on when this is going to happen next. 

> Is there a way to automatically determine if a camera supports sensor temp? I understand that only 2 cameras support it today but I'd like the Allsky program to be able to do _something_ to know if [...]

If a sensor does not have a thermal readout option, the control will not exist in the metadata returned out to the application.  This alone ought to give you indication of whether the camera supports this or not.

6by9, 2023-06-21:
It looks like IMX296 does have a temperature sensor (although docs do say "precision is not guaranteed"), however it's not apparently in the embedded data, and needs to be triggered to take a reading.

We did discuss with mainline Linux folks over how to expose temperature, and as a V4L2 control was one option. It became a bit of a mine field though, so using the embedded data into the pipeline handler was the simplest approach for those sensors that support it.

IMX219 also has a temperature sensor and the result is in the metadata, but it isn't clear if it is a trigger for a single conversion or for constant readout. It's also fairly coarse as the granularity is 0.8333 degC (5/6).

naushir, 2023-06-21:
> It looks like IMX296 does have a temperature sensor (although docs do say "precision is not guaranteed"), however it's not apparently in the embedded data, and needs to be triggered to take a reading.

Correct, from my discussions with Sony, the IMX296 does not return this in embedded data.  Bit annoying, but oh well...

> We did discuss with mainline Linux folks over how to expose temperature, and as a V4L2 control was one option. It became a bit of a mine field though, so using the embedded data into the pipeline [...]

We currently don't use imx219 embedded data as there seems to be issues with it, I can't recall exactly what right now.  Probably worth another look at some point.

6by9, 2023-06-21:
> We currently don't use imx219 embedded data as there seems to be issues with it, I can't recall exactly what right now. Probably worth another look at some point.

Comment in cam_helper_im219 says that it randomly returns junk register values.

naushir, 2023-06-23:
Closing this now as the work is completed.

closed 2023-06-23 by naushir