raspberrypi/rpicam-apps #524: libcamera-still taking twice as long as --shutter

EricClaeys, 2023-06-21:
With the Module 3 camera, which has focus control, when I run `libcamera-still` it takes roughly twice as long to return as the --shutter time.
~~~
time LIBCAMERA_LOG_LEVELS=ERROR,FATAL libcamera-still --output '/tmp/x.jpg' --metadata \
     /home/pi/allsky/config/overlay/extra/libcamera.txt --metadata-format txt --timeout 1 \
    --nopreview --width 4056 --height 3040 --shutter 5000000 --autofocus-mode manual
Still capture image received

real	0m11.568s
user	0m0.391s
sys	0m0.174s
~~~

It does this regardless of the shutter value and with or without ` --autofocus-mode manual`.
The documentation doesn't mention this.  Is there another setting I need to keep this 2x from happening?  It doesn't happen with the HQ camera.

naushir, 2023-06-21:
I think this command is not doing what you are expecting it to do.  By default ``libcamera-still`` sets a fixed framerate of 30fps if a value is not provided on the command line.  Framerate overrides shutter speed in priority, so when you ask for a 5s exposure time, that value will be truncated to what the framerate would allow.  You can fix this by adding ``--framerate 0`` to the above command line option so that framerate follows the shutter speed you request.

However, there are implications of this.  On startup we always run our 3A algorithms for a number of frames to converge.  This will now happen at a rate of 1/5 fps resulting in an incredibly slow capture time.  If you provide all manual values, the algorithms do not have to converge and can run immediately.  To do this, add ``--gain 1`` and ``--awbgains 1.5,1.8`` and ``--immediate`` to the command line.  The ``--awbgains`` values will need adapting to your scene and illuminant.  Best way to do this is see what the auto algorithms come up with, and just plug in the same values manually.

EricClaeys, 2023-06-24:
@naushir,  I shortened the command being run to make it easier to see.  I should have posted the whole command:
~~~
libcamera-still --output '/tmp/x.jpg' --metadata /home/pi/allsky/config/overlay/extra/libcamera.txt \
    --metadata-format txt --timeout 1 \
    --analoggain 1 --awbgains 2.51,2.1 --saturation 1 --contrast 1 --sharpness 1 --brightness 0 --quality 100 \
    --nopreview --width 4056 --height 3040 --shutter 5000000 --autofocus-mode manual
~~~

Running on the **Module 3** always results in the "real time" taking twice as long as the shutter time, as if something (focus??) was still in auto mode trying to converge on the correct value.
Runing on an **HQ** camera (and without `-autofocus-mode manual`) always results in a "real time" close to the shutter time.

I have never gotten `--immediate` to work so don't use it. IIRC it caused libcamera-still to never return.

If it helps, here's the output of "--metadata":
~~~
ExposureTime=59997881
AfState=1
AnalogueGain=7.699248
AfPauseState=0
FrameDuration=59999143
SensorTemperature=42.000000
LensPosition=1.000000
AeLocked=false
DigitalGain=1.006644
Lux=0.001810
ColourGains=[ 2.510000, 2.100000 ]
ColourTemperature=4966
SensorBlackLevels=[ 4096, 4096, 4096, 4096 ]
FocusFoM=2640
ColourCorrectionMatrix=[ 1.570767, -0.277466, -0.293301, -0.393810, 1.793847, -0.400047, -0.070043, -0.515764, 1.585810 ]
ScalerCrop=(575, 0)/3458x2592
SensorTimestamp=433269217699000
~~~

naushir, 2023-06-26:
Ah, ok this helps!  I can see why it's going wrong now.  It's to do with this line:
https://github.com/raspberrypi/libcamera/blob/06bbff9c23d5a85acd65595216c6b75094891f72/src/ipa/raspberrypi/cam_helper_imx708.cpp#L59C98-L59C98

The imx708 camera helper is telling us to drop one frame on every mode switch.  So when compared with the imx477, you will be capturing 1 extra frame for the entire capture cycle.  Given your entire capture cycle is a single frame, this has effectively doubled your capture time as you pointed out.  

Pinging @njhollinghurst - the comment suggests this seems to be needed for HDR modes.  What about non-HDR modes (like the case here), can we get away with removing this frame drop?

njhollinghurst, 2023-06-26:
Yes, I think it's specific to HDR mode. The IMX708 seems to refer to its own statistics from the previous frame, when doing HDR stitch and tonemap, so the first frame is noticeably different. The extra delay is not needed in non-HDR modes.

Detecting HDR is a little tricky, as the "WIDE_DYAMIC_RANGE" control currently bypasses libcamera altogether. When enabled, the camera mode will always be 2304x1296 with a maximum of 30.0 fps.

naushir, 2023-06-26:
> I have never gotten `--immediate` to work so don't use it. IIRC it caused libcamera-still to never return.

The following command does seem to work for me (using the ``--immediate`` flag).

```
libcamera-still --output test.jpg --timeout 1 --analoggain 1 --awbgains 2.51,2.1 --saturation 1 --contrast 1 --sharpness 1 --brightness 0 --quality 100 --nopreview --width 4056 --height 3040 --shutter 5000000 --autofocus-mode manual --immediate
```

And actually, because of a small bug in the helper, it actually will give you the result you are looking for, i.e. no dropping any frames for capture.  However, please don't rely on this, I'll create a proper fix for all of this and submit to libcamera shortly.

EricClaeys, 2023-06-26:
@naushir 

> However, please don't rely on this, I'll create a proper fix for all of this and submit to libcamera shortly.

Don't rely on what?  Don't use `--immediate` ?

naushir, 2023-06-27:
> @naushir
> 
> > However, please don't rely on this, I'll create a proper fix for all of this and submit to libcamera shortly.
> 
> Don't rely on what? Don't use `--immediate` ?

If you use ``--immediate``, it should give you the result you are looking for, i.e. the capture cycle  time for Camera v3 will be the same as HQ camera.  However, I have also pushed an official fix to the libcamera team to integrate, so eventually you won't need to use ``--immediate`` if you don't want to.

closed 2023-07-10 by naushir

EricClaeys, 2023-07-10:
Was this only a problem with the RPi Module 3 (IMX708) camera?  If so, do other cameras not drop the first frame when not in HDR mode?

Since we won't know if a user has the fix, will using `--immediate` cause any problems if they have the fix?

naushir, 2023-07-11:
> Was this only a problem with the RPi Module 3 (IMX708) camera? If so, do other cameras not drop the first frame when not in HDR mode?

Yes, we only drop the first frame for IMX708 in HDR mode - which as far as I can tell, you are not using.  Some sensors do need a frame dropped on startup, some don't.  This is entirely sensor dependent.

> 
> Since we won't know if a user has the fix, will using `--immediate` cause any problems if they have the fix?

If you are always fixing the shutter speed, analogue gain and white balance gains, you should use ``--immediate``  as it will give you the smallest latency for a capture.

almowery, 2025-08-28:
naushir

I hope you can help me with this.  I am trying to use a picamera2 app called indi_pylibcamera.  I have been using a the rpicam-still command with the
 --immediate option with great success.  I wanted to try using this app which works with picamera2 and I am seeing different results.  
 
 When I run picamera with script 5sec works fine 5sec result, raw mode.
 When i run picamera with a loop repeated the 5sec jumps to 10sec.
 
 The indi_pylibcamera app is worse it always takes double the time and often even longer.  I am confused what is the difference between rpicam-still and picamera2
 and then indi_pylibcamera which is supposed to be built on picamera2  why aren't they the same?
 
I do not know if the driver(.dtb)from the kernal, something in libcamera, or if it is just a setting in indi_pylibacamera is not making the --immediate mode.

The driver/camera I am using is imx283 which went from a developer created driver to incorporation into the latest kernal build.

I thought the fix you put in from this post was to make the switch --immediate stop preview frames.  Is this correct?

You seem to have a real handle on this problem with some cameras I hope you cna help me solve this.

I have included the py script that repeats the exposures in a loop in case this helps.

#!/usr/bin/python3

"""
test program for taking raw still images the same was as it is done by indi_pylibcamera
"""

import time
import datetime
import pprint
from picamera2 import Picamera2
from libcamera import controls
from playhouse.sqlite_udf import total_seconds

picam2 = Picamera2()  # TODO: select correct camera if you have more than one connected!

# Print camera information. The "config" and "controls" settings below must fit to the capabilities
# of the camera!
print('Camera properties:')
pprint.pprint(picam2.camera_properties)
print()
print("Raw sensor modes:")
pprint.pprint(picam2.sensor_modes)
print()
print('Camera controls:')
pprint.pprint(picam2.camera_controls)
print()

# We want to make still images.
config = picam2.create_still_configuration(queue=False, buffer_count=2)

# We are interested in raw images only. But the driver can not provide raw images without a
# processed ("main") one. To avoid wasting time and memory for calculating the "main" image we 
# configure it here with a pretty small size.
config["main"]["size"] = (240, 190)

# The raw image is the important one. Its configuration mut fit to the capabilities of the camera,
# which can be determined with "picam2.sensor_modes". The following settings are for IMX477.
config["raw"] = {"size": (4576, 3648), "format": 'SRGGB12'}

# Set configuration
picam2.align_configuration(config)
picam2.configure(config)

# Set the exposure controls. This must fit t the capabilities of your camera! Possible controls and
# their value ranges can be determined with "picam2.camera_controls"
# The following is for IMX477.
controls = {
    # Exposure time and analog gain are the most important controls for raw images!
    "ExposureTime": int(5 * 1e6),  # unit: microsecond
    "AnalogueGain": 16.0,
    # Disable all control loops which take multiple exposures to settle.
    "AeEnable": False,  # disable auto-exposure control loop which would need many exposures for one image!
    "AwbEnable": False, # disable the auto-widebalance control loop
    # "AfMode": "MANUAL", # For cameras with auto-focus we also need to disable the auto-focus control loop.
    # The following is just to be sure.
    "ExposureValue": 0.0, # not sure if this has an effect on raw images
    "NoiseReductionMode": controls.draft.NoiseReductionModeEnum.Off, # disable "star-eater" - should not have an effect on raw images but who knows
}
picam2.set_controls(controls)

# Take a number of images in a loop.
for n in range(5):
    TimeStamp_Start = datetime.datetime.now()

    # Start camera.
    picam2.start()

    # Start an exposure. We use the asynchronous capture functions because the "indi_pylibcamera" needs to
    # stay responsive during exposure.
    job = picam2.capture_arrays(
        ["raw"],
        wait=False, # don't block
        # a "signal_function" can be given here which gets called at exposure end
    )

    # Wait until end of exposure and read image.
    (array, ), metadata = picam2.wait(job)

    # "array" is a numpy array with raw image data. "metadata" describes this image.
    # You can examine the meta data with "pprint.pprint(metadata)"

    # Stop camera.
    picam2.stop()

    TimeStamp_Stop = datetime.datetime.now()
    TimeNeeded = (TimeStamp_Stop - TimeStamp_Start).total_seconds()

    # Report CPU time needed for exposure. This should not differ too much from image exposure time!
    print(f'Exposure {n+1} took {TimeNeeded:.3f} sec, {array.mean()=:.3f}, {array.std()=:.3f}.')

    # If needed: wait some time between exposures to simulate frame processing.
    time.sleep(2.0)  # unit: seconds

print("Finished.")

Thank You.