raspberrypi/rpicam-apps #527: API to get at CameraManager or Camera list

anjok, 2023-07-01:
For the INDI libcamera driver, we need to get at the camera list before actually opening a camera. Also, we need to be able to Open and Close the camera on a regular basis. Unfortunately, the API that is present needs the camera manager as input.

Can you add a means for that to get the list? Currently, I cloned the code into the driver and  added a method to get a pointer to CameraManager and run the same code that you run for discovering cameras:

https://github.com/indilib/indi-3rdparty/blob/master/indi-libcamera/libcamera-apps/core/libcamera_app.hpp#L118
https://github.com/indilib/indi-3rdparty/blob/master/indi-libcamera/indi_libcamera.cpp#L595C22-L595C22

I can't reuse the one you added to LibCameraApp::GetCameraList() as it takes a std::unique_pointer argument and I can't return the one you created for the instance.

This is of course not very usable, as you'll need to recompile this each time something changes.

naushir, 2023-07-01:
Hi, unfortunately it's a bit hard for me to guess exactly what API changes you need here without more context.   If you are able to create a basic PR with the changes, we would be happy to discuss this in more detail and merge the appropriate changes.

Vulisha, 2023-07-01:
Hi, let me jump in with probable guess, anjok may confirm if my suspicions are true. 

In INDI we do not know what is the camera is used as many may be used, and INDI is. made as universal driver so we need to get info what model of camera is connected and its properties before we Open the camera, and in current api we need to know Camera before we can start it and use GetCameraList.

naushir, 2023-07-02:
The only way to enumerate cameras in libcamera is through the camera manager.

Again, I am a bit unclear on how you want to do things in your INDI framework - I've never used it before. If you could create a PR with some proposed API changes that would work for you we can get something suitable merged for your. The PR does not have to be functionally complete to start with.

anjok, 2023-07-02:
Here you go:

https://github.com/raspberrypi/libcamera-apps/pull/528/files

I  have no idea if this is how std:unique_ptr is supposed to get used, though. Since we have an instance of LibcameraApp in the driver, we might as well add an instance method 

	std::vector<std::shared_ptr<libcamera::Camera>> GetCameras();

which calls the static method with it's CameraManager, but my std:: C++ skills aren't that up to speed.

anjok, 2023-07-02:
The PR above is what I added to out local copy of the source files. If you'd add something like this, we don'T have to maintain that copy anymore (provided the user has whatever version of the libcamera-apps lib then contains it)

EricClaeys, 2023-07-08:
What timing, I was just going to create an Issue to request a way to query capabilities of the attached cameras.

Our Allsky software calls `libcamera-still` with options to set shutter speed, contrast, does it support temperature, does it support focus, etc. etc.

We have a front end where users enter those values which our program then adds to the `libcamera-still` command line.
The front end lists the minimum value, maximum value, and default for all the camera's capabilities that have that info.  For example, contrast goes from 0 to 15.99 with a default of 1.

We currently have that data hard-coded for every RPi or RPi-compatible camera, which is a nightmare and very prone to error.  It also means we can't support new cameras until we add that data.

What I'm requesting is some way to query all that information and have it returned in either a "txt" or "json" format, as is done with the metadata.  I can provide all the info we'd need.

naushir, 2023-07-10:
@anjok https://github.com/raspberrypi/libcamera-apps/pull/528 should now be merged.

@EricClaeys there is a way to get this information (with a caveat).  The output of ``libcamera-hello --list-cameras -v`` will give you something like what you want as printed below:
```
pi@pi4:~/libcamera-apps/build $ libcamera-hello --list-cameras -v
Available cameras
-----------------
0 : imx477 [4056x3040] (/base/soc/i2c0mux/i2c@1/imx477@1a)
    Modes: 'SRGGB10_CSI2P' : 1332x990 [120.05 fps - (696, 528)/2664x1980 crop]
           'SRGGB12_CSI2P' : 2028x1080 [50.03 fps - (0, 440)/4056x2160 crop]
                             2028x1520 [40.01 fps - (0, 0)/4056x3040 crop]
                             4056x3040 [10.00 fps - (0, 0)/4056x3040 crop]

    Available controls for 4056x3040 SRGGB12_CSI2P mode:
    ----------------------------------------------------
    AeConstraintMode : [0..3]
    AeEnable : [false..true]
    AeExposureMode : [0..3]
    AeMeteringMode : [0..3]
    AnalogueGain : [1.000000..22.260870]
    AwbEnable : [false..true]
    AwbMode : [0..7]
    Brightness : [-1.000000..1.000000]
    ColourGains : [0.000000..32.000000]
    Contrast : [0.000000..32.000000]
    ExposureTime : [114..674191602]
    ExposureValue : [-8.000000..8.000000]
    FrameDurationLimits : [100000..694434742]
    NoiseReductionMode : [0..4]
    Saturation : [0.000000..32.000000]
    ScalerCrop : [(0, 0)/64x64..(0, 0)/4056x3040]
    Sharpness : [0.000000..16.000000]
```

The code that retrieves the properties can be found here:
https://github.com/raspberrypi/libcamera-apps/blob/69122be6235f1a757cd16b8ecde367981db4afbd/core/options.cpp#L222

The caveat here is that the settings above are strictly mode specific and not necessarily global for all modes.  So you might have to configure each mode and retrieve the properties individually one-by-one if you need to.

naushir, 2023-07-10:
@EricClaeys if you need more help with this feel free to open a new issue.  I'll close this one down now as the original problem is resolved.

closed 2023-07-11 by naushir