raspberrypi/rpicam-apps #533: [FEATURE REQUEST] Additional entries in "--list-cameras -v"

EricClaeys, 2023-07-14:
This is a follow-on to Issue #527.

I assume the `-v` option to `libcamera-hello --list-cameras` is new?  It's not on my Bullseye from May.

Can the following information be added to the output?

* A line indicating if a camera supports focus?  Perhaps `Focus :[true]`
* For color sensors, a line for the bayer matrix so we don't have to parse the "Modes" lines?  e.g., `bayer : [RGGB]`
* A separate line for the bit depth so we don't have to parse the "Modes" lines? e.g., `bitdepth : [12]`
* Lines indicating the rotation and / or flip the sensor (or software) supports.
* Will `ColourGains ` only be in the output for color cameras?  I realize all the currently supported sensors are color but in case there is ever a mono sensor I'd like to be able to detect that.

naushir, 2023-07-14:
> I assume the -v option to libcamera-hello --list-cameras is new? It's not on my Bullseye from May.

Correct, this will be available in our next release that is being prepared right now.

Most what you want is already directly available.  Here's an example for the Camera V3 output:

```
Available cameras
-----------------
0 : imx708_wide [4608x2592] (/base/soc/i2c0mux/i2c@1/imx708@1a)
    Modes: 'SRGGB10_CSI2P' : 1536x864 [120.13 fps - (768, 432)/3072x1728 crop]
                             2304x1296 [56.03 fps - (0, 0)/4608x2592 crop]
                             4608x2592 [14.35 fps - (0, 0)/4608x2592 crop]

    Available controls for 4608x2592 SRGGB10_CSI2P mode:
    ----------------------------------------------------
    AeConstraintMode : [0..3]
    AeEnable : [false..true]
    AeExposureMode : [0..3]
    AeMeteringMode : [0..3]
    AfMetering : [0..1]
    AfMode : [0..2]
    AfPause : [0..2]
    AfRange : [0..2]
    AfSpeed : [0..1]
    AfTrigger : [0..1]
    AfWindows : [(0, 0)/0x0..(65535, 65535)/65535x65535]
    AnalogueGain : [1.122807..16.000000]
    AwbEnable : [false..true]
    AwbMode : [0..7]
    Brightness : [-1.000000..1.000000]
    ColourGains : [0.000000..32.000000]
    Contrast : [0.000000..32.000000]
    ExposureTime : [26..112015443]
    ExposureValue : [-8.000000..8.000000]
    FrameDurationLimits : [69669..220535845]
    LensPosition : [0.000000..32.000000]
    NoiseReductionMode : [0..4]
    Saturation : [0.000000..32.000000]
    ScalerCrop : [(0, 0)/64x64..(0, 0)/4608x2592]
    Sharpness : [0.000000..16.000000
```

> * A line indicating if a camera supports focus?  Perhaps `Focus :[true]`

None of the AF controls (e.g. AfMode, AfTrigger, ...) will be present if there is no actuator present in the module.

> * Lines indicating the rotation and / or flip the sensor (or software) supports.

All cameras will always support horizontal and vertical flips unconditionally.  Hence there's no need for a control here.

> * Will `ColourGains ` only be in the output for color cameras?  I realize all the currently supported sensors are color but in case 
there is ever a mono sensor I'd like to be able to detect that.

Correct, colour related controls (e.g. ColourGains, Saturation, AwbMode, etc...) will not be present in the list for a mono sensor.

> * For color sensors, a line for the bayer matrix so we don't have to parse the "Modes" lines?  e.g., `bayer : [RGGB]`
> * A separate line for the bit depth so we don't have to parse the "Modes" lines? e.g., `bitdepth : [12]`

These 2 are the ones that are not directly available.  As you said, parsing the mode lines is possible but a bit cumbersome.  I'll see if I can change things to perhaps include it like follows:
```
0 : imx708_wide [4608x2592 10-bit RGGB] (/base/soc/i2c0mux/i2c@1/imx708@1a)
```

naushir, 2023-07-14:
https://github.com/raspberrypi/libcamera-apps/pull/535 ought to add the bayer order and bit depth to the output now.

EricClaeys, 2023-07-14:
Thanks @naushir.  I appreciate your team being willing to implement changes for the Allsky project.

EricClaeys, 2023-07-15:
@naushir   Roughly when will users be able to get the new version with `--list-cameras -v` ?
And I assume they'll need to manually do an "apt upgrade" ?

EricClaeys, 2023-07-15:
Will the line with the camera number, e.g.,
~~~
0 : imx708_wide [4608x2592 10-bit RGGB] (/base/soc/i2c0mux/i2c@1/imx708@1a)
~~~
ALWAYS have the three items in brackets?  For example, can I safely do:
~~~
sscanf(string, "%d : %s [%dx%d %d-bit %s]" ...
~~~
and be guaranteed I'll get 6 matches?
If not, would it be possible for the bit-depth and bayer pattern to be on separate lines so no fancy parsing is needed?

Also, looking at the code, mono cameras have `MONO` for the bayer pattern?

Eric

naushir, 2023-07-17:
Hope to get the release out at some point this week.  However, he specific change for the bit depth and bayer order unfortunately did not make it into this release so we will have to wait for the next release for that specific change.

naushir, 2023-07-17:
> Will the line with the camera number, e.g.,
> 
> ```
> 0 : imx708_wide [4608x2592 10-bit RGGB] (/base/soc/i2c0mux/i2c@1/imx708@1a)
> ```
> 
> ALWAYS have the three items in brackets? For example, can I safely do:
> 
> ```
> sscanf(string, "%d : %s [%dx%d %d-bit %s]" ...
> ```
> 
> and be guaranteed I'll get 6 matches?

As long as the libcamera APIs don't change and hide the relevant information, this string should always have the same format.  And ``MONO`` string will be used for mono sensors.

naushir, 2023-07-18:
These changes have now been merged, and eventually will get to the apt packages.  Closing this now...

closed 2023-07-18 by naushir

EricClaeys, 2023-08-18:
@naushir   I have version 1.2.1.1 of libcamera-aps (which apt upgrade says is the latest) but the bit depth and bayer aren't on the first line.  I get:
~~~
0 : imx708_wide [4608x2592] (/base/soc/i2c0mux/i2c@1/imx708@1a)
~~~
Have your changes made it to apt yet?