raspberrypi/rpicam-apps #534: ERROR: *** no cameras available *** bullseye

mart-e, 2023-07-14:
This is a followup of the old #168 that was closed for inactivity

**Describe the bug**
I have a raspberry pi 4 with the camera module 2.
I have an install using Raspberry Pi OS Bullseye on a SSD and I can not get the camera to work.
Note that I have a previous install on an SD card where the camera does work, so I know it's not a faulty camera.

```
$ libcamera-hello
Made X/EGL preview window
[0:15:00.576343596] [6374] INFO Camera camera_manager.cpp:299 libcamera v0.0.4+22-923f5d70
ERROR: *** no cameras available ***
```

**Bug report**
```
$ camera-bug-report -t 5 -o report.txt -c "libcamera-hello"
```
[report.txt](https://github.com/raspberrypi/libcamera-apps/files/12048540/report.txt)

mart-e, 2023-07-14:
Ok I think I found it. I compared both reports and it seems that in the old one I had

```
# Automatically load overlays for detected cameras
camera_auto_detect=1
```
and in the new
```
# Automatically load overlays for detected cameras
start_x=1
```
No idea where this difference comes from but adding `camera_auto_detect=1` between the two lines seems to fix my issue.

Leaving open if one wants to add information on what caused this but feel free to close it if nothing to add :)

naushir, 2023-07-14:
Thanks for the update.  I'll close this down.  Users are always free to use the search function if they want to look at past issue :)

closed 2023-07-14 by naushir