raspberrypi/rpicam-apps #539: mjpeg - high cpu usage

Lillifee, 2023-07-19:
Hey, first of all, thanks for your great work!

I'm currently developing a web application that allows users to stream, capture pictures, and record videos from their Raspberry Pi camera via a web interface using libcamera.  [RaspiCam](https://github.com/Lillifee/raspiCam) 

As a fallback option to h264, I use MJPEG for streaming the live video. However, I've noticed that the MJPEG codec tends to result in high CPU usage. (https://github.com/Lillifee/raspiCam/issues/58) Is this behavior normal? I'm curious to know if there are any plans to support a hardware-accelerated MJPEG encoder in the future. Alternatively, do you have any suggestions on how to reduce CPU usage and enhance overall performance? These findings are based on testing with a Raspberry Pi 4 and an HQ camera.

```
libcamera-vid --denoise cdn_off --nopreview --width 1640 --height 1232 --codec MJPEG --timeout 0 --output ./test.mjpeg
```
[image]

Comparing to other solutions such as the camera-streamer, which seems to utilizes hardware encoding for the MJPEG stream, the CPU usage with the current MJPEG settings is considerably high. In the camera-streamer implementation available on https://github.com/ayufan/camera-streamer, the CPU usage is around 10%.

Here is the result of the camera-bug-report, but I believe it may not be pertinent to this particular bug report. [bug.txt](https://github.com/raspberrypi/libcamera-apps/files/12091378/bug.txt)

Thanks in advance!

naushir, 2023-07-19:
You are correct about the fact that libcamera-vid uses a software JPEG encoder (libjpeg/libjpeg-turbo) so the CPU usage is significantly worse than the HW encoder variant.  Unfortunately at this time (and because of compatibility reasons), we don't have concrete plans to implement HW jpeg encoding in libcamera-apps.

One thing you can try to help reduce CPU usage is to lower the number of threads used for the software encoder.  You can do this by adjusting ``NUM_THREADS`` in ``mjpeg_encoder.hpp`` here:
https://github.com/raspberrypi/libcamera-apps/blob/07fc4619dc107384ee072f34db3be2d6ae297e2a/encoder/mjpeg_encoder.hpp#L29

Lillifee, 2023-07-19:
Ok i understand. 👍
Thank you so much for the fast response!

closed 2023-07-19 by Lillifee