raspberrypi/rpicam-apps #542: How are we supposed to compile the source code from now on?

FacuM, 2023-07-22:
The recently pushed commit 9e17265 implies that we shouldn't rely on CMake files anymore in order to build the project.

My question is, then, how are we supposed to build it? The official blog post provided [here](https://www.raspberrypi.com/documentation/computers/camera_software.html#building-libcamera-apps) doesn't provide an alternative to the CMake-based compilation procedure.

I think this commit should either be **reverted** or a proper method be published ASAP in order to avoid breaking workflows that compile on bleeding-edge changes.

naushir, 2023-07-22:
The documentation changes have already been made - see here: https://github.com/raspberrypi/documentation/pull/2992.
There's a wait on getting it published on the documentation website now, that's it.

I should note that since libcamera does not yet have a stable API/ABI, you will inevitably hit unavoidable breakages if you work on the bleeding edge of the tree.  This is why we publish official releases for our users.

closed 2023-07-22 by naushir

FacuM, 2023-07-22:
> The documentation changes have already been made - see here: [raspberrypi/documentation#2992](https://github.com/raspberrypi/documentation/pull/2992). There's a wait on getting it published on the documentation website now, that's it.
> 
> I should note that since libcamera does not yet have a stable API/ABI, you will inevitably hit unavoidable breakages if you work on the bleeding edge of the tree. This is why we publish official releases for our users.

Got it, thanks, I'm rewriting my build script now!