raspberrypi/rpicam-apps #543: [BUG] Update your docs before removing cmake...

Vulisha, 2023-07-23:
**Describe the bug**
 You have removed cmake files and kept documentation explaining how to build with cmake and no docs how to build with meson/ninja... Plain simple as that. 

Please prepare documentation before removing something or do not remove it untill docs are ready... I am capable of reverting to. previous commit but many peoole are not. And I. am. forced to build because you introduced breaking change with changing to libcamera.so0.1.0 and that broke all dependancies.

More care should be taken when introducing breaking changes as well ensure backwards compatibility as much as possible.

naushir, 2023-07-23:
The documentation changes have already been made - see here: https://github.com/raspberrypi/documentation/pull/2992.
There's a wait on getting it published on the documentation website now, that's it.

I should note that since libcamera does not yet have a stable API/ABI, you will inevitably hit unavoidable breakages if you work on the bleeding edge of the tree. This is why we publish official releases for our users.

closed 2023-07-23 by naushir

Vulisha, 2023-07-24:
Well TBH you could have updated readme.MD file and just note that new build guide is on the way. I have lost few hours that I could have used on good opensource dev, on trying to find out why everything broke down and how could I be so incompetent and do a pull something wrong or what...