raspberrypi/rpicam-apps #545: libcamera-apps find no camera if gpu_mem=16 in /boot/config.txt

k323r, 2023-07-31:
**Describe the bug**
If gpu_mem is set to 16 either manually or through `raspi-config` the libcamera-app, such as `libcamera-hello`, fail to find a camera. 

If `gpu_mem` is set to 32, it works. While it makes sense, that a minimum of GPS memory might be necessary for the libcamera-apps, the error message can be misleading. 

**hardware**
- raspberry pi 4 4GB RAM
- raspberry pi 3 cam noir wide angle
- raspberry pi 3 cam noir
- raspberry pi 2 cam

**`camera-bug-report` output:**
With 16 MB of RAM: [16MB_RAM_libcamera-bug.txt](https://github.com/raspberrypi/libcamera-apps/files/12217636/16MB_RAM_libcamera-bug.txt)
With 32 MB of RAM (working): [32MB_RAM_libcamera-bug.txt](https://github.com/raspberrypi/libcamera-apps/files/12217645/32MB_RAM_libcamera-bug.txt)

**additional context**
we are building a video tracking application and love the "new" libcamera stack, great work and thanks!

naushir, 2023-08-01:
``gpu_mem`` is not something that users ought to play with.  In fact, changing the value can and will have a very detrimental effect on system operation as you found out.  Camera buffer allocations happen from the CMA heap, which is entirely different and unrelated to the ``gpu_mem`` param.

Can you expand on what you are trying to achieve here - is there a specific reason to lower ``gpu_mem``?

k323r, 2023-08-01:
I was experimenting with the `gpu_mem` parameter as I was dealing with an `could not allocate memory` bug in a call to OpenCV. I was thrown off by `libcamera-hello` throwing a `no cameras aviailable` error. I first thought it was a hardware error and only, after some time, found that if I adjusted gpu_mem to values > 16 MB did the error go away. Knowing that libcamera does not actually use RAM reserved by `gpu_mem` I thought it might be a useful bug report.

naushir, 2023-08-01:
Thanks, yes all these options and  parameters can be very confusing at times!  If there is anything more we can help with on this issue, let me know.  If not, feel free to close this down.

k323r, 2023-08-01:
Thanks for the swift response, any idea, why the `no camera found` error actually gets thrown if only 16 MB of GPU RAM is available? Again, thanks for the great software and all the effort!

closed 2023-08-01 by k323r

naushir, 2023-08-01:
I think the reason is because we allocate a small amount of memory for  state objects for various multimedia components in the firmware through the gpu_mem heap.  If these allocations fail, the entire functionality of the block will be disabled (e.g. camera in this case).

6by9, 2023-08-01:
Taking gpu_mem down to 16MB switches to the cutdown firmware. That removes all the camera probing (could be overcome by manually adding dtoverlay=imx708 or similar to config.txt), but also the shared memory service to the firmware, and the ISP and codecs drivers. With no ISP, libcamera isn't going to get very far.

https://www.raspberrypi.com/documentation/computers/configuration.html#start-elf-start_x-elf-start_db-elf-start_cd-elf-start4-elf-start4x-elf-start4db-elf-start4cd-elf
> start_cd.elf removes support for hardware blocks such as cameras, codecs and 3D as well as having initial framebuffer limitations.

k323r, 2023-08-02:
Thanks for the explanations @naushir @6by9, learned a ton!