raspberrypi/rpicam-apps #546: [BUG] Cannot achieve high framerate video recording using Camera Module v3

azsde, 2023-08-01:
Hello everyone,

I have a Camera Module V3 plus a Raspberry Pi 3b+, and I wish to record video at the maximum supported framerate by a specific resolution.

The official [camera module 3 specification](https://www.raspberrypi.com/products/camera-module-3) says:

`Common video modes: 1080p50, 720p100, 480p120`

So, from what I understand, the camera module V3 can be used to record 1080p footage at 50 FPS.

So I followed the official [instructions](https://www.raspberrypi.com/documentation/computers/camera_software.html#high-framerate-capture) and tried the following command:

`libcamera-vid --level 4.2 --framerate 50 --width 1920 --height 1080 --save-pts timestamp.pts -o video.264 -t 10000 --denoise cdn_off -n`

Then, when trying to play the .264 file using VLC, the output video doesn't have any duration and play faster than it should, like if the footage was sped up.

I then tried using ffmpeg to "convert" the .264 file into a .mp4 file:

`ffmpeg -i video.264 -r 50 -c:v copy video.mp4`

This results in a 6 seconds long video instead of a 10 seconds one, and the video still plays as if the video was sped up.

[image]
I then tried to use mkvmerge instead:

`mkvmerge -o video.mkv --timecodes 0:timestamp.pts video.264`

This results in a 10 seconds long video, but with a 33.663 FPS framerate instead of 50 FPS. The video plays just fine (the footage doesn't seem sped up), but this is not the wanted FPS.

[image]

Am I doing something wrong here ? Or is it a bug ?

I've attached the [bug.txt](https://github.com/raspberrypi/libcamera-apps/files/12229297/bug.txt) report to this issue.

naushir, 2023-08-01:
First I would suggest that you remove the ``-t 10000`` and use ``--frames 300`` (for example).  The ``-t`` option does not give you exact control of the encoded stream, it's merely a timer at the application level to control shutdown.  ``--frames`` give an exact number of frames to encode.

Secondly, you should try creating an mkv directly from ``libcamera-vid`` to avoid having to generate a container externally.  This will embed the correct timestamps with the frame data directly:

```
libcamera-vid --level 4.2 --framerate 50 --width 1920 --height 1080 --codec libav -o video.mkv --frames 300 --denoise cdn_off -n
```

Can you report on your results using the above command line?

azsde, 2023-08-01:
Trying to use `--codec libav` result in a very low quality video:

[image]

naushir, 2023-08-01:
Sorry, you should set the video bitrate ``--bitrate`` to an appropriate value as well.  Perhaps something like 2 to 5 Mbps?

azsde, 2023-08-01:
Setting the bitrate does result in a normal video, that being said, MediaInfo does not report the FPS when inspecting the video, is there a way to check besides MediaInfo ?

naushir, 2023-08-01:
You can use a script provided in this repo:
https://github.com/raspberrypi/libcamera-apps/blob/main/utils/timestamp.py

That will will extract all timestamp information from the container file.

azsde, 2023-08-01:
Unfortunately I get the following error:

```
python3 timestamp.py video.mkv

Traceback (most recent call last):
  File "/home/azsde/timestamp.py", line 72, in <module>
    avg = sum(diffs) / len(diffs)
ZeroDivisionError: division by zero
```

naushir, 2023-08-01:
Hmmm, that should not happen.  Would you be able to attach this file for me to have a look at please?

naushir, 2023-08-01:
An alternative would be to use your original command line (but use ``--frames`` instead of ``-t``) and run ``python3 timestamp.py timestamp.pts``.

azsde, 2023-08-01:
The same issue is happening with the timestamp.pts file: 
[timestamp.zip](https://github.com/raspberrypi/libcamera-apps/files/12230050/timestamp.zip)

naushir, 2023-08-01:
Something seems to have changed in the container timestamp format.  You will need to change the script as follows to get it working:
```diff
diff --git a/utils/timestamp.py b/utils/timestamp.py
index a486ae8b85fb..024396f75078 100755
--- a/utils/timestamp.py
+++ b/utils/timestamp.py
@@ -21,7 +21,7 @@ def read_times_pts(file):

 def read_times_container(file):
-    cmd = ['ffprobe', file, '-hide_banner', '-select_streams', 'v', '-show_entries', 'frame=pts_time', '-of', 'csv=p=0']
+    cmd = ['ffprobe', file, '-hide_banner', '-select_streams', 'v', '-show_entries', 'frame=pkt_pts_time', '-of', 'csv=p=0']
     r = subprocess.run(cmd, stdout=subprocess.PIPE, stderr=subprocess.DEVNULL, text=True)
     if r.returncode:
         raise RuntimeError(f'ffprobe failed to run with command:\n{" ".join(cmd)}')
```

naushir, 2023-08-01:
With the above change, I get the following output:
```
Total: 300 frames (299 samples)
Average: 30.281 ms / 33.024 fps
Minimum: 20.000 ms at frame 266
Maximum: 59.000 ms at frame 290
Outliers: 0 (100%) 299 (10.0%) 299 (1.0%) 299 (0.1%)
```
So with frame drops, the record framerate comes out at 33.024 fps.

azsde, 2023-08-01:
Alright, so what is most likely happening here is the hardware limitation of the RPI3 b+ are not allowing me to go beyond ~33 FPS in 1080p ?

Should I try the same command on a RPI 4 ?

naushir, 2023-08-01:
Yes, I would suggest either reduce the resolution/framerate or try a Pi 4 platform.

PS I'm now going to be away from work so unfortunately won't be able to look at this until I'm back in a couple of weeks.

closed 2023-08-01 by azsde