raspberrypi/rpicam-apps #547: [BUG] libcamera-apps not compiling due to a member in 'controls'

fabianmpa, 2023-08-01:
Compiling with instructions from https://www.raspberrypi.com/documentation/computers/camera_software.html#building-libcamera-apps throws an error in file **libcamera_app.cpp**

Pre conditions:

1.  Install libcamera through:
>   meson build --buildtype=release -Dpipelines=rpi/vc4 -Dipas=rpi/vc4 -Dv4l2=true -Dgstreamer=enabled -Dtest=false -Dlc-compliance=disabled -Dcam=disabled -Dqcam=enabled -Ddocumentation=disabled -Dpycamera=enabled -Dudev=auto

> ninja -C build -j2

> sudo ninja -C build install

Now, when trying to compile libcamera-apps I am running:

> meson setup build -Denable_libav=true -Denable_drm=true -Denable_egl=true -Denable_qt=true -Denable_opencv=true -Denable_tflite=false

> meson compile -C build -j2

and that gives me the following output:

INFO: autodetecting backend as ninja
INFO: calculating backend command to run: /usr/bin/ninja -C /home/pi4/libcamera-apps/build -j 2
ninja: Entering directory `/home/pi4/libcamera-apps/build'
[3/47] Compiling C++ object libcamera_app.so.1.2.1.p/core_libcamera_app.cpp.o
FAILED: libcamera_app.so.1.2.1.p/core_libcamera_app.cpp.o 
c++ -Ilibcamera_app.so.1.2.1.p -I. -I.. -Icore -I/usr/local/include/libcamera -I/usr/include -I/usr/include/aarch64-linux-gnu -I/usr/include/libpng16 -I/usr/include/libdrm -I/usr/local/include -I/usr/include/aarch64-linux-gnu/qt5/QtCore -I/usr/include/aarch64-linux-gnu/qt5 -I/usr/include/aarch64-linux-gnu/qt5/QtWidgets -I/usr/include/aarch64-linux-gnu/qt5/QtGui -I/usr/local/include/opencv4 -fdiagnostics-color=always -Wall -Winvalid-pch -Wextra -Wpedantic -Werror -std=c++17 -O3 -pedantic -Wno-unused-parameter -faligned-new -D_FILE_OFFSET_BITS=64 -Wno-psabi -ftree-vectorize -DLIBAV_PRESENT=1 -DLIBDRM_PRESENT=1 -DLIBEGL_PRESENT=1 -DQT_PRESENT=1 -fPIC -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CORE_LIB -pthread -DBOOST_ALL_NO_LIB -MD -MQ libcamera_app.so.1.2.1.p/core_libcamera_app.cpp.o -MF libcamera_app.so.1.2.1.p/core_libcamera_app.cpp.o.d -o libcamera_app.so.1.2.1.p/core_libcamera_app.cpp.o -c ../core/libcamera_app.cpp
../core/libcamera_app.cpp: In member function ‘void LibcameraApp::StartCamera()’:
../core/libcamera_app.cpp:623:59: error: ‘AeFlickerMode’ is not a member of ‘controls’
  623 |  if (options_->flicker_period && !controls_.get(controls::AeFlickerMode) &&
      |                                                           ^~~~~~~~~~~~~
../core/libcamera_app.cpp:624:39: error: ‘AeFlickerMode’ is not a member of ‘controls’
  624 |   camera_->controls().find(&controls::AeFlickerMode) != camera_->controls().end() &&
      |                                       ^~~~~~~~~~~~~
../core/libcamera_app.cpp:625:39: error: ‘AeFlickerPeriod’ is not a member of ‘controls’
  625 |   camera_->controls().find(&controls::AeFlickerPeriod) != camera_->controls().end())
      |                                       ^~~~~~~~~~~~~~~
../core/libcamera_app.cpp:627:27: error: ‘AeFlickerMode’ is not a member of ‘controls’
  627 |   controls_.set(controls::AeFlickerMode, controls::FlickerManual);
      |                           ^~~~~~~~~~~~~
../core/libcamera_app.cpp:627:52: error: ‘FlickerManual’ is not a member of ‘controls’
  627 |   controls_.set(controls::AeFlickerMode, controls::FlickerManual);
      |                                                    ^~~~~~~~~~~~~
../core/libcamera_app.cpp:628:27: error: ‘AeFlickerPeriod’ is not a member of ‘controls’
  628 |   controls_.set(controls::AeFlickerPeriod, options_->flicker_period.get<std::chrono::microseconds>());
      |                           ^~~~~~~~~~~~~~~
[4/47] Compiling C++ object libcamera_app.so.1.2.1.p/core_post_processor.cpp.o
ninja: build stopped: subcommand failed.

Hardware used: Raspberry Pi 4 Model B 2GB
OS: Bullseye 64 bit flashed through Imager

naushir, 2023-08-02:
As @6by9 said, if you want to build and run with top of tree libcamera-apps, you will have to build and install libcamera from top of tree if the upstream repo (https://git.linuxtv.org/libcamera.git/) to pull in the new flicker controls.

UmonDaisuke, 2023-08-02:
Hello,
thansk a lot for ractive support. I tried with repo (https://git.linuxtv.org/libcamera.git) and compilation was diretly fine without any code update. Nevertheless, "/user/local/bin/libcamera-vid -o test1.h264 -t 5000 -n" is still generating ERROR: *** Invalid time string provided *** while "/user/bin/libcamera-vid -o test1.h264 -t 5000 -n" is fine. Do you me top open a new issue ?

here is my libbcamera compilation process:

#!/bin/bash
set -e
echo "Compilation of libcamera for 32-bit OS"
sudo rm -rf libcamera
sudo apt install -y python3-pip git
sudo pip3 install jinja2
sudo apt install -y libboost-dev
sudo apt install -y libgnutls28-dev openssl libtiff5-dev pybind11-dev
#sudo apt install -y qtbase5-dev libqt5core5a libqt5gui5 libqt5widgets5
sudo apt install -y meson
sudo apt install -y cmake
sudo pip3 install pyyaml ply
sudo pip3 install --upgrade meson
#sudo apt install -y libglib2.0-dev libgstreamer-plugins-base1.0-dev
git clone https://git.linuxtv.org/libcamera.git
cd libcamera
meson build --buildtype=release -Dpipelines=rpi/vc4 -Dipas=rpi/vc4 -Dv4l2=true -Dgstreamer=disabled -Dtest=false -Dlc-compliance=disabled -Dcam=disabled -Dqcam=disabled -Ddocumentation=disabled -Dpycamera=enabled
ninja -C build   # use -j 2 on Raspberry Pi 3 or earlier devices
sudo ninja -C build install

6by9, 2023-08-02:
Different issue.
It sounds like https://github.com/raspberrypi/libcamera-apps/pull/535/commits/79e850bdfd4832f3844d45cba048adc2bb68772a isn't quite as backward compatible as hoped.

Altering your command line to use ```-t 5000ms``` or ```-t 5s``` would work around it.

UmonDaisuke, 2023-08-02:
Built "/user/local/bin/libcamera-vid" works with --version,--help, --list-cameras but  the simplest command `/user/local/bin/libcamera-vid -o test1.h264` fails even without -t parameter or `-t 5s` has no effect. Do you think of any log I can collect for supporting debbugging?

naushir, 2023-08-02:
Hmmm, I've tested many permutations (including the one above) to ensure backward compatibility. Not sure what's gone wrong here.

I'll only be able to check this once I'm back from holiday in a couple of weeks.

UmonDaisuke, 2023-08-04:
Hi,
this is solved. Here is the proess I used for compiling only libcamera-apps (and skipping the last commit)
Thanks a lot to @naushir  and @6by9  for support
Enjoy holidays

 #!/bin/bash
set -e
#echo "Updating and upgrading"
sudo apt update -y
sudo apt upgrade -y

#echo "Installing tools"
sudo apt install -y cmake python3-pip git
sudo pip3 install ninja meson

#echo "Downloading libraries"
sudo apt install -y libcamera-dev libepoxy-dev libjpeg-dev libtiff5-dev
sudo apt install -y qtbase5-dev libqt5core5a libqt5gui5 libqt5widgets5
sudo apt install -y libboost-program-options-dev libdrm-dev libexif-dev
sudo apt install -y libavcodec-dev libavdevice-dev libavformat-dev libswresample-dev

#echo "Downloading source"
rm -rf libcamera-apps
git clone https://github.com/raspberrypi/libcamera-apps.git
cd libcamera-apps
git checkout 9e172655e257972e5da45d59a720192516f7d769

echo "Compiling libamera-apps"
cd libcamera-apps/
meson setup build -Denable_libav=true -Denable_drm=true -Denable_egl=false -Denable_qt=false -Denable_opencv=false -Denable_tflite=false -Dneon_flags=armv8-neon
meson compile -C build # use -j1 on Raspberry Pi 3 or earlier devices
sudo meson install -C build
sudo ldconfig # this is only necessary on the first build
echo "¨Please run a new terminal window for testing"

naushir, 2023-08-22:
Resolving now, feel free to re-open another issue for the ``-t`` argument if you continue having problems.

closed 2023-08-22 by naushir

jacksmi1961, 2023-09-24:
For those dealing with ERROR: *** Invalid time string provided ***, it is due to LIBAV not being included. If the preprocessor LIBAV_PRESENT is not defined then the libav control options do not get initialized and then empty strings get parsed as time strings and the exception occurs. Including #define LIBAV_PRESENT 1 in your code fixes the issue