raspberrypi/rpicam-apps #548: [QUESTION] -v Options versus Available Controls

EricClaeys, 2023-08-03:
Is there a better way to ask questions about libcamera?  
&nbsp;

The `-v` option to libcamera-still outputs two main sections:
1. **Options**
2. **Available controls**

If I run `libcamera-still -v`, is everything in the `Options` section (except "verbose") the default value?  For example, `contrast: 1`.  Is 1 the default?

I need to determine the default value for each Available Control so am trying to map it to its corresponding Option.  Would you please review this and let me know of any mistakes and fill in the missing pieces?  This is for the Module 3 camera.

Thanks - Eric

~~~
NoiseReductionMode : [0..4]
	denoise: auto
ScalerCrop : [(0, 0)/128x128..(0, 0)/4608x2592]
	??
Sharpness : [0.000000..16.000000]
	sharpness: 1
AwbEnable : [false..true]
	awb: auto
AwbMode : [0..7]
	??
FrameDurationLimits : [17849..112075593]
	??   framerate: 30 ??
ExposureValue : [-8.000000..8.000000]
	ev: 0
Brightness : [-1.000000..1.000000]
	brightness: 0
AfMode : [0..2]
	??
AfWindows : [(0, 0)/0x0..(65535, 65535)/65535x65535]
	??   autofocus-window: all ??
AfSpeed : [0..1]
	autofocus-speed: normal
AfTrigger : [0..1]
	??
AfRange : [0..2]
	autofocus-range: normal
AfPause : [0..2]
	??
AfMetering : [0..1]
	??
LensPosition : [0.000000..32.000000]
	??
ExposureTime : [13..0]
	??
AeExposureMode : [0..3]
	exposure: normal
AeEnable : [false..true]
	??
AeConstraintMode : [0..3]
	??
AeMeteringMode : [0..3]
	metering: centre
Contrast : [0.000000..32.000000]
	contrast: 1
Saturation : [0.000000..32.000000]
	saturation: 1
AnalogueGain : [1.122807..16.000000]
	??	Isn't the minimum gain 1.0 ?
ColourGains : [0.000000..32.000000]
	??
~~~

naushir, 2023-08-14:
``NoiseReductionMode`` - Default to ``NoiseReductionModeHighQuality``
``ScalerCrop`` - The scaler crop is sensor and mode dependent, and this in turn depends on the application using the device.  Typically, it would be ``(0, 0)/4608x2592`` for the Camera v3.
``AwbEnable`` - Default to ``true``
``AwbMode`` - Default to ``AwbAuto``
``FrameDurationLimits`` - Default to ``[1/33ms, 1/33ms]'' (30fps)
``ExposureTime`` and ``AnalogueGain`` - No default, only set when a user control is passed in with an explicit value.  Otherwise AE controls the exposure time.
``AeExposureMode`` - Default to ``ExposureNormal``
``AeEanbled`` - Default to ``true``
``AeConstraintMode`` - Default set in the config file to ``ConstraintNormal``
``AeMeteringMode`` - Default set in the config file to ``MeteringCentreWeighted``
``ColourGains `` - No default, only set when a user control is passed in with an explicit value.  Otherwise AWB controls the colour gains.

@njhollinghurst can you look at the focus/lens ones (wrt libcamera-still) as I'll probably get them wrong.

njhollinghurst, 2023-08-14:
- `AfMode`: The libcamera API default is `AfModeManual`. But libcamera-still defaults `AfModeContinuous` (unless a `--lens-position` or `--autofocus-on-capture` command-line option is used, when it will be `AfModemanual` or `AfModeAuto` respectively).
- `AfTrigger`: This is a write-only control with no default. Initially not triggered (roughly `AfTriggerCancel`). libcamera-still will send `AfTriggerStart` at the end of the viewfinder interval after `--autofocus-on-capture`.
- `AfPause`: This is a write-only control with no default. Initially not paused (roughly `AfPauseResume`).
- `AfMetering`: Default is `AfMeteringAuto`. libcamera-still will set it to `AfMeteringWindows` if the `--autofocus-window` command-line option was used.
- `AfWindows`: The default is an empty list of windows. Note that in the libcamera API, multiple windows are allowed, measured in raw sensor pixels; but in libcamera-still only a single window may be given, in fractions of the sensor size.
- `LensPosition`: The default for the V3 camera with `AfModeManual` and `AfRangeNormal` is 1.0 (nominal distance 1m).

naushir, 2023-08-17:
Any more info needed for this?  If not, I'll close it down shortly.

EricClaeys, 2023-08-17:
Yes, I have additional questions I will post in a day or two.
Thanks for asking.
Eric

EricClaeys, 2023-08-18:
How can I determine the default values for ColourGains, ExposureTime, and AnalogueGain?
Would it be possible for the "Available controls" section to contain the default value in addition to the min and max values?

Do you know why the output of `libcamera-still --list-cameras -v` would be different with, and without `--timeout 1` ?
See AnalogueGain and ExposureTime.
On my imx708 chip I've seen `ExposureTime : [13..0]` which I assume is invalid.

[image]

[image]

naushir, 2023-08-18:
> How can I determine the default values for ColourGains, ExposureTime, and AnalogueGain?

When the AWB and AGC are enabled (this is by default), they adapt the values of these controls based on the image scene and statistics.  As such, they don't really have an inherent default value.  When AWB/AGC are disabled then the value for these controls are based on what the users sets.

>Would it be possible for the "Available controls" section to contain the default value in addition to the min and max values?

It may be possible, I will have to look into this.

> 
> Do you know why the output of `libcamera-still --list-cameras -v` would be different with, and without `--timeout 1` ? See AnalogueGain and ExposureTime. On my imx708 chip I've seen `ExposureTime : [13..0]` which I assume is invalid.
> 
> [image]
> 
> [image]

Hmmm, this does look a bit strange.  On the HQ camera (imx477) the values are identical, as you would expect.  I'll have to try and get hold of an ov5647 module to see what may be going wrong here.  They are a bit hard to find now!

EricClaeys, 2023-08-18:
If I recall correctly, the way to turn off AWB is to specify the red and blue values, so I agree the concept of a default doesn't apply.

Since the values in those screenshots are what the camera can do, shouldn't the values be the same regardless of any command-line arguments?

On my imx708 chip I've seen `ExposureTime : [13..0]`.  Any idea why?

Thanks for considering adding defaults to the "Available controls" section.  It will make determining the defaults much easier since a mapping between control name and the "Options" name won't be needed.

naushir, 2023-08-18:
> On my imx708 chip I've seen `ExposureTime : [13..0]`. Any idea why?

Again, that's quite strange.  My output shows ```ExposureTime : [26..112015443]```.  Does it occasionally show different values?

naushir, 2023-08-18:
> Would it be possible for the "Available controls" section to contain the default value in addition to the min and max values?

Unfortunately this does not work as expected:

```
Available cameras
-----------------
0 : imx477 [4056x3040 12-bit RGGB] (/base/soc/i2c0mux/i2c@1/imx477@1a)
    Modes: 'SRGGB10_CSI2P' : 1332x990 [120.05 fps - (696, 528)/2664x1980 crop]
           'SRGGB12_CSI2P' : 2028x1080 [50.03 fps - (0, 440)/4056x2160 crop]
                             2028x1520 [40.01 fps - (0, 0)/4056x3040 crop]
                             4056x3040 [10.00 fps - (0, 0)/4056x3040 crop]

    Available controls for 4056x3040 SRGGB12_CSI2P mode:
    ----------------------------------------------------
    AeConstraintMode : [0..3] [0]
    AeEnable : [false..true] [<ValueType Error>]
    AeExposureMode : [0..3] [0]
    AeFlickerMode : [0..1] [0]
    AeFlickerPeriod : [100..1000000] [<ValueType Error>]
    AeMeteringMode : [0..3] [0]
    AnalogueGain : [1.000000..22.260870] [<ValueType Error>]
    AwbEnable : [false..true] [<ValueType Error>]
    AwbMode : [0..7] [0]
    Brightness : [-1.000000..1.000000] [0.000000]
    ColourGains : [0.000000..32.000000] [<ValueType Error>]
    Contrast : [0.000000..32.000000] [1.000000]
    ExposureTime : [114..674191602] [<ValueType Error>]
    ExposureValue : [-8.000000..8.000000] [0.000000]
    FrameDurationLimits : [100000..694434742] [<ValueType Error>]
    NoiseReductionMode : [0..4] [0]
    Saturation : [0.000000..32.000000] [1.000000]
    ScalerCrop : [(0, 0)/64x64..(0, 0)/4056x3040] [(2, 0)/4053x3040]
    Sharpness : [0.000000..16.000000] [1.000000]
```

It seems like some control types cannot print default values so you get this ``<ValueType Error>`` displayed which is not pretty.  I may have to defer this for now and speak to the libcamera team about it in more detail.

EricClaeys, 2023-08-18:
It normally displays `ExposureTime : [114..674191602]`, although if another libcamera-still is running it gives `ExposureTime : [0..66666]` (as well as different AnalogueGain and ScalerCrop values).
I realize running it when another libcamera-still is running is unusual it seems odd that the output would be different.  I can make sure the Allsky software doesn't run a second libcamera-still.

Your mockup with the default values would be perfect if the `<ValueType Error>` had the default or said `none` as appropriate.

Interesting that the two boolean values have `<ValueType Error>` as well as ColourGains and ExposureTime which don't have defaults.  Do AeFlickerPeriod and FrameDurationLimits have defaults?

naushir, 2023-08-18:
> I realize running it when another libcamera-still is running is unusual but it seems odd that the output would be different. I can make sure the Allsky software doesn't run a second libcamera-still.

I think this would be advisable.  Although, I'm still not sure why it would make any difference.

> Do AeFlickerPeriod and FrameDurationLimits have defaults?

AeFlickerPeriod  has no intrinsic default, like gain/exposure the value only gets consumed if you set the appropriate control.  FrameDurationLimits is set to 1/33ms by default.

naushir, 2023-08-24:
Any further questions related to this?  If not I'll close it shortly.

EricClaeys, 2023-08-24:
Will you be speaking to the libcamera team about the `<ValueType Error>`?  If so, should I open a new Issue for it?
If not, then please close this Issue.

naushir, 2023-08-25:
Yes I'll chat with them and discuss this.  No need to open a new issue, hopefully it should "just work" after they make some changes since the << operator for ControlInfo will take care if the new output.

closed 2023-08-25 by naushir