raspberrypi/rpicam-apps #549: [BUG]  bullseye lite : libcamera-vid : ERROR V4L2 : Unable to set controls:Remote I/O error

UmonDaisuke, 2023-08-06:
Hello,

I can't acquire video with 2 new cameras v3 on 2 different raspberry Pis but it works for camera v2 and raspivid (with associated legacy settings in raspi-config.). V4L2 i2c controls seems to be the culprit. Could you help me? 

Here is my process.

After flashing, updating, upgrading bullseye 32 bits lite on 2 sd cards for:
- 1 Pi 4 B with v3 camera 
- Pi zero v2 with another pi v3 camera,  

I detect cameras with `libcamera-vid --list-cameras` 
Available cameras
-----------------
0 : imx708_wide_noir [4608x2592] (/base/soc/i2c0mux/i2c@1/imx708@1a)
    Modes: 'SRGGB10_CSI2P' : 1536x864 [120.13 fps - (768, 432)/3072x1728 crop]
                             2304x1296 [56.03 fps - (0, 0)/4608x2592 crop]
                             4608x2592 [14.35 fps - (0, 0)/4608x2592 crop]

But I cant acquire any video with `libcamera-vid -o test.h264 -t 500 -n`

**Bug report**
pi@dev0:~ $ libcamera-vid -o test.h264 -t 500 -n
[0:32:46.873435656] [6460]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f
[0:32:46.991776953] [6461]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media4 and ISP device /dev/media1
[0:32:46.991903711] [6461]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[0:32:46.992950090] [6460]  INFO Camera camera.cpp:1033 configuring streams: (0) 640x480-YUV420
[0:32:46.993485899] [6461]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected unicam format: 1536x864-pBAA
[0:32:46.994860182] [6461] ERROR V4L2 v4l2_device.cpp:353 'dw9807 10-000c': Unable to set controls: Remote I/O error
[0:32:47.155226516] [6461] ERROR V4L2 v4l2_device.cpp:353 'dw9807 10-000c': Unable to set controls: Remote I/O error
....
[0:32:47.654925441] [6461] ERROR V4L2 v4l2_device.cpp:353 'dw9807 10-000c': Unable to set controls: Remote I/O error
Halting: reached timeout of 500 milliseconds.

UmonDaisuke, 2023-08-08:
Hello,
after few analysis, in spite of long list of error messages, video is produced but it seems that autofocus control is disabled. Can it be linked to this remote I/O error?
Thanks for your support

naushir, 2023-08-14:
This looks to be an issue with communicating with the module actuator over I2C.  I would normally say that the cable may need re-seating, but I2C for the sensor access seems to be working fine.  Is this a problem that has recently started occurring with the camera board?

UmonDaisuke, 2023-08-14:
Hello,
I would say that this problem appeared 2 to 3 weeks ago. Meanwhile, I add the same behavior with a new couple pi zéro v2 / camera module v3. I double Checked wiring. Can I collect some logs to go further into investigation ?
Many thanks for support.
Michael

naushir, 2023-08-15:
So you are getting this error with multiple Camera v3 boards?  I would suggest starting from scratch again with a fresh Raspberry Pi OS sd card image to see if you still get the error.

naushir, 2023-08-21:
If there are no further updates, I'll close this one down for now.

naushir, 2023-08-22:
Closing now, please do reopen if you continue having issues with a fresh Raspberry Pi OS image.

closed 2023-08-22 by naushir

UmonDaisuke, 2023-08-23:
Hello ,

sorry for dealy, i'm just back from holiday. A fresh bullseye installation on a Pi 4b followed by 
```
sudo apt update
sudo apt full-upgrade
```

leads to the same issue. Videofile is recorded but always these v4l2 error messages 

pi@sierra:~ $ libcamera-vid -o test.h264 -n
[0:21:44.909356211] [1292]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f
[0:21:45.049081044] [1293]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media4 and ISP device /dev/media0
[0:21:45.049163414] [1293]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[0:21:45.050133794] [1292]  INFO Camera camera.cpp:1033 configuring streams: (0) 640x480-YUV420
[0:21:45.050779547] [1293]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected unicam format: 1536x864-pBAA
[0:21:45.052291034] [1293] ERROR V4L2 v4l2_device.cpp:353 'dw9807 10-000c': Unable to set controls: Remote I/O error
[0:21:45.212678739] [1293] ERROR V4L2 v4l2_device.cpp:353 'dw9807 10-000c': Unable to set controls: Remote I/O error
[0:21:45.245857870] [1293] ERROR V4L2 v4l2_device.cpp:353 'dw9807 10-000c': Unable to set controls: Remote I/O error
[0:21:45.279398147] [1293] ERROR V4L2 v4l2_device.cpp:353 'dw9807 10-000c': Unable to set controls: Remote I/O error
[0:21:45.312571704] [1293] ERROR V4L2 v4l2_device.cpp:353 'dw9807 10-000c': Unable to set controls: Remote I/O error
[0:21:45.345834945] [1293] ERROR V4L2 v4l2_device.cpp:353 'dw9807 10-000c': Unable to set controls: Remote I/O error
......

naushir, 2023-08-23:
@UmonDaisuke thanks for the update.  Can I ask when and where did you purchase this module from?  I presume it was working fine at some point in time?

UmonDaisuke, 2023-08-23:
@naushir  I bought original camera modules V3 at Digikey. Yes It has worked for a while. I unplugged and replugged the sensor (like I did for years with v2) for adding and extension ribbon and it was still working. Please notice that with or without this ribbon v4l2 messages are the sames for my 3 cameras v3 (classical, noir, wide noir).   This is realy annoying as autofocus is disabled and I can't set it manually.

UmonDaisuke, 2023-08-23:
@naushir , topics closed. wrong connection. sorry for time spent on this false issue. 
Knd regards