raspberrypi/rpicam-apps #550: [BUG]

medem23, 2023-08-09:
**Describe the bug**
A clear and concise description of what the bug is.

**Bug report**
Please use the ``camera-bug-report`` tool to create a bug report, and upload it here.

The bug report tool uses the following syntax:

```
camera-bug-report -t <timeout in seconds> -o <output file> -c "<libcamera-app command>"
```

For example,

```
camera-bug-report -t 5 -o bug.txt -c "libcamera-still -t 1000 -o test.jpg"
```
will attempt to run libcamera-still and timeout after 5 seconds if the application has stalled. The script will generate a ``bug.txt`` file that captures all the output logs and system state to help us debug the issue.

You can also run without a timeout:

```
camera-bug-report -o bug.txt -c "libcamera-vid -t 0 -o test.264"
```
This will run ``libcamera-vid`` indefinitely until either you press ``Ctrl+C`` or the application terminates, after which the necessary output logs and system state will be captured.

If you cannot run your application through the ``camera-bug-report`` tool, run it without the ``-c`` command line argument **after running the camera application.**  In these cases, please also provide the command line used to run the application, as well as any output generated during the run.

**Additional context**
Add any other context about the problem here.

naushir, 2023-08-14:
This seems to be an I2C error when communicating with the sensor.  Do you by any chance have a touchscreen installed on the Pi?  If so, a recent fix for excessive I2C accesses has been merged (https://github.com/raspberrypi/linux/pull/5545) which may solve your problem.

medem23, 2023-08-14:
Yes I do have a touchscreen installed.

naushir, 2023-08-17:
Did you manage to try the kernel fix and see if it fixes your problem?

naushir, 2023-08-21:
If there are no further updates, I'll close this one down for now.

medem23, 2023-08-21:
I updated the kernels today and it seems to work (will need to run longer test)., Thank you very much for the help.

naushir, 2023-08-22:
Great, thanks for confirming!

closed 2023-08-22 by naushir