raspberrypi/rpicam-apps #551: [BUG] Camera Module 3 not detected on RPi4b 4G Ubuntu 22.04

atticusrussell, 2023-08-13:
**Describe the bug**
Camera module 3 on RPi 4b 4GB Ubuntu 22.04. (Ubuntu 22.04.3 LTS (GNU/Linux 5.15.0-1034-raspi aarch64))
Compiled libcamera and libcamera-apps but get the message "no cameras available". I have confirmed that this exact hardware functions when using Raspbian, where I can run libcamera.

**Bug report**
[report0.txt](https://github.com/raspberrypi/libcamera-apps/files/12329647/report0.txt)

I ran the bug report utility again as root to get dmesg info:
[report1.txt](https://github.com/raspberrypi/libcamera-apps/files/12329652/report1.txt)

**Additional context**

```
$ cat /boot/firmware/config.txt
[all]
kernel=vmlinuz
cmdline=cmdline.txt
initramfs initrd.img followkernel

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Enable the audio output, I2C and SPI interfaces on the GPIO header. As these
# parameters related to the base device-tree they must appear *before* any
# other dtoverlay= specification
dtparam=audio=on
dtparam=i2c_arm=on
dtparam=spi=on

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Comment out the following line if the edges of the desktop appear outside
# the edges of your display
disable_overscan=1

# If you have issues with audio, you may try uncommenting the following line
# which forces the HDMI output into HDMI mode instead of DVI (which doesn't
# support audio output)
#hdmi_drive=2

# Enable the serial pins
enable_uart=1

# Autoload overlays for any recognized cameras that are attached
# to the CSI port. Please note this is for libcamera support, *not* for
# the legacy camera stack
camera_auto_detect=1
# dtoverlay=imx708
# start_x=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Config settings specific to arm64
arm_64bit=1
dtoverlay=dwc2

# Display on fan hat
dtoverlay=gpio-fan,gpiopin=14,temp=60000
gpu_mem=128

# add pi shutdown switch (manual tweak)
dtoverlay=gpio-shutdown,gpio_pin=17,active_low=1,gpio_pull=up,debounce=1000

[... 270 lines cut ...]
start_x=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Config settings specific to arm64
arm_64bit=1
dtoverlay=dwc2

# Display on fan hat
dtoverlay=gpio-fan,gpiopin=14,temp=60000
gpu_mem=128

# add pi shutdown switch (manual tweak)
dtoverlay=gpio-shutdown,gpio_pin=17,active_low=1,gpio_pull=up,debounce=1000

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable the USB2 outputs on the IO board (assuming your CM4 is plugged into
# such a board)
dtoverlay=dwc2,dr_mode=host
```

6by9, 2023-08-13:
Adding start_x=1 is a dead end as Camera Module 3 is not supported by the legacy stack. It'll identify, but will never work.

Raspberry Pi don't control the Ubuntu kernel builds. Are the relevant modules built? ```modinfo imx708``` and ```modinfo dw9807-vcm```. Does imx708.dtbo exist in /boot/overlays (or whatever the Ubuntu equivalent is)?

```sudo vclog -m``` from https://github.com/raspberrypi/utils/tree/master/vclog can read the firmware logs on systems which don't have vcdbg. That should tell you if the firmware has failed to apply the overlay for whatever reason.

atticusrussell, 2023-08-14:
First, thank you @6by9 for your help. TLDR it appears I'm missing those modules. Not sure if I'd have to build a new kernel or something. 

Following are the results of those commands:
```
atticus@rospi:~  
$ modinfo imx708
modinfo: ERROR: Module imx708 not found.
```
```
atticus@rospi:~  
$ modinfo dw9807-vcm
modinfo: ERROR: Module dw9807-vcm not found.
```

```
atticus@rospi:~  
$ ls /boot/firmware/overlays/im*
/boot/firmware/overlays/imx219.dtbo      /boot/firmware/overlays/imx296.dtbo.bak
/boot/firmware/overlays/imx219.dtbo.bak  /boot/firmware/overlays/imx378.dtbo
/boot/firmware/overlays/imx258.dtbo      /boot/firmware/overlays/imx378.dtbo.bak
/boot/firmware/overlays/imx258.dtbo.bak  /boot/firmware/overlays/imx477.dtbo
/boot/firmware/overlays/imx290.dtbo      /boot/firmware/overlays/imx477.dtbo.bak
/boot/firmware/overlays/imx290.dtbo.bak  /boot/firmware/overlays/imx519.dtbo
/boot/firmware/overlays/imx296.dtbo      /boot/firmware/overlays/imx519.dtbo.bak
```

I was only able to get "bus error" from vclog, but I have included my install process in hopes there is some error I made.
```
atticus@rospi:~/github  
$ git clone git@github.com:raspberrypi/utils.git
Cloning into 'utils'...
remote: Enumerating objects: 399, done.
remote: Counting objects: 100% (178/178), done.
remote: Compressing objects: 100% (95/95), done.
remote: Total 399 (delta 113), reused 127 (delta 83), pack-reused 221
Receiving objects: 100% (399/399), 150.38 KiB | 4.30 MiB/s, done.
Resolving deltas: 100% (201/201), done.

atticus@rospi:~/github  
$ cd utils/

atticus@rospi:~/github/utils  (master) 
$ mkdir build

atticus@rospi:~/github/utils  (master) 
$ cd build/

atticus@rospi:~/github/utils/build  (master) 
$ cmake ..
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/atticus/github/utils/build

atticus@rospi:~/github/utils/build  (master) 
$ make
[ 20%] Building C object dtmerge/CMakeFiles/dtmerge.dir/dtmerge.c.o
/home/atticus/github/utils/dtmerge/dtmerge.c:30:10: fatal error: libfdt.h: No such file or directory
   30 | #include <libfdt.h>
      |          ^~~~~~~~~~
compilation terminated.
make[2]: *** [dtmerge/CMakeFiles/dtmerge.dir/build.make:76: dtmerge/CMakeFiles/dtmerge.dir/dtmerge.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:188: dtmerge/CMakeFiles/dtmerge.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

atticus@rospi:~/github/utils/build  (master) 
$ sudo apt install libfdt-dev

atticus@rospi:~/github/utils/build  (master) 
$ make

atticus@rospi:~/github/utils/build  (master) 
$ sudo ./vclog/vclog -m
Bus error
```

atticusrussell, 2023-08-16:
I successfully recompiled the linux kernel and enabled imx708.
```
atticus@rospi:~  
$ modinfo imx708
filename:       /lib/modules/5.15.0-1035-raspi/kernel/drivers/media/i2c/imx708.ko
license:        GPL v2
description:    Sony IMX708 sensor driver
author:         David Plowman <david.plowman@raspberrypi.com>
srcversion:     51A9A1F441F97D6CA01D94F
alias:          of:N*T*Csony,imx708C*
alias:          of:N*T*Csony,imx708
depends:        videodev,v4l2-fwnode,mc,v4l2-async
intree:         Y
name:           imx708
vermagic:       5.15.0-1035-raspi SMP preempt mod_unload modversions aarch64
sig_id:         PKCS#7
signer:         Build time autogenerated kernel key
sig_key:        6E:9D:24:04:B5:FA:5E:1E:32:88:94:0F:6C:17:E8:79:D9:3F:13:00
sig_hashalgo:   sha512
signature:      27:62:F1:95:1C:D5:09:0F:A0:CF:E4:E5:81:BB:B9:9A:73:9B:2D:46:
		9A:59:DF:FB:67:AF:61:BF:CC:78:43:6A:75:D6:6A:90:A6:98:03:4D:
		40:11:18:B3:BE:95:BE:7E:2C:12:7F:99:59:96:6E:BD:28:37:2C:0F:
		CC:76:CC:48:90:E6:5B:B5:35:85:E5:E4:5F:2C:12:0D:D2:A6:37:B5:
		66:EC:5A:6D:A8:12:8A:98:38:9C:E7:B6:03:47:15:B7:FA:0C:2F:96:
		7A:2B:8B:7E:3B:1C:8F:7F:61:28:23:7E:94:17:A0:33:7D:BA:53:6F:
		16:0F:DB:9D:E0:C9:01:DB:B8:23:49:FE:31:A6:89:1E:12:63:19:AD:
		47:78:7B:4D:1A:DE:EE:46:8A:A6:B7:77:DB:46:68:25:87:3D:B8:E6:
		E3:41:3F:AF:57:9A:84:BC:43:95:6B:E2:03:8F:10:62:9D:C0:FF:61:
		09:E5:95:29:C2:AA:B8:E9:4D:95:88:85:CB:C2:C1:8B:E0:67:B9:64:
		70:8C:02:F7:8D:97:C1:CD:32:80:3D:BA:30:61:97:32:ED:D7:0D:38:
		53:98:D9:6C:C6:0A:67:88:93:67:EE:E8:67:27:3A:08:68:24:C9:38:
		29:81:72:84:E5:2D:D1:6A:32:5E:95:63:87:C4:65:74:92:37:D8:84:
		0D:D4:69:94:56:59:0F:50:EB:AC:B5:3E:AA:88:1B:02:20:BF:C2:85:
		05:9D:CF:2D:38:8E:DC:FF:E4:5C:D3:48:56:F8:FE:67:42:6C:3F:25:
		7E:52:14:CA:31:5D:6C:CA:A9:08:A6:FF:C3:6D:8F:92:E3:D1:91:B5:
		A8:0E:ED:AC:36:93:02:6A:81:D6:0F:80:F4:D4:12:1B:69:39:9E:8B:
		FE:08:BB:96:B2:45:B6:ED:49:26:35:38:9C:89:E1:A3:5B:F0:1D:B0:
		B1:E0:42:FB:63:0E:8E:DD:D5:D1:36:EE:A9:CD:3E:5E:93:74:03:C5:
		6A:80:ED:E5:E3:56:A5:BA:09:D5:D8:8F:D4:50:A4:EB:B5:70:2C:1C:
		CB:99:9A:EA:5C:BB:8A:4F:51:9E:C3:0B:52:13:29:8B:86:79:EB:B7:
		06:87:DE:05:21:6C:57:C6:FD:45:EE:44:70:2E:00:CD:0B:EB:19:2B:
		F0:34:E4:42:88:BE:75:EE:35:34:F8:89:7E:C1:80:3C:55:5F:20:A8:
		3B:A3:E4:00:10:76:DE:8E:61:6D:F6:E7:67:A4:BE:5A:03:73:EE:AD:
		D6:EC:D5:71:EF:4E:BE:70:16:32:C9:78:2A:35:02:C9:14:47:61:0A:
		34:3E:6D:A0:4A:1D:35:6D:C9:6E:99:F9
```

however I forgot to make sure dw9807-vcm was enabled
```
atticus@rospi:~  
$ modinfo dw9807-vcm
modinfo: ERROR: Module dw9807-vcm not found.

```

and imx708.dtbo still doesn't exist as an overlay, so I'm gonna need to work on that.

atticusrussell, 2023-08-16:
@6by9 I recompiled the kernel again and this time added the imx708 dtbo and the dw9807-vcm, but am running into the same issue. any ideas?

```
atticus@rospi:~  
$ modinfo dw9807-vcm
filename:       /lib/modules/5.15.0-1035-raspi/kernel/drivers/media/i2c/dw9807-vcm.ko
license:        GPL v2
description:    DW9807 VCM driver
author:         Chiang, Alan
srcversion:     E09A567F71350C88CB79109
alias:          of:N*T*Cdongwoon,dw9807-vcmC*
alias:          of:N*T*Cdongwoon,dw9807-vcm
depends:        v4l2-async,mc,videodev
intree:         Y
name:           dw9807_vcm
vermagic:       5.15.0-1035-raspi SMP preempt mod_unload modversions aarch64
sig_id:         PKCS#7
signer:         Build time autogenerated kernel key
sig_key:        36:2D:70:E4:AC:D8:C0:CA:1C:4A:D6:84:E1:AD:C7:FD:00:F3:06:F9
sig_hashalgo:   sha512
signature:      AF:19:A3:93:9F:53:6D:7B:EC:CE:93:EF:F5:6C:4A:B6:85:B0:C0:23:
		53:27:77:69:1A:2B:2D:88:51:DD:A9:2A:AA:0B:28:AC:26:32:7B:53:
		3B:92:9D:18:9D:9B:16:21:82:98:CE:E9:13:D6:B0:02:49:FA:DE:7F:
		72:D1:36:2B:FF:87:AD:F9:52:2F:EA:5A:CD:43:DE:09:6C:DC:0F:29:
		84:56:74:A9:02:3C:B7:11:BA:7A:1C:49:63:54:6B:E9:88:9A:E3:33:
		D3:4A:F1:C6:7C:4C:01:83:F9:C0:46:A1:61:9E:C8:1C:7C:BE:E2:AD:
		8F:AC:CC:DA:D5:37:CD:95:04:6B:E0:F4:2B:3D:EB:14:8A:07:89:B3:
		20:D8:AF:7A:A4:B3:5E:15:B8:EE:44:4D:E7:F1:A9:7D:1C:9E:08:99:
		42:53:64:C5:E2:F9:C7:77:FC:37:50:9F:B6:FE:4B:25:0D:AD:2A:04:
		99:ED:B2:AE:A6:C1:49:B7:6F:D3:11:30:08:27:5D:C9:01:B8:D6:A1:
		53:4C:44:38:EF:0E:A5:A4:CB:2C:C8:70:D3:90:A9:E0:9D:EC:76:C2:
		39:93:1F:B6:A3:E7:06:C0:BF:3E:75:F9:88:8B:CB:99:2A:7B:8D:74:
		F1:B3:A4:95:7A:FC:FB:7E:C9:0F:F8:2A:4A:44:B2:90:70:0E:8D:FC:
		17:DD:F9:73:32:08:E9:34:D5:30:02:00:4A:88:43:00:67:C1:46:28:
		1E:8D:3D:E3:29:69:2D:6B:FA:3F:93:3F:A7:7A:D2:66:70:D0:4A:85:
		BB:D3:1A:B8:FF:D7:FA:4C:23:EA:0A:FA:55:73:DE:17:68:11:4E:B1:
		0D:E8:95:87:54:34:C5:0A:25:50:44:79:ED:7F:B1:44:E5:88:F4:BD:
		B5:79:DE:68:73:69:EC:8F:83:A6:FE:00:38:5C:FD:5A:A3:8A:83:F0:
		48:54:1F:94:53:89:66:9C:BC:CE:C4:F4:81:98:4A:6A:29:84:43:5A:
		06:E9:CA:14:E2:71:C0:68:75:F8:FF:0C:33:34:78:FC:E3:CB:7A:1F:
		C6:22:D6:A9:23:3D:7A:C4:CC:66:59:CC:49:E7:DA:0D:D5:B7:25:D5:
		76:BB:CC:A7:66:36:0E:13:19:1C:EA:C7:8E:4F:5E:70:0F:E5:F5:81:
		29:4E:83:49:31:FB:69:C2:0F:60:FB:B1:B7:6A:EC:A0:28:7A:D4:46:
		A1:FC:E3:13:38:EF:D0:0A:C3:47:39:0B:12:B7:A7:FB:02:8A:82:49:
		D1:D7:BD:B4:06:9E:1A:FE:0C:4C:C6:78:63:7E:DE:35:6C:A8:0F:FD:
		4C:0B:29:E0:C3:69:3E:15:95:56:D2:42
atticus@rospi:~  
$ ls /boot/firmware/overlays/imx*
/boot/firmware/overlays/imx219.dtbo      /boot/firmware/overlays/imx378.dtbo
/boot/firmware/overlays/imx219.dtbo.bak  /boot/firmware/overlays/imx378.dtbo.bak
/boot/firmware/overlays/imx258.dtbo      /boot/firmware/overlays/imx477.dtbo
/boot/firmware/overlays/imx258.dtbo.bak  /boot/firmware/overlays/imx477.dtbo.bak
/boot/firmware/overlays/imx290.dtbo      /boot/firmware/overlays/imx519.dtbo
/boot/firmware/overlays/imx290.dtbo.bak  /boot/firmware/overlays/imx519.dtbo.bak
/boot/firmware/overlays/imx296.dtbo      /boot/firmware/overlays/imx708.dtbo
/boot/firmware/overlays/imx296.dtbo.bak
atticus@rospi:~  
$ modinfo imx708
filename:       /lib/modules/5.15.0-1035-raspi/kernel/drivers/media/i2c/imx708.ko
license:        GPL v2
description:    Sony IMX708 sensor driver
author:         David Plowman <david.plowman@raspberrypi.com>
srcversion:     51A9A1F441F97D6CA01D94F
alias:          of:N*T*Csony,imx708C*
alias:          of:N*T*Csony,imx708
depends:        videodev,v4l2-fwnode,mc,v4l2-async
intree:         Y
name:           imx708
vermagic:       5.15.0-1035-raspi SMP preempt mod_unload modversions aarch64
sig_id:         PKCS#7
signer:         Build time autogenerated kernel key
sig_key:        36:2D:70:E4:AC:D8:C0:CA:1C:4A:D6:84:E1:AD:C7:FD:00:F3:06:F9
sig_hashalgo:   sha512
signature:      6E:19:23:FD:BD:1A:2A:AC:83:08:C2:3E:24:4E:15:58:9D:56:D0:55:
		89:F9:E3:17:1D:9E:C4:95:2E:37:DF:01:64:A3:86:DF:CE:27:93:BC:
		72:CD:9E:42:23:3F:66:B6:8E:BF:5F:31:FC:A4:BA:F8:38:9D:AA:B6:
		D6:94:23:2D:19:B3:47:54:83:4A:6F:AB:5F:44:7B:40:F8:C8:AC:23:
		83:C3:C9:44:12:A0:7F:E4:A0:99:24:16:42:40:6F:65:A5:E0:9A:4B:
		94:E4:17:1D:3F:C3:B7:64:A6:43:D0:90:69:94:E6:D5:D0:A4:68:3A:
		40:C5:5B:89:0E:9C:48:FB:5F:00:AF:FB:B1:4A:DA:8D:BD:BC:3E:6A:
		CF:A3:F0:42:D9:20:C7:26:61:CA:91:DC:1D:52:26:2C:27:75:80:BC:
		5E:3C:B2:AE:3E:AC:BB:07:FA:91:BA:9E:0E:36:E3:51:C2:F1:87:07:
		F5:9A:52:79:4A:A5:AD:2D:96:7E:BB:CE:3F:9D:EB:F6:14:67:F3:60:
		41:C4:A9:D3:FE:A4:81:40:EA:B2:47:9A:8B:70:80:75:60:3B:8C:2D:
		DC:40:87:6B:D2:98:1D:00:55:BE:B9:D2:89:0B:46:D1:FA:D2:E0:32:
		AD:A2:AF:03:7C:51:6E:7B:BF:B3:9C:BC:30:E0:2D:F3:E1:D5:CB:E1:
		4F:AF:AC:B2:D5:BC:82:29:2C:8B:8A:BC:3A:92:53:DA:A4:9D:2D:23:
		C9:EF:00:B0:49:52:F5:A0:AD:44:97:73:3E:62:C6:F8:38:C8:4F:C5:
		2D:F3:0F:C3:D2:B6:6D:DA:9C:10:A4:62:BE:F5:E6:3F:E6:5F:60:96:
		D6:70:16:6F:94:F7:80:94:A4:B6:DE:17:E2:B5:BA:41:27:93:5B:59:
		37:09:00:EA:DC:26:21:F4:CC:0E:03:24:FE:43:50:59:91:1A:89:44:
		95:6A:5D:90:12:61:94:BF:E0:CC:AA:B0:C8:69:88:00:A8:15:62:6F:
		04:80:91:BF:A5:13:73:FF:97:10:14:11:1A:1A:24:73:C0:AD:2F:E7:
		CB:A2:D7:D0:2C:76:F4:09:3D:D5:70:AD:48:22:BD:75:8E:43:B0:19:
		77:F1:F1:53:10:73:08:46:37:B1:B0:91:9C:46:C9:AD:1F:65:54:72:
		4F:EE:DD:05:E3:5C:E4:84:D4:4F:3F:B3:BE:51:3C:15:42:66:B5:CF:
		D7:E0:4D:10:A9:95:91:9D:5A:96:B0:81:43:72:28:C4:07:4D:99:50:
		38:0E:DD:3E:8F:53:9C:7D:40:07:D3:05:5E:55:73:B7:1A:E7:EF:74:
		52:D6:FE:78:B9:1F:62:77:47:95:74:28
atticus@rospi:~  
$ libcamera-hello 
Preview window unavailable
[0:01:29.796286333] [1656]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+19-960d0c1e
ERROR: *** no cameras available ***
```

atticusrussell, 2023-08-16:
A bit of progress!! With the modified/recompiled kernel I manually specified the camera in `/boot/firmware/config.txt`
and got this error:
```
atticus@rospi:~  
$ libcamera-hello 
Preview window unavailable
[0:01:17.528089267] [1548]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+19-960d0c1e
[0:01:17.568858211] [1549] ERROR V4L2 v4l2_device.cpp:93 'imx708': Failed to open V4L2 device '': No such file or directory
[0:01:17.568991952] [1549] ERROR RPI vc4.cpp:209 Failed to register camera imx708: -22
ERROR: *** no cameras available ***
```

```
atticus@rospi:~  
$ cat /boot/firmware/config.txt
[all]
kernel=vmlinuz
cmdline=cmdline.txt
initramfs initrd.img followkernel

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Enable the audio output, I2C and SPI interfaces on the GPIO header. As these
# parameters related to the base device-tree they must appear *before* any
# other dtoverlay= specification
dtparam=audio=on
dtparam=i2c_arm=on
dtparam=spi=on

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Comment out the following line if the edges of the desktop appear outside
# the edges of your display
disable_overscan=1

# If you have issues with audio, you may try uncommenting the following line
# which forces the HDMI output into HDMI mode instead of DVI (which doesn't
# support audio output)
#hdmi_drive=2

# Enable the serial pins
enable_uart=1

# Autoload overlays for any recognized cameras that are attached
# to the CSI port. Please note this is for libcamera support, *not* for
# the legacy camera stack
camera_auto_detect=0
dtoverlay=imx708

l Automatically load overlays for detected DSI displays
display_auto_detect=1

# Config settings specific to arm64
arm_64bit=1
dtoverlay=dwc2

# Display on fan hat
dtoverlay=gpio-fan,gpiopin=14,temp=60000
gpu_mem=128

# add pi shutdown switch (manual tweak)
dtoverlay=gpio-shutdown,gpio_pin=17,active_low=1,gpio_pull=up,debounce=1000

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable the USB2 outputs on the IO board (assuming your CM4 is plugged into
# such a board)
dtoverlay=dwc2,dr_mode=host
```

not sure if maybe something is still lacking in the kernel or if it might be a simpler config error here.

SachetK, 2023-08-16:
`camera_auto_detect` should be set to 1 to enable libcamera support (I believe).

naushir, 2023-08-22:
That suggests the sensor probed correctly, but the device driver could not be opened for some reason.

@atticusrussell do you see any error messages in ``dmesg`` after boot related to the imx708?
Also does the following command successfully give an output:
```
v4l2-ctl -d /dev/v4l-subdev0 --list-subdev-mbus-codes
```
?

Also, could you run ``camera-bug-report`` and attach the output again after your changes to get probe working?

Another thing to try is to see if another camera works in your environment.  Do you have another Raspberry Pi camera that you could use?

6by9, 2023-08-22:
I'd guess that the media controller node has been created, but due to some dependent device not having probed the /dev/videoN node hasn't been registered.
I can recreate that by blacklisting dw9807_vcm (I happen to be on a 6.5 kernel here, but that shouldn't matter):
```
pi@raspberrypi:~ $ media-ctl -p -d 4
Media controller API version 6.5.0

Media device information
------------------------
driver          unicam
model           unicam
serial          
bus info        platform:fe801000.csi
hw revision     0x0
driver version  6.5.0

Device topology
- entity 1: imx708 (2 pads, 0 link)
            type V4L2 subdev subtype Sensor flags 0
	pad0: Source
	pad1: Source
```
```
pi@raspberrypi:~ $  sudo cat /sys/kernel/debug/v4l2-async/pending_async_subdevices 
imx708:
 [fwnode] dev=10-000c, node=/soc/i2c0mux/i2c@1/dw9817@c
unicam fe801000.csi:```
ie the imx708 node is waiting for dw9817 to enumerate. 
/dev/video0 doesn't exist until all dependencies are met, which calls [unicam_async_complete](https://github.com/raspberrypi/linux/blob/rpi-6.1.y/drivers/media/platform/bcm2835/bcm2835-unicam.c#L3137) and that registers the /dev/videoN nodes.

atticusrussell, 2023-08-23:
@naushir the following is the output of bug-report (with the later mentioned USB HD web camera disconnected for simplicity): 
[bugreport3.txt](https://github.com/raspberrypi/libcamera-apps/files/12414049/bugreport3.txt)

```
atticus@rospi:~  
$ v4l2-ctl -d /dev/v4l-subdev0 --list-subdev-mbus-codes
Cannot open device /dev/v4l-subdev0, exiting.
```

The output of searching dmesg for the sensor only results in the name of the kernel that I modified (not predicting that it would show up as a false positive in dmesg when picking the name)
```
atticus@rospi:~  
$ dmesg | grep imx708
[    0.000000] Linux version 5.15.0-1035-raspi (root@dev-5587) (aarch64-linux-gnu-gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #38+imx708dw9807~20230815 SMP PREEMPT Tue Aug 15 23:04:05 EDT 20 (Ubuntu 5.15.0-1035.38+imx708dw9807~20230815-raspi 5.15.111)

```

I found a generic USB "HD Webcam" without branding on it that works on my pi with libcamera. Here is the output of dmesg (nothing grepped) with it connected, as well as the non-working RPi Camera Module 3 connected 
```
atticus@rospi:~  
$ dmesg 
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.15.0-1035-raspi (root@dev-5587) (aarch64-linux-gnu-gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #38+imx708dw9807~20230815 SMP PREEMPT Tue Aug 15 23:04:05 EDT 20 (Ubuntu 5.15.0-1035.38+imx708dw9807~20230815-raspi 5.15.111)
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000016000000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000037ffffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] On node 0, zone DMA32: 16384 pages in unavailable ranges
[    0.000000] percpu: Embedded 29 pages/cpu s80472 r8192 d30120 u118784
[    0.000000] pcpu-alloc: s80472 r8192 d30120 u118784 alloc=29*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v3a
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 1742098
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 983808
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=DC:A6:32:19:5D:BE vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 dwc_otg.lpm_enable=0 console=tty1 root=LABEL=writable rootfstype=ext4 rootwait fixrtc quiet splash
[    0.000000] Unknown kernel command line parameters "fixrtc splash", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x0000000034000000-0x0000000038000000] (64MB)
[    0.000000] Memory: 3415444K/3997696K available (14272K kernel code, 2666K rwdata, 5188K rodata, 5760K init, 878K bss, 254572K reserved, 327680K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 44164 entries in 173 pages
[    0.000000] ftrace: allocated 173 pages with 5 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[... 542 lines cut ...]
[   23.822525] Bluetooth: HCI UART protocol ATH3K registered
[   23.822558] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   23.822711] Bluetooth: HCI UART protocol Intel registered
[   23.822823] Bluetooth: HCI UART protocol Broadcom registered
[   23.822848] Bluetooth: HCI UART protocol QCA registered
[   23.822853] Bluetooth: HCI UART protocol AG6XX registered
[   23.822877] Bluetooth: HCI UART protocol Marvell registered
[   24.312048] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   24.312071] Bluetooth: BNEP filters: protocol multicast
[... cut ...]
[  743.699806] audit: type=1400 audit(1692111661.603:51): apparmor="DENIED" operation="open" profile="/usr/sbin/ntpd" name="/etc/ssl/openssl.cnf" pid=5180 comm="ntpd" requested_mask="r" denied_mask="r" fsuid=0 ouid=0
[  745.510930] audit: type=1400 audit(1692111663.415:52): apparmor="DENIED" operation="open" profile="/usr/sbin/ntpd" name="/snap/bin/" pid=5212 comm="ntpd" requested_mask="r" denied_mask="r" fsuid=0 ouid=0
[  745.521632] audit: type=1400 audit(1692111663.427:53): apparmor="DENIED" operation="open" profile="/usr/sbin/ntpd" name="/etc/ssl/openssl.cnf" pid=5218 comm="ntpd" requested_mask="r" denied_mask="r" fsuid=0 ouid=0

```

naushir, 2023-08-23:
As @6by9 mentioned, this looks like the sensor probe has not fully completed (likely due waiting for dw9817 to enumerate), and the media device is not fully registered.

6by9, 2023-08-23:
5.15 is pretty old in our book.
https://github.com/torvalds/linux/commit/517fd2b6a058bc4f23735ab9a9e6b0d5c56876d1 to add v4l2-async/pending_async_subdevices was added in 5.12, so that implies that nothing is registering for v4l2-async.

```lsmod``` should include at least imx708, bcm2835_unicam, videobuf2_dma_contig, videobuf2_memops, videobuf2_v4l2, videodev, videobuf2_common, v4l2_async, dw9807_vcm.
Not having a unicam Media Controller node registered implies that bcm2835_unicam isn't loading, let alone the rest.

Sorry, but this is getting to the stage of being Ubuntu specific and therefore probably better to be asking Canonical about. Or switch to the rpi-6.1.y kernel and our standard defconfig to ensure you've built everything necessary.

atticusrussell, 2023-08-23:
Thanks for all the help. I'm pretty new to messing with the kernel, so have a few questions about how to proceed with the update. 

AFAIK the kernel goes under any distro-specific customization, so in theory could I simply compile 6.1y from [raspberrypi/linux](https://github.com/raspberrypi/linux) and apply the .deb patches to Ubuntu?

Thus far I have been copying changes from raspberrypi/linux over to the [Ubuntu provided linux-raspi kernel](https://git.launchpad.net/ubuntu/+source/linux-raspi/) on the branch [applied/ubuntu/jammy-updates 5.15.0-1035.38 (patches applied)](https://git.launchpad.net/ubuntu/+source/linux-raspi/commit/?h=applied/ubuntu/jammy-updates).

Can I run a kernel newer than 5.15 with ubuntu 22.04 (jammy)? I notice that there's a 6.2.X kernel in the ubuntu rpi-kernel repo tagged with mantic (23.10) and lunar (23.04).  If so, does it matter whether I get the newer kernel from raspberrypi/linux or rpi-kernel, and would either choice be better?

Thanks again for all your help.

atticusrussell, 2023-08-24:
@6by9 I managed to update my kernel to  `6.2.0-1004-raspi` using a package from Ubuntu Lunar, as detailed in [this askubuntu post I made](https://askubuntu.com/a/1483547/1721050), however my issues with libcamera and the Raspberry Pi Camera Module 3 persist. 

I think I have the correct drivers now. I have now reproduced the issue you predicted with imx708 as a pending_async_subdevice.

```
atticus@rospi:~  
$ uname -a
Linux rospi 6.2.0-1004-raspi #5-Ubuntu SMP PREEMPT Mon Apr  3 11:15:14 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux

atticus@rospi:~  
$ sudo cat /sys/kernel/debug/v4l2-async/pending_async_subdevices
imx708:
 [fwnode] dev=10-000c, node=/soc/i2c0mux/i2c@1/dw9817@c
unicam fe801000.csi:

atticus@rospi:~  
$ libcamera-hello 
Preview window unavailable
[0:13:33.922153143] [5185]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+19-960d0c1e
[0:13:33.961853643] [5186] ERROR V4L2 v4l2_device.cpp:93 'imx708': Failed to open V4L2 device '': No such file or directory
[0:13:33.961984865] [5186] ERROR RPI vc4.cpp:209 Failed to register camera imx708: -22
ERROR: *** no cameras available ***
```

Not sure what my next step should be. How do I get dw9817 to enumerate, like you mentioned. 

Also I have manually specified the sensor:
```
atticus@rospi:~  
$ cat /boot/firmware/config.txt 
[all]
kernel=vmlinuz
cmdline=cmdline.txt
initramfs initrd.img followkernel

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Enable the audio output, I2C and SPI interfaces on the GPIO header. As these
# parameters related to the base device-tree they must appear *before* any
# other dtoverlay= specification
dtparam=audio=on
dtparam=i2c_arm=on
dtparam=spi=on

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Comment out the following line if the edges of the desktop appear outside
# the edges of your display
disable_overscan=1

# If you have issues with audio, you may try uncommenting the following line
# which forces the HDMI output into HDMI mode instead of DVI (which doesn't
# support audio output)
#hdmi_drive=2

# Enable the serial pins
enable_uart=1

# Autoload overlays for any recognized cameras that are attached
# to the CSI port. Please note this is for libcamera support, *not* for
# the legacy camera stack
camera_auto_detect=0
dtoverlay=imx708

l Automatically load overlays for detected DSI displays
display_auto_detect=1

# Config settings specific to arm64
arm_64bit=1
dtoverlay=dwc2

# Display on fan hat
dtoverlay=gpio-fan,gpiopin=14,temp=60000
gpu_mem=128

# add pi shutdown switch (manual tweak)
dtoverlay=gpio-shutdown,gpio_pin=17,active_low=1,gpio_pull=up,debounce=1000

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable the USB2 outputs on the IO board (assuming your CM4 is plugged into
# such a board)
dtoverlay=dwc2,dr_mode=host
```

atticusrussell, 2023-08-25:
looks like I may not have the necessary driver in this kernel but I'll look into it further:
```
atticus@rospi:~  
$ sudo modprobe dw9817
[sudo] password for atticus: 
modprobe: FATAL: Module dw9817 not found in directory /lib/modules/6.2.0-1004-raspi

```

naushir, 2023-08-25:
@atticusrussell any particular reason you are using Ubuntu and not a Raspberry Pi OS image?  In the latter, Camera v3 will "just work" as intended.

6by9, 2023-08-25:
The VCM driver is dw9807_vcm. It supports multiple compatible strings to differentiate between dw9807 and dw9817
```
pi@raspberrypi:~ $ modinfo dw9807-vcm 
filename:       /lib/modules/6.1.32-v8+/kernel/drivers/media/i2c/dw9807-vcm.ko.xz
license:        GPL v2
description:    DW9807 VCM driver
author:         Chiang, Alan
srcversion:     1F6C21AE82A3684591E1B1D
alias:          of:N*T*Cdongwoon,dw9817-vcmC*
alias:          of:N*T*Cdongwoon,dw9817-vcm
alias:          of:N*T*Cdongwoon,dw9807-vcmC*
alias:          of:N*T*Cdongwoon,dw9807-vcm
depends:        v4l2-async,videodev,mc
intree:         Y
name:           dw9807_vcm
vermagic:       6.1.32-v8+ SMP preempt mod_unload modversions aarch64
```

atticusrussell, 2023-08-25:
> @atticusrussell any particular reason you are using Ubuntu and not a Raspberry Pi OS image?  In the latter, Camera v3 will "just work" as intended.

If your question is about why as an OS, I am using Ubuntu in order to have compatibility with ROS as this computer is for my robot.

If you're asking about why I'm not trying to use a Raspberry pi kernel like from the `raspberrypi/linux` repo, I am still open to trying that. The reason I have not yet is that I was able to install the ubuntu 6.2.X kernel without compiling it myself (which is slightly tedious), and that I was slightly apprehensive to mix kernels and userspaces  (although the kernel intended for raspian should be fine to run Ubuntu I think).

atticusrussell, 2023-08-25:
@6by9 
> The VCM driver is dw9807_vcm. It supports multiple compatible strings to differentiate between dw9807 and dw9817
 
Ah as I was suspecting. Thanks for confirming! I'll first try enabling it in menuconfig when recompiling the Ubuntu kernel 6.2.x, and if that fails I'll look for any recompiled `raspberrypi/linux`  6.1y `.deb` files, and failing that I'll try to compile `raspberrypi/linux` and install that kernel.

That order of operations seem sane?

naushir, 2023-09-07:
Any further updates on this issue?  If not, I'll close this shortly.

atticusrussell, 2023-09-07:
I took a bit of a break from trying to solve this issue and focused on just using a USB webcam I have. I still intend to try with a different kernel version, but I might not get around to it for a week or so. If there is a more relevant place for me to file an issue I would say go ahead and close the issue after I re-file it, but otherwise might be good to leave open in case others come across this issue.

naushir, 2023-09-14:
I'll close this one now.  Please do create another issue if you still run into problems when you revisit this.

closed 2023-09-14 by naushir