raspberrypi/rpicam-apps #555: [QUESTION] - unable to request buffers

EricClaeys, 2023-08-16:
One of our Allsky users has an RPi HQ camera that occasionally produces the error below.  Sometimes it produces 4 in a row, at which point our software stops.  Any ideas on how to debug this?

Before calling `libcamera-still` our software kills any currently running `libcamera-still` processes (or at least it's supposed to).

Thanks - Eric

~~~
[111:06:04.333128904] [2815] ERROR V4L2 v4l2_videodevice.cpp:1241 /dev/video14[16:cap]: Unable to request 1 buffers: Cannot allocate memory
[111:06:04.333312828] [2815] ERROR RPI raspberrypi.cpp:1133 Failed to allocate buffers
ERROR: *** failed to start camera ***
~~~

pmccudden, 2023-08-18:
Here's my Pi:
[image]
 and I'm using a Pi HQ camera.  Could it be that my Pi board only has 1GB or RAM?

naushir, 2023-08-22:
That shows a 1GB memory size.  Depending on what else might be running on the system, this might be a bit tight on memory and fail to allocate capture buffers.  Our most recent updates to the libcamera libraries have made some improvements in camera buffer allocations, this may help with your intermittent failures.  To update, you need to run

```
sudo apt update
sudo apt upgrade -y
```

naushir, 2023-09-01:
Any further updates on this issue?

pmccudden, 2023-09-05:
The problem was still there.  I got my hands on an 8GB Pi, however, and the problem seems to be gone with it - apparently it was a memory issue?

EricClaeys, 2023-09-05:
Many other people are using the HQ camera on Pi's with 1/2 GB or 1 GB systems, so I doubt it's the physical memory.
During installation of Allsky did you let it create swap space?

pmccudden, 2023-09-05:
I'm almost certain I did - I certainly did not actively chose NOT to.  I may try a complete reinstallation on the 1 GB Pi and see what happens.  I followed this video: https://www.youtube.com/watch?v=7TGpGz5SeVI

naushir, 2023-09-05:
1GB ought to work (just about).  It might possibly/probably be some background process or task that would be eating away some memory.

Conrad414, 2023-09-14:
My Allsky system is running on a Raspberry Pi 3B+ now with the latest version of Raspbian Bullseye OS with a 128Mb GPU memory, 4Gb swap file, no overlay file system, GL driver enabled and the latest version of Allsky.

The following issue has arisen. When I start Allsky from the Web GUI, it takes four pictures and then hangs with the message "ERROR: maximum number of consecutive errors of 4 reached; capture program stopped. (September 14, 07:21:25 AM)". However, if I close the Web GUI and browser immediately after starting Allsky, (or launch from the CLI) the program functions normally, eventually producing all the outputs - timelapse, startrails etc. Running a "libcamera-still --timelapse" also works with no problems so the camera cable is fine.

I have attached the log and RPi debug files pertaining to the above error message, hopefully in the right manner.

[65:22:36.000215746] [17183] [1;31mERROR [1;37mV4L2 [1;34mv4l2_videodevice.cpp:1241 [0;32m/dev/video14[16:cap]: [0mUnable to request 1 buffers: Cannot allocate memory
ERROR: *** failed to allocate capture buffers ***

[capture_RPi_debug.txt](https://github.com/raspberrypi/libcamera-apps/files/12610486/capture_RPi_debug.txt)
[allsky.log](https://github.com/raspberrypi/libcamera-apps/files/12610505/allsky.log)

EricClaeys, 2023-09-14:
@naushir, please note @Conrad414's issue is identical to @pmccudden's.

naushir, 2023-09-15:
> with a 128Mb GPU memory

How have you set this up?  Unless there is a very specific need, you should leave CMA/GPU allocations set to the defaults provided in our standard OS images.  This could possibly cause the memory issues you are encountering.

Conrad414, 2023-09-15:
I have changed the GPU memory back to 76Mb which I believe is the default and behavior is unchanged. cat /proc/me info reports 262144kB Cma total, 164788 free

naushir, 2023-09-27:
Sorry for the delay in replying to this.  Can you send me the output of ``libcamera-hello --version`` to see what version of libcamera you are running?

Conrad414, 2023-09-28:
Hi Naushir, I sent this output of libcamera-hello yesterday, not sure if it got through.

[libcamer-hello.txt](https://github.com/raspberrypi/libcamera-apps/files/12746947/libcamer-hello.txt)

naushir, 2023-09-28:
Thanks for that.  To be honest, there is very little I can recommend to do right now.  One thing to do would be to try and increase the CMA memory space.  To do this, edit ``/boot/config.txt`` and replace the following line:
```
dtoverlay=vc4-kms-v3d
```
with
```
dtoverlay=vc4-kms-v3d,cma-512
```
and see if that helps.

On the side, we are working with the libcamera team to improve memory usage, but that requires some changes to the libcamera stack.  This is an ongoing development, so I can't give an exact timeframe when it would be ready.

Conrad414, 2023-09-28:
Thanks, will give it a try. Can at least live with the problem as long as I
keep the Allsky preview window closed.

pmccudden, 2023-09-28:
My problem disappeared when I got a Raspberry Pi with more memory, so I'm assuming that was the issue

EricClaeys, 2023-09-28:
@Conrad414,   FYI, @naushir asked for the output of `libcamera-hello --version`, not `libcamera-hello`.

Conrad414, 2023-09-28:
Oh dear! Sorry about that. Here is the output of libcamera-hello --version

libcamera-hello --version
libcamera-apps build: 7e4d3d71867f 17-07-2023 (07:34:42)
libcamera build: v0.0.5+83-bde9b04f

naushir, 2023-09-28:
Thanks, that verifies that you are using a very recent version of the libraries.

naushir, 2023-10-05:
Unfortunately I cannot really think of anything more I can do here about this memory issue.  However, we are working with the libcamera team on optimisations that improve the memory footprint, so things should hopefully improve in the future...

closed 2023-10-05 by naushir

Conrad414, 2023-10-05:
Naushir,

Thanks for your input, I can manage by closing the preview window. Perhaps
it's just too much for a Pi3?

Look forward to hearing if libcamera solves the problem.

Best regards

Conrad

orzaez, 2024-03-26:
I have the same issue, I´m working with a Pi Zero 2W have 512 MB. Thats my config.txt
`# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=6

# Disable compensation for displays with overscan
disable_overscan=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]
dtoverlay=arducam-64mp
[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]
camara auto detect=0`

Pls help!!!1

6by9, 2024-03-26:
`max_framebuffers=6` is a nonsense. The max the firmware supports is 2.

Use of arducam-64mp means you're using Arducam's fork of libcamera, as they've never upstreamed support for that sensor. https://github.com/arducam/libcamera, and support through https://forum.arducam.com/c/cameras-for-raspberry-pi/7

https://forum.arducam.com/t/how-to-use-arducam-64mp-arducam-64mp-faq/2848/2 is recommending you assign 512MB to the CMA heap. That would be all the RAM on your Pi02W (it won't work), and the CMA heap is also used by the display pipeline.
Each 64MPix image is 76MB. You literally can't fit many of those into 512MB.

zoldaten, 2025-03-13:
try this - https://retropie.org.uk/forum/topic/24436/drm_ioctl_mode_create_dumb-failed-cannot-allocate-memory/8