raspberrypi/rpicam-apps #556: libcamera-apps on x86_64

vivien, 2023-08-17:
Due to the integration of libcamera, the code for libcamera-apps is now independent of Raspberry Pi hardware. My objective is to execute the applications on an x86_64 laptop using the built-in front webcam to prove this. I have successfully compiled both libcamera and libcamera-apps. While the 'cam' and 'qcam' applications function as expected, the execution of any binaries from libcamera-apps results in the subsequent errors. Any insights into the potential cause of this issue?

```
$ cam -l
[13:07:08.913420228] [171841]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+19-960d0c1e
Available cameras:
1: Internal front camera (\_SB_.PC00.XHCI.RHUB.HS08-8:1.0-13d3:5271)
2: Internal front camera (\_SB_.PC00.XHCI.RHUB.HS08-8:1.2-13d3:5271)

$ libcamera-hello --version
libcamera-apps build: 92ee0fc0583a-intree 16-08-2023 (16:53:34)
libcamera build: v0.1.0+19-960d0c1e

$ libcamera-hello
Made X/EGL preview window
[13:07:21.478930727] [171850]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+19-960d0c1e
Stream configuration adjusted
[13:07:21.531294418] [171850]  INFO Camera camera.cpp:1033 configuring streams: (0) 640x360-R8
terminate called after throwing an instance of 'std::runtime_error'
  what():  Preview windows only support YUV420
Aborted (core dumped)

$ libcamera-hello --lores-width 400 --lores-height 300
Made X/EGL preview window
[13:07:29.270298355] [171888]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+19-960d0c1e
ERROR: *** failed to generate viewfinder configuration **
```

closed 2023-08-17 by naushir

6by9, 2023-08-17:
The first error is thrown from https://github.com/raspberrypi/libcamera-apps/blob/main/core/libcamera_app.cpp#L998
If you can determine a format that is supported by your pipeline, amending that check, and changing the format requested at https://github.com/raspberrypi/libcamera-apps/blob/main/core/libcamera_app.cpp#L301 and similar from ```libcamera::formats::YUV420``` may get it working.

I suspect that cam and qcam just leave pixel format for the streams at the default as set by the pipeline handler. Some of the code paths may well just work with that, but they haven't been tested.

vivien, 2023-08-17:
Thanks @6by9! How do I inspect the formats supported by my pipeline? Is there an appropriate common line option that I may use to force one? If not, what do we want to implement here, `--pixel-format`?

6by9, 2023-08-17:
As a first test, drop https://github.com/raspberrypi/libcamera-apps/blob/main/core/libcamera_app.cpp#L998
```
		if (item.stream->configuration().pixelFormat != libcamera::formats::YUV420)
			throw std::runtime_error("Preview windows only support YUV420");
```
and remove the 
```
configuration_->at(0).pixelFormat = libcamera::formats::YUV420;
```
at https://github.com/raspberrypi/libcamera-apps/blob/main/core/libcamera_app.cpp#L301
AIUI That will leave the ```configuration_->at(0).pixelFormat``` at the default as set by the pipeline handler. Print it out for your own benefit.
The pipeline may well fail slightly further on, but one thing at a time.

A first glance says that the DRM preview code should just work. Note that you need to be at a basic console for DRM preview to work as DRM/KMS only allows one master at a time.
EGL preview is hard coded to YUV420 at https://github.com/raspberrypi/libcamera-apps/blob/main/preview/egl_preview.cpp#L379. You may be able to lift code from https://github.com/6by9/drm_mmal/blob/x11/drm_mmal.c#L479 to provide the relevant EGL config.
QT preview likewise assumes YUV420.

vivien, 2023-08-29:
I discovered that I have two cameras on my laptop, the first one being some sort of night vision sensor next to the normal one...

```
~/libcamera-apps$ LIBCAMERA_LOG_LEVELS=0 libcamera-hello --list-cameras
Available cameras
-----------------
0 : Integrated RGB Camera: Integrat [640x360] (\_SB_.PC00.XHCI.RHUB.HS08-8:1.2-13d3:5271)
    Modes: 'R8' : 640x360 [nan fps - (0, 0)/0x0 crop]

1 : Integrated RGB Camera: Integrat [1920x1080] (\_SB_.PC00.XHCI.RHUB.HS08-8:1.0-13d3:5271)
    Modes: 'MJPEG' : 320x180 [nan fps - (0, 0)/0x0 crop]
                     320x240 [nan fps - (0, 0)/0x0 crop]
                     352x288 [nan fps - (0, 0)/0x0 crop]
                     424x240 [nan fps - (0, 0)/0x0 crop]
                     640x360 [nan fps - (0, 0)/0x0 crop]
                     640x480 [nan fps - (0, 0)/0x0 crop]
                     848x480 [nan fps - (0, 0)/0x0 crop]
                     960x540 [nan fps - (0, 0)/0x0 crop]
                     1280x720 [nan fps - (0, 0)/0x0 crop]
                     1920x1080 [nan fps - (0, 0)/0x0 crop]
           'YUYV' : 320x180 [nan fps - (0, 0)/0x0 crop]
                    320x240 [nan fps - (0, 0)/0x0 crop]
                    352x288 [nan fps - (0, 0)/0x0 crop]
                    424x240 [nan fps - (0, 0)/0x0 crop]
                    640x360 [nan fps - (0, 0)/0x0 crop]
                    640x480 [nan fps - (0, 0)/0x0 crop]
                    848x480 [nan fps - (0, 0)/0x0 crop]
                    960x540 [nan fps - (0, 0)/0x0 crop]
                    1280x720 [nan fps - (0, 0)/0x0 crop]
                    1920x1080 [nan fps - (0, 0)/0x0 crop]
```

I have libav, drm, egl, opencv and qt set to YES in the build configuration, and compiled libcamera-apps with the following patch:

```diff
~/libcamera-apps$ git diff
diff --git a/core/libcamera_app.cpp b/core/libcamera_app.cpp
index d969a59..9a39654 100644
--- a/core/libcamera_app.cpp
+++ b/core/libcamera_app.cpp
@@ -298,7 +298,9 @@ void LibcameraApp::ConfigureViewfinder()
        }
 
        // Now we get to override any of the default settings from the options_->
+#if 0
        configuration_->at(0).pixelFormat = libcamera::formats::YUV420;
+#endif
        configuration_->at(0).size = size;
        if (options_->viewfinder_buffer_count > 0)
                configuration_->at(0).bufferCount = options_->viewfinder_buffer_count;
@@ -995,8 +997,10 @@ void LibcameraApp::previewThread()
                                preview_cond_var_.wait(lock);
                }
 
+#if 0
                if (item.stream->configuration().pixelFormat != libcamera::formats::YUV420)
                        throw std::runtime_error("Preview windows only support YUV420");
+#endif
 
                StreamInfo info = GetStreamInfo(item.stream);
                FrameBuffer *buffer = item.completed_request->buffers[item.stream];
```

From a basic console, I get no preview:

```
$ LIBCAMERA_LOG_LEVELS=0 libcamera-hello --camera 1 -t 1000
Preview window unavailable
[0:45:53.060978676] [2195] DEBUG IPAModule ipa_module.cpp:334 ipa_ipu3.so: IPA module /usr/local/lib/x86_64-linux-gnu/libcamera/ipa_ipu3.so is signed
[0:45:53.061048973] [2195] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/local/lib/x86_64-linux-gnu/libcamera/ipa_ipu3.so'
[... 374 lines cut ...]
QtPreview: unexpected colour space Rec709/Rec709/Rec601/Limited
[0:49:25.103493313] [4050] DEBUG V4L2 v4l2_videodevice.cpp:1693 /dev/video0[43:cap]: Queueing buffer 3
[0:49:25.136614389] [4050] DEBUG V4L2 v4l2_videodevice.cpp:1764 /dev/video0[43:cap]: Dequeuing buffer 0
[0:49:25.136748066] [4050] DEBUG Request request.cpp:122 Request(20:C:0/1:0)
QtPreview: unexpected colour space Rec709/Rec709/Rec601/Limited
[0:49:25.139444271] [4050] DEBUG V4L2 v4l2_videodevice.cpp:1693 /dev/video0[43:cap]: Queueing buffer 0
[0:49:25.168567370] [4050] DEBUG V4L2 v4l2_videodevice.cpp:1764 /dev/video0[43:cap]: Dequeuing buffer 1
[0:49:25.168695945] [4050] DEBUG Request request.cpp:122 Request(21:C:0/1:0)
QtPreview: unexpected colour space Rec709/Rec709/Rec601/Limited
[0:49:25.171388382] [4050] DEBUG V4L2 v4l2_videodevice.cpp:1693 /dev/video0[43:cap]: Queueing buffer 1
[0:49:25.204667842] [4050] DEBUG V4L2 v4l2_videodevice.cpp:1764 /dev/video0[43:cap]: Dequeuing buffer 2
[0:49:25.204804042] [4050] DEBUG Request request.cpp:122 Request(22:C:0/1:0)
QtPreview: unexpected colour space Rec709/Rec709/Rec601/Limited
[0:49:25.207474378] [4050] DEBUG V4L2 v4l2_videodevice.cpp:1693 /dev/video0[43:cap]: Queueing buffer 2
[0:49:25.236646237] [4050] DEBUG V4L2 v4l2_videodevice.cpp:1764 /dev/video0[43:cap]: Dequeuing buffer 3
[0:49:25.236783434] [4050] DEBUG Request request.cpp:122 Request(23:C:0/1:0)
[0:49:25.237024703] [4041] DEBUG Camera camera.cpp:1235 Stopping capture
[0:49:25.237218193] [4050] DEBUG V4L2 v4l2_videodevice.cpp:1693 /dev/video0[43:cap]: Queueing buffer 3
[0:49:25.239333107] [4050] DEBUG Request request.cpp:122 Request(24:X:0/1:0)
[0:49:25.239392876] [4050] DEBUG Request request.cpp:122 Request(25:X:0/1:0)
[0:49:25.239413649] [4050] DEBUG Request request.cpp:122 Request(26:X:0/1:0)
[0:49:25.239432263] [4050] DEBUG Request request.cpp:122 Request(27:X:0/1:0)
[0:49:25.239482604] [4050] DEBUG V4L2 v4l2_videodevice.cpp:1539 /dev/video0[43:cap]: Releasing buffers
[0:49:25.239525837] [4050] DEBUG V4L2 v4l2_videodevice.cpp:1254 /dev/video0[43:cap]: 0 buffers requested.
  ```

vivien, 2023-09-04:
@6by9 Any idea what to look into next?

6by9, 2023-09-04:
As @naushir has said, neither x86 nor UVC are not the target for the Pi libcamera-apps. The fact that you're trying to send MJPEG compressed images to the display without decompression is going to fail.

If you look at [cam](https://git.linuxtv.org/libcamera.git/tree/src/apps/cam) you'll see it has a dependency on libjpeg.
[qcam](https://git.linuxtv.org/libcamera.git/tree/src/apps/qcam/format_converter.cpp#n187) uses QT which has a mechanism for handling (M)JPEG via [QImage's loadFromData](https://doc.qt.io/qt-5/qimage.html#loadFromData).

If you select YUYV (assuming your webcam supports it), then you might get somewhere. The DRM path certainly isn't going to though, as the hardware doesn't support it. I don't know about EGL.