raspberrypi/rpicam-apps #557: Package not found on raspbian bookworm.

bsasealteam6, 2023-08-21:
Here is a list of all my Repos:
Hit:1 http://security.debian.org/debian-security bookworm-security InRelease
Hit:2 http://deb.debian.org/debian bookworm InRelease
Hit:3 http://deb.debian.org/debian bookworm-updates InRelease
Hit:4 https://downloads.1password.com/linux/debian/arm64 stable InRelease
Hit:5 http://deb.debian.org/debian bookworm-backports InRelease
Hit:6 http://archive.raspberrypi.org/debian bookworm InRelease
Hit:7 http://packages.azlux.fr/debian bookworm InRelease
Hit:8 https://deb.nodesource.com/node_18.x bookworm InRelease
Get:9 https://pkgs.tailscale.com/stable/debian bookworm InRelease
Hit:10 https://download.docker.com/linux/debian bookworm InRelease
Hit:11 https://www.deb-multimedia.org bookworm InRelease

And libcamera-apps and libcamera-apps-lite are not found.  Am I missing a repo, or is it not available on Raspbian Bookworm yet?

naushir, 2023-08-22:
@bsasealteam6 can I check where you got this Raspberry Pi OS Bookworm image from?

BlackNet, 2023-08-22:
It's yet to be available.  Being worked on what we are told.

naushir, 2023-08-22:
In which case, I think we are jumping the gun a bit here.  Once the Raspberry Pi bookworm image is available, it will have the relevant libraries for libcamera and libcamera-apps.

BlackNet, 2023-08-22:
I can't say for the OP but mine was from the beta forums. Also, I am heavy into the 3D printing community and we live by libcamera.

bsasealteam6, 2023-08-22:
I'm going to be honest, I tried the changing sources.list to bookworm.  Im used to debian, so i assumed it would be the same

naushir, 2023-08-23:
As @6by9 mentioned, the Bookwork OS image is still not publicly available and changing the sources.list to point to Bookworm will not work.