raspberrypi/rpicam-apps #561: [BUG]libcamera-hello  preview window unavailable

guxiao-tetras, 2023-08-29:
**Describe the bug**
libcamera hello failes to start preview window bug but is able to run.
**Bug report**
--------------------------------------------------------------------------------
libcamera-apps Bug Report
--------------------------------------------------------------------------------
Date: 29-08-2023 (04:32:19)
Command: /usr/local/bin/camera-bug-report -t 5 -o bugreport -c libcamera-hello

--------------------------------------------------------------------------------
Hardware information
--------------------------------------------------------------------------------
** hostname **
raspberrypi

** cat /proc/cpuinfo **
processor	: 0
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2835
Revision	: d03115
Serial		: 100000002352a334
Model		: Raspberry Pi 4 Model B Rev 1.5

--------------------------------------------------------------------------------
Configuration
--------------------------------------------------------------------------------
** cat /boot/cmdline.txt **
console=serial0,115200 console=tty1 root=PARTUUID=b383bb6d-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
** cat /boot/config.txt **
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

[... 1052 lines cut ...]
#123 (30.01 fps) exp 33222.00 ag 4.69 dg 1.00
#124 (30.02 fps) exp 33222.00 ag 4.69 dg 1.00
#125 (30.02 fps) exp 33222.00 ag 4.69 dg 1.00
#126 (30.02 fps) exp 33222.00 ag 4.69 dg 1.00
#127 (30.02 fps) exp 33222.00 ag 4.69 dg 1.00
#128 (30.02 fps) exp 33222.00 ag 4.69 dg 1.00
#129 (30.01 fps) exp 33222.00 ag 4.69 dg 1.00
#130 (30.02 fps) exp 33222.00 ag 4.69 dg 1.00
#131 (30.02 fps) exp 33222.00 ag 4.69 dg 1.00
#132 (30.02 fps) exp 33222.00 ag 4.69 dg 1.00
#133 (30.01 fps) exp 33222.00 ag 4.69 dg 1.00
#134 (30.02 fps) exp 33222.00 ag 4.69 dg 1.00
#135 (30.01 fps) exp 33222.00 ag 4.69 dg 1.00
#136 (30.02 fps) exp 33222.00 ag 4.69 dg 1.00
Error: ***** TIMEOUT *****

**Additional context**
Add any other context about the problem here.

I did a little debug and found the the problem is here.
DrmPreview::DrmPreview(Options const *options) : Preview(options), last_fd_(-1), first_time_(true)
**if (!drmIsMaster(drmfd_)) {**
this will fail and throw an exception.
_throw std::runtime_error("DRM preview unavailable - not master");_
BTW when handling the exception, the error msg is thrown away, which makes it hard to locate the problem.

naushir, 2023-08-29:
You cannot use the DRM preview in a GUI windowing environment as that is the DRM master.  DRM preview will only work in console mode.  In a windowing environment, you should be using the EGL or QT preview modes.

```
> ** libcamera-hello --version **
> libcamera-apps build: https://github.com/raspberrypi/libcamera-apps/commit/a8e6df7bf18c0e418d7f220ba6fd84788f3a691a-intree-dirty 28-08-2023 (10:11:26)
> libcamera build: v0.0.5+84-27f1d7d4
```

You seem to be running a custom build of libcamera-apps.  What's most likely happened is the EGL preview libraries have not been built because of missing dependencies, and the application falls back to DRM preview and fails like above.  You could either use ``--qt-preview`` to force QT preview mode (assuming the libraries have been built) or rebuild with the correct dependences so that EGL preview is available.  Instructions for this can be found [here](https://www.raspberrypi.com/documentation/computers/camera_software.html).

guxiao-tetras, 2023-08-29:
> You cannot use the DRM preview in a GUI windowing environment as that is the DRM master. DRM preview will only work in console mode. In a windowing environment, you should be using the EGL or QT preview modes.
> 
> ```
> > ** libcamera-hello --version **
> > libcamera-apps build: a8e6df7bf18-intree-dirty 28-08-2023 (10:11:26
> > libcamera build: v0.0.5+84-27f1d7d4
> ```
> 
> You seem to be running a custom build of libcamera-apps. What's most likely happened is the EGL preview libraries have not been built because of missing dependencies, and the application falls back to DRM preview and fails like above. You could either use `--qt-preview` to force QT preview mode (assuming the libraries have been built) or rebuild with the correct dependences so that EGL preview is available. Instructions for this can be found [here](https://www.raspberrypi.com/documentation/computers/camera_software.html).

thanks for your reply.
I tried --qt-preview wiht no luck.  preview window unavailable.
I then built libepoxy but still, no.
Then I tried meson setup with --wipe.
somehow this did the trick.

I found 
> if enable_egl and x11_deps.found() and epoxy_deps.found()
>     libcamera_app_dep += [x11_deps, epoxy_deps]
>     libcamera_app_src += files('egl_preview.cpp')
>     cpp_arguments += '-DLIBEGL_PRESENT=1'
> else
>     enable_egl = false
> endif
in meson.build.
Is it possible that _enable_egl_ is not updated when dependencies are met after the first build?

I don't understand what you said about libdrm. did you mean libdrm is supposed to be used in pure console env w/o X, like tty4?

Anyway, problem solved. Thank you.

naushir, 2023-08-30:
> Is it possible that enable_egl is not updated when dependencies are met after the first build?

That's correct, you will need to rerun meson setup to pick up the new dependencies.

> I don't understand what you said about libdrm. did you mean libdrm is supposed to be used in pure console env w/o X, like tty4?

Correct, DRM preview will only work in a console environment since there can only be one DRM master in use.  If X was running, it would be the DRM master.

If you have no more issues with this, feel free to close this down.

guxiao-tetras, 2023-08-30:
> 

Really thank you very much. I learned  a lot from this issue.

closed 2023-08-30 by guxiao-tetras