raspberrypi/rpicam-apps #562: [BUG] RPi V1 camera's reported maximum ExposureTime incorrect?

EricClaeys, 2023-09-02:
**Describe the bug**
RPi V1 camera's reported maximum ExposureTime not consistent with camera information.

**Bug report**
When an RPi V1 camera is connected, running `libcamera-still --list-cameras -v` returns:
~~~
Available cameras
-----------------
0 : ov5647 [2592x1944] (/base/soc/i2c0mux/i2c@1/ov5647@36)
    Modes: 'SGBRG10_CSI2P' : 640x480 [58.92 fps - (16, 0)/2560x1920 crop]
                             1296x972 [43.25 fps - (0, 0)/2592x1944 crop]
                             1920x1080 [30.62 fps - (348, 434)/1928x1080 crop]
                             2592x1944 [15.63 fps - (0, 0)/2592x1944 crop]

    Available controls for 2592x1944 SGBRG10_CSI2P mode:
    ----------------------------------------------------
    AeConstraintMode : [0..3]
    AeEnable : [false..true]
    AeExposureMode : [0..3]
    AeMeteringMode : [0..3]
    AnalogueGain : [1.000000..63.937500]
    AwbEnable : [false..true]
    AwbMode : [0..7]
    Brightness : [-1.000000..1.000000]
    ColourGains : [0.000000..32.000000]
    Contrast : [0.000000..32.000000]
    ExposureTime : [130..969249]
    ExposureValue : [-8.000000..8.000000]
    FrameDurationLimits : [63965..1065021]
    NoiseReductionMode : [0..4]
    Saturation : [0.000000..32.000000]
    ScalerCrop : [(0, 0)/64x64..(0, 0)/2592x1944]
    Sharpness : [0.000000..16.000000]
~~~

This implies the maximum ExposureTime is 969249 us or about 1 second.  However, per [this link](https://www.waveshare.com/wiki/RPi_Camera_(G)#libcamera-still) the maximum is 6 seconds:
[image]

naushir, 2023-09-04:
I'm not entirely familiar with the Camera v1, but in this case, I would trust what ``libcamera-apps`` says as it computes the exposure time from the blanking limits advertised by the sensor driver.

I do recall that ov5647 had a special mode in the firmware driver that would slow the clocks down to get a larger exposure time, but this may not have been ported across to the kernel driver (particularly because it is not easily workable with the V4L2 APIs).  @6by9 might be able to confirm this.

I will make an update to the documentation to fix this error to the above table.

6by9, 2023-09-04:
Links to 3rd party websites are erroneous - 3rd parties can write what they like and we have no control over it.

OV5647 base configuration is limited at ~1second. Way back I added an extra mode to the legacy driver that messed with the PLL settings, and possibly line length as well.
Adding control of line length to the V4L2 driver should be trivial, and libcamera already sets that where possible.
Changing PLL settings is more troublesome as it is then a compromise between max exposure times and max frame rate, and there is no sensible control to do that.

I'll see if there is a quick win available on line length.
Doing a diff on modes 2 & 3 from https://github.com/raspberrypi/raspiraw/blob/master/ov5647_modes.h (which is the same register set as the firmware uses), HTS is 2844 in mode 2, and as a 13bit value can go up to 8191, so that would get us back up to ~2.8secs.
PLL change is from a multiplier of 0x69 (105 decimal) to 0x34 (52 decimal), which would halve the clock rate. I may resort to a module parameter, but that also then means pixel rate and other V4L2 params have to change.

6by9, 2023-09-04:
> The "FE without FS" check was added to fix (i think!) a problem with the imx219 at fast framerates.

4e7680cadd832d49c8ce52c67c459df1b3e7232c
```
media: bcm2835-unicam: Correctly handle FS + FE ISR condtion
    
    If we get a simultaneous FS + FE interrupt for the same frame, it cannot be
    marked as completed and returned to userland as the framebuffer will be refilled
    by Unicam on the next sensor frame. Additionally, the timestamp will be set to 0
    as the FS interrupt handling code will not have run yet.
    
    To avoid these problems, the frame is considered dropped in the FE handler,
    and will be returned to userland on the subsequent sensor frame.
```

I just don't quite see how the timestamp isn't set at the FE. The lines_done is the only obviously quirky thing. Time for some extra logging, but may have to wait.

EricClaeys, 2023-09-04:
> I'm not entirely familiar with the Camera v1, but in this case, I would trust what `libcamera-apps` says as it computes the exposure time from the blanking limits advertised by the sensor driver.
> 
> I will make an update to the documentation to fix this error to the above table.

The RPi documentation also has 6 seconds as the max exposure, and the user who noted this discrepancy is able to take 6-second shots:
~~~
libcamera-still -o html/allsky/6s.jpg --shutter 6000000 --gain 1 --awbgains 1,1 --immediate
~~~

naushir, 2023-09-04:
I think what would happen is that libcamera would truncate the shutter speed down to 0.9 seconds as the kernel driver advertises that as the maximum limit.

6by9, 2023-09-04:
Even stranger. I've got 3.1s/frame from the 1296x972 mode with no issues, but the full res 2592x1944 is doing something funny at the CSI2 level as soon as the horizontal blanking control goes above 545.

No more time for this now, so I'll have to park it. Patch pushed to https://github.com/6by9/linux/tree/rpi-6.1.y-ov5647/

naushir, 2023-09-05:
Sounds like this might need further investigating when we have more time/resources.   For now, I think the right thing would be to update the documentation to reflect the actual shutter limit to resolve this.

naushir, 2023-09-05:
Documentation has been updated.  I'll close this one down now.

closed 2023-09-05 by naushir