raspberrypi/rpicam-apps #564: [BUG] no cameras available issue (Raspberry Pi 4 Model B Rev 1.5)

tsofu, 2023-09-14:
**Describe the bug**
Test environment: Camera module 2 (sony IMX2198-megapixel sensor) is connected to Raspberry Pi 4 (Raspberry Pi 4 Model B Rev 1.5). We're going to capture a camera image.

Issue
We are going to use the libcamera module.
However, when you enter the libcamera-hello command, the error message "no camera available" is checked.
Could we know what kind of problem this symptom is?

**Bug report**
[bug_libcamera.txt](https://github.com/raspberrypi/libcamera-apps/files/12603208/bug_libcamera.txt)

**Additional context**
Image capture works fine with the raspistill -o test.jpg command.

naushir, 2023-09-14:
It looks like you have enabled the legacy camera stack (hence ``raspistill`` works).  Please flash a fresh SD card image, and don't do anything else to enable camera.  The libcamera stack is enabled by default on all our images and ``libcamera-hello`` should just work as intended.

closed 2023-09-14 by naushir

tsofu, 2023-09-15:
@naushir 
Thank you.
It helped me solve the problem.