raspberrypi/rpicam-apps #569: dma heap | failed to queue buffer

schoolpost, 2023-09-25:
I am testing the dma_heap branch and after building libcamera-apps from source, this it the output I am getting:

```
~/dma_heap/libcamera-apps $ libcamera-hello -t 0
Made DRM preview window
[1:53:51.857752748] [84940]  INFO Camera camera_manager.cpp:299 libcamera v0.0.4+23-06bbff9c-dirty (2023-09-07T02:52:53-06:00)
[1:53:51.867337327] [84941]  WARN RPI raspberrypi.cpp:1367 Mismatch between Unicam and CamHelper for embedded data usage!
[1:53:51.868152976] [84941]  INFO RPI raspberrypi.cpp:1783 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[1:53:51.868415242] [84941]  INFO RPI raspberrypi.cpp:1486 Registered camera /base/soc/i2c0mux/i2c@1/imx283@1a to Unicam device /dev/media3 and ISP device /dev/media2
[1:53:51.869310511] [84940]  INFO Camera camera.cpp:1028 configuring streams: (0) 2736x1824-YUV420
[1:53:51.869681432] [84941]  INFO RPI raspberrypi.cpp:861 Sensor: /base/soc/i2c0mux/i2c@1/imx283@1a - Selected sensor format: 2796x1846-SRGGB12_1X12 - Selected unicam format: 2796x1846-pRCC
[1:53:52.602303120] [84941] ERROR RPISTREAM rpi_stream.cpp:243 Failed to queue buffer for ISP Output0
[1:53:52.602344893] [84941] ERROR RPISTREAM rpi_stream.cpp:243 Failed to queue buffer for ISP Output0
[1:53:52.602664005] [84941] ERROR RPISTREAM rpi_stream.cpp:243 Failed to queue buffer for ISP Output0
```

Using a CM4 / 4GB model with a IMX283 sensor and custom kernel driver. ( this camera has been tested and does work with an older version of libcamera-apps, so I do not believe it's a hardware issue. )

schoolpost, 2023-09-25:
Here is another instance of the error with a different app:

```
~/dma_heap/libcamera-apps $ libcamera-raw -t 0
[2:39:27.029404587] [112337]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+84-27f1d7d4-dirty (2023-09-25T15:00:44-06:00)
[2:39:27.134568915] [112338]  WARN RPI vc4.cpp:383 Mismatch between Unicam and CamHelper for embedded data usage!
[2:39:27.136981248] [112338]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/imx283@1a to Unicam device /dev/media3 and ISP device /dev/media0
[2:39:27.137160110] [112338]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/local/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Stream configuration adjusted
[2:39:27.140992470] [112337]  INFO Camera camera.cpp:1033 configuring streams: (0) 640x480-YUV420 (1) 2796x1846-SRGGB12_CSI2P
[2:39:27.142405164] [112338]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx283@1a - Selected sensor format: 2796x1846-SRGGB12_1X12 - Selected unicam format: 2796x1846-pRCC
[2:39:27.861140638] [112338] ERROR RPISTREAM rpi_stream.cpp:249 Failed to queue buffer for Unicam Image
[2:39:27.861192099] [112338] ERROR RPISTREAM rpi_stream.cpp:249 Failed to queue buffer for Unicam Image
[2:39:27.861216690] [112338] ERROR RPISTREAM rpi_stream.cpp:249 Failed to queue buffer for Unicam Image
[2:39:27.861238726] [112338] ERROR RPISTREAM rpi_stream.cpp:249 Failed to queue buffer for Unicam Image
[2:39:27.894393798] [112338] ERROR RPISTREAM rpi_stream.cpp:249 Failed to queue buffer for ISP Output0
[2:39:27.894473739] [112338] ERROR RPISTREAM rpi_stream.cpp:249 Failed to queue buffer for ISP Output0
[2:39:27.894514922] [112338] ERROR RPISTREAM rpi_stream.cpp:249 Failed to queue buffer for ISP Output0
[2:39:27.894552809] [112338] ERROR RPISTREAM rpi_stream.cpp:249 Failed to queue buffer for ISP Output0
[2:39:27.894590882] [112338] ERROR RPISTREAM rpi_stream.cpp:249 Failed to queue buffer for ISP Output0
terminate called after throwing an instance of 'std::out_of_range'
  what():  _Map_base::at
Aborted
```
version info: 
```
~/dma_heap/libcamera-apps $ libcamera-raw --version
libcamera-apps build: 6058483639a2-intree-dirty 25-09-2023 (15:18:00)
libcamera build: v0.0.5+84-27f1d7d4-dirty (2023-09-25T15:00:44-06:00)
```

naushir, 2023-09-26:
You need to build/install the latest version of libcamera from git.libcamer.org.  This has the necessary libcamera side changes for the dma_heap branch to work correctly.

naushir, 2023-09-26:
@schoolpost I assume this is now working to some degree?  Would you be able to comment on whether this change gets you any improved performance in your application?

schoolpost, 2023-09-26:
Yes, installing the latest 0.1.0 version of libcamera solved the issue. 

And from preliminary tests, it does seem to provide a noticeable improvement in performance.

Thanks!