raspberrypi/rpicam-apps #570: Recording Low-Res with IMX708 (Camera Module 3 Wide)

GuyPerets106, 2023-09-27:
Hi , I'm using RPi CM4 8GB RAM 32GB eMMC , with Wavehsrae Mini IO Board , and two Module 3 cameras connected.
I'm trying to record 640x400 video , but it turns out that the lowest resolution mode is 1536x864 and it's cropped. I need that wide FOV.
The next mode is uncropped 2304x1296 but it does not allow high framerate (>100fps).
currently the command I use is `libcamera-vid --mode 1536:864:8 --width 640 --height 400 --framerate 100 -n --denoise off --level 4.2 --hdr 0 --shutter <X> --gain <Y> -o test.h264`
If I specify the `--mode 2304x1296:8` I get uncropped video but get a lot of frame skips so high framerate is not available in that mode. 
I'm not even sure the specification of an 8-bit video is even possible with that sensor , so I'd appreciate clarification about this too. 
There is no option for recording low-resolution and high-framerate video with that sensor? 
Thanks!

naushir, 2023-09-27:
Unfortunately if you want the faster framerate modes, you are going to have to use the cropped 1536x864 10-bit non-hdr mode.  Note that the sensor driver does not support 8-bit output, so you have to use 10-bits.

Another thing to mention is that since you are using ``--hdr``, you will be limited to a single mode 2304x1296 and that can only run at 30fps.  The non-hdr 2304x1296 can run at 56fps max.

If you do need 100fps with a wide horizontal FoV, I suggest you use the wide-angle Camera v3 variant (if you are not already) and use the cropped 1536x864 mode.

GuyPerets106, 2023-09-27:
> Unfortunately if you want the faster framerates, you are going to have to use the cropped 1536x864 10-bit non-hdr mode. Note that the sensor driver does not support 8-bit output, so you have to use 10-bits.
> 
> Another thing to mention is that since you are using `--hdr`, you will be limited to a single mode 2304x1296 and that can only run at 30fps. The non-hdr 2304x1296 can run at 56fps max.
> 
> If you do need 100fps with a wide horizontal FoV, I suggest you use the wide-angle Camera v3 variant (if you are not already) and use the cropped 1536x864 mode.

Thanks for replying. 
Yes , I'm using x2 Module 3 Wide cameras on my setup. 
So when I specify `--width 640 --height 400` the sensor automatically captures 1536x864 (or any higher-res mode) frames and then just lowers the resolution to 640x400? 
Meaning there's no way to record high-framerate videos with this sensor as for now? (or until RPi 5 comes out with a better encoder)

GuyPerets106, 2023-09-27:
> > So when I specify `--width 640 --height 400` the sensor automatically captures 1536x864 (or any higher-res mode) frames and then just lowers the resolution to 640x400?
> 
> When you specify `--width 640 --height 400`, the sensor will run generate a 1536x864 resolution image (cropped), and the ISP will further crop that to match the 4:2.5 aspect ratio you have chosen and rescale down to 640x400. If you want to avoid the ISP doing any cropping/rescaling, you can specify `--width 1536 --height 864` instead.
> 
> > Meaning there's no way to record high-framerate videos with this sensor as for now? (or until RPi 5 comes out with a better encoder)
> 
> With a 1536x864 stream, I would expect the encoder to keep up somewhere in the region of 80-90fps. You could then lower the output resolution slightly (but keep the 16:9 aspect ratio) to find the working point that achieves your 100 fps target.

Thank you for clarifying.

closed 2023-09-27 by GuyPerets106