raspberrypi/rpicam-apps #571: [BUG] No cameras available! / Illegal instruction

LeifSec, 2023-09-29:
**Describe the bug**
I tested a lot of combination for `dtoverlay` and  `camera_auto_detect` and for all I get for `libcamera-hello --list-cameras`
```
No cameras available!
``` 
and for `libcamera-hello` (-n)
```
Illegal instruction
```

**Bug report**
I used `/ibcamera-apps/utils/camera-bug-report -t 10000 -c "libcamera-hello -t 5000`
and attached file is for `camera_auto_detect=1`and no `dtoverlay`.
[bug-report.txt](https://github.com/raspberrypi/libcamera-apps/files/12764416/bug-report.txt)

**Additional context**
* `vcgencmd get_camera`is always `supported=0 detected=0, libcamera interfaces=0`
* For some  `dtoverlay` `dmesg` is reporting e.g. `imx519 10-001a: failed to read chip id 519, with error -5`- since I am using Raspberry Pi Camera HD this should be consequential.
* `uname -a`:  ```Linux *** 6.1.54-v7+ #1682 SMP Wed Sep 20 15:16:25 BST 2023 armv7l GNU/Linux```
* `lsb_release -a`
 ```
 No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm
```
* `vcgencmd version`
 ```
Sep 21 2023 18:44:30 
Copyright (c) 2012 Broadcom
version 3094eda5aadf2c719edd2c68d1900884abc3dbca (clean) (release) (start) ```
* `libcamera-hello --version`
``l`ibcamera-apps build: 83e5ae54bfc1-intree 29-09-2023 (07:14:00)
libcamera build: v0.0.5+84-27f1d7d4-dirty (2023-09-29T09:33:50+01:00)
```
* enabling  stack and using `raspistill` the camera is working.

6by9, 2023-09-29:
> For some dtoverlay dmesg is reporting e.g. imx519 10-001a: failed to read chip id 519, with error -5- since I am using Raspberry Pi Camera HD this should be consequential.

imx519 is an Arducam module, not Raspberry Pi.

```camera_auto_detect=1``` should detect any of the camera modules made by Raspberry Pi:
- v1, ov5647 5MPix
- v2, imx219 8MPix
- v3, imx708 12MPix with AF
- HQ, imx477
- Global Shutter, imx296

I'm assuming your HD is a typo for HQ, in which case the correct line would be ```dtoverlay=imx477```. Any of the others will break the system if ```camera_auto_detect=1``` is still active.

The media-ctl output is showing part of a unicam node, but with no sensor driver registered.

Can you please give the output from:
- ```sudo vclog -m```
- ```sudo cat /sys/kernel/debug/devices_deferred```
- ```sudo cat /sys/kernel/debug/v4l2-async/pending_async_subdevices```

LeifSec, 2023-09-29:
Yes - it is a HQ (not a HD) - sorry for mistake.

I now have
* `camera_auto_detect=1`
* `dtoverlay=imx477`

`dmesg` still reports
``` 
[   19.009442] ov5647 10-0036: Unsupported clock frequency: 24000000
[   19.009489] ov5647: probe of 10-0036 failed with error -22
[   19.374451] imx477 10-001a: failed to read chip id 477, with error -5
[   19.377526] imx477: probe of 10-001a failed with error -5
```
while `ov5647 `is for sure comment out in `boot/config.txt` .

* `sudo vclog -m`
```
```
* `sudo cat /sys/kernel/debug/devices_deferred`
```
``` 
* `sudo cat /sys/kernel/debug/v4l2-async/pending_async_subdevices`
```
unicam 3f801000.csi:
 [fwnode] dev=10-001a, node=/soc/i2c0mux/i2c@1/imx477@1a/port/endpoint
```

naushir, 2023-10-05:
Can you try this again with a fresh OS image on a spare sd-card and see if you get the same problems?

LeifSec, 2023-10-05:
Then I will need to use Raspberry Pi OS 11  because 12 (bookworm) is not yet [released](https://www.raspberrypi.com/software/operating-systems/).
I will try that again.Nevertheless I did already so in the past but maybe without trying `dtoverlay=imx477`.

I also tried direct [Debian 12 Image](https://raspi.debian.net/tested-images/) and try to install `libcamera` there. Finally I used  Raspberry Pi OS 11 and updated the package sources to Raspberry Pi OS 12 which are already available. Because corresponding packages are also missing there I built everything from source.

naushir, 2023-10-05:
> I will try that again.Nevertheless I did already so in the past but maybe without trying `dtoverlay=imx477`.

On a fresh image, you should not have to make any modifications/changes to get the camera working.

LeifSec, 2023-10-05:
* Fresh [Debian 11 Raspberry PI OS image](https://downloads.raspberrypi.org/raspios_full_armhf/images/raspios_full_armhf-2023-05-03/2023-05-03-raspios-bullseye-armhf-full.img.xz)
* No changes
* Still `libcamera-hello` -> `ERROR: *** no cameras available ***`
* `apt-get install libcamera-apps` and `
camera-bug-report -t 10000 -c "libcamera-hello -t 5000` attached [bug-report.txt](https://github.com/raspberrypi/libcamera-apps/files/12819145/bug-report.txt)
* replaced `dtoverlay=vc4-kms-v3d` by `dtoverlay=imx477` in `config.txt`, reboot -> same error
* activate legacy camera support, reboot, `raspistill`-> `mmal: Cannot read camera info, keeping the defaults for OV5647 ...`
*  replaced `dtoverlay=imx477`  by  `dtoverlay=vc4-kms-v3d`  again, reboot, `raspistill` -> works - thus, I am sure that the camera and it's connection is OK.

naushir, 2023-10-05:
> * Fresh [Debian 11 Raspberry PI OS image](https://downloads.raspberrypi.org/raspios_full_armhf/images/raspios_full_armhf-2023-05-03/2023-05-03-raspios-bullseye-armhf-full.img.xz)
> * No changes
> * Still `libcamera-hello` -> `ERROR: *** no cameras available ***`
....
> * replaced `dtoverlay=imx477`  by  `dtoverlay=vc4-kms-v3d`  again, reboot, `raspistill` -> works - thus, I am sure that the camera and it's connection is OK.

This seems like quite a contradictory result.  I cannot see any way where the legacy stack would work and not the standard libcamera stack.  Where did you purchase camera board from, and has it ever worked in the past?

LeifSec, 2023-10-06:
> This seems like quite a contradictory result. I cannot see any way where the legacy stack would work and not the standard libcamera stack. Where did you purchase camera board from,

From an official (German) distributor - it is for sure not a clone. 
> and has it ever worked in the past?

Never with `libcamera` and always with `raspistill`. 
I am not sure any more if `vcgencmd get_camera` outputted some value != 0 in the past - but I could not reproduce it.

naushir, 2023-10-06:
Do you have another Raspberry Pi board to try this camera out with?  Or alternatively, do you have another Raspberry Pi camera board to try out?

LeifSec, 2023-10-12:
I tried now with a fresh Raspbian 12 image which is available since yesterday. I tried a second Raspberry 2B and a Raspberry 3 - same results (just the `Illegal instruction`error disappeared.)

LeifSec, 2023-10-12:
I tried now with a fresh Raspbian 12 image which is available since yesterday. I tried a second Raspberry 2B and a Raspberry 3 - same results (just the `Illegal instruction`error disappeared.)  
And the legacy camera option via `raspi-config`in final Raspbian 12 has been remove ...

naushir, 2023-10-12:
Can you provide the full output of the following command:
```bash
LIBCAMERA_LOG_LEVELS=*:0 libcamera-hello -t 0
```
?

LeifSec, 2023-10-12:
```
 LIBCAMERA_LOG_LEVELS=*:0 libcamera-hello -t 0
Preview window unavailable
[2:21:32.660197647] [1629] DEBUG IPAModule ipa_module.cpp:334 ipa_rkisp1.so: IPA module /usr/lib/arm-linux-gnueabihf/libcamera/ipa_rkisp1.so is signed
[2:21:32.660414882] [1629] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/lib/arm-linux-gnueabihf/libcamera/ipa_rkisp1.so'
[2:21:32.660824509] [1629] DEBUG IPAModule ipa_module.cpp:334 ipa_rpi_pisp.so: IPA module /usr/lib/arm-linux-gnueabihf/libcamera/ipa_rpi_pisp.so is signed
[2:21:32.661053045] [1629] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/lib/arm-linux-gnueabihf/libcamera/ipa_rpi_pisp.so'
[2:21:32.661344133] [1629] DEBUG IPAModule ipa_module.cpp:334 ipa_rpi_vc4.so: IPA module /usr/lib/arm-linux-gnueabihf/libcamera/ipa_rpi_vc4.so is signed
[2:21:32.661546889] [1629] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/lib/arm-linux-gnueabihf/libcamera/ipa_rpi_vc4.so'
[2:21:32.661689386] [1629] DEBUG IPAModule ipa_module.cpp:334 ipa_vimc.so: IPA module /usr/lib/arm-linux-gnueabihf/libcamera/ipa_vimc.so is signed
[2:21:32.661808758] [1629] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/lib/arm-linux-gnueabihf/libcamera/ipa_vimc.so'
[2:21:32.661985108] [1629]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+52-a858d20b
[2:21:32.662434838] [1632] DEBUG Camera camera_manager.cpp:69 Starting camera manager
[2:21:32.681388434] [1632] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "bcm2835-codec" created from /dev/media1
[2:21:32.681495723] [1632] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media1 due to 5 missing dependencies
[2:21:32.684257694] [1632] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media1 found
[2:21:32.684319672] [1632] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media1: bcm2835-codec
[2:21:32.684812421] [1632] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "bcm2835-isp" created from /dev/media0
[2:21:32.684877472] [1632] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media0 due to 4 missing dependencies
[2:21:32.685374440] [1632] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "bcm2835-isp" created from /dev/media2
[2:21:32.685445897] [1632] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media2 due to 4 missing dependencies
[2:21:32.687685171] [1632] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media0 found
[2:21:32.687756680] [1632] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media0: bcm2835-isp
[2:21:32.689907414] [1632] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media2 found
[2:21:32.689965746] [1632] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media2: bcm2835-isp
[2:21:32.690454694] [1632] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "unicam" created from /dev/media3
[2:21:32.690521776] [1632] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media3: unicam
[2:21:32.691372382] [1632] DEBUG Camera camera_manager.cpp:113 Found registered pipeline handler 'PipelineHandlerISI'
[2:21:32.691477640] [1632] DEBUG Camera camera_manager.cpp:113 Found registered pipeline handler 'PipelineHandlerRkISP1'
[2:21:32.691578367] [1632] DEBUG Camera camera_manager.cpp:113 Found registered pipeline handler 'PipelineHandlerPiSP'
[2:21:32.691659928] [1632] DEBUG RPI pisp.cpp:837 Unable to acquire a CFE instance
[2:21:32.691699927] [1632] DEBUG Camera camera_manager.cpp:113 Found registered pipeline handler 'PipelineHandlerVc4'
[2:21:32.691742739] [1632] DEBUG DeviceEnumerator device_enumerator.cpp:312 Successful match for media device "unicam"
[2:21:32.691822633] [1632] DEBUG DeviceEnumerator device_enumerator.cpp:312 Successful match for media device "bcm2835-isp"
[2:21:32.691875288] [1632] DEBUG RPI vc4.cpp:179 Unable to acquire a Unicam instance
[2:21:32.692110491] [1632] DEBUG Camera camera_manager.cpp:113 Found registered pipeline handler 'SimplePipelineHandler'
[2:21:32.692172208] [1632] DEBUG Camera camera_manager.cpp:113 Found registered pipeline handler 'PipelineHandlerUVC'
[2:21:32.692216791] [1632] DEBUG Camera camera_manager.cpp:113 Found registered pipeline handler 'PipelineHandlerVimc'
ERROR: *** no cameras available ***
```
with fresh [Raspian12](https://downloads.raspberrypi.com/raspios_full_armhf/images/raspios_full_armhf-2023-10-10/2023-10-10-raspios-bookworm-armhf-full.img.xz)  on Raspberry3

naushir, 2023-10-12:
Thank you.  That shows that the camera has been detected by the kernel (since a Unicam media device is present), yet libcamera fails to load the device.

Can you attach the file generated from the output of 
```bash
camera-bug-report
``
please?

LeifSec, 2023-10-12:
[bug-report.txt](https://github.com/raspberrypi/libcamera-apps/files/12883520/bug-report.txt)

6by9, 2023-10-12:
```
[   11.202867] ov5647: probe of 10-0036 failed with error -5
```
ov5647 is a V1 camera module, not HQ. 

What does ```sudo i2cdetect -y 10``` report? You may need to ```sudo modprobe i2c-dev``` first to allow userspace access.

naushir, 2023-10-12:
Thanks.  From that log:

> [   11.202867] ov5647: probe of 10-0036 failed with error -5

This looks to be the same as the original problem, so the sensor has failed to probe.  This definitely looks to be a hardware issue to me.  It might be a cable issue - try removing and replacing the camera cable, ensuring it is seated correctly in the catches.

LeifSec, 2023-10-12:
> ```
> [   11.202867] ov5647: probe of 10-0036 failed with error -5
> ```
> 
> ov5647 is a V1 camera module, not HQ.
> 
It is labeled with `Raspberry Pi Camera`
> What does `sudo i2cdetect -y 10` report? 
``` 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                   
``` 

You may need to `sudo modprobe i2c-dev` first to allow userspace access.
-> no effect

LeifSec, 2023-10-13:
Just changing SD-card to one with Raspberry OS 11 and using `raspistill`it work -> cabling is OK.

But you are right (sorry for confusing): The camera is not a HQ camera.

Using a HQ camera it works with `libcamera` - so the problem at my side is only with `ov5647 - V1 camera module`. As we are using quite only HQ camera this is not a large problem.

Thank for your help.

naushir, 2023-10-13:
Glad you got it resolved.  Closing now.

closed 2023-10-13 by naushir

6by9, 2023-10-13:
If there's only a problem with OV5647, then Raspberry Pi haven't sold that module since 2016 when v2 / IMX219 was released (https://www.raspberrypi.com/news/new-8-megapixel-camera-board-sale-25/). Omnivision had stated that the sensor was end of life in 2014.

Either your module is very old, or it's made by a third party. We can't control what third parties do, so whilst we do our best on compatibility, it can't be guaranteed.