raspberrypi/rpicam-apps #573: [BUG] camera not detected

TiJof, 2023-10-04:
**camera isn't detected**
I want to use a waveshare 5MP 160° FOV camera, seems based on OV5647 (https://www.waveshare.com/product/robotics/piracer-pro-ai-kit.htm).

The error message is : ERROR: *** no cameras available ***

No modification in the config.txt (just i2c, but the same issue without).

**Bug report from the tools **
```
--------------------------------------------------------------------------------
libcamera-apps Bug Report
--------------------------------------------------------------------------------
Date: 03-05-2023 (05:15:20)
Command: /usr/bin/camera-bug-report -c libcamera-still -t 1000 -o test.jpg

--------------------------------------------------------------------------------
Hardware information
--------------------------------------------------------------------------------
** hostname **
grab

** cat /proc/cpuinfo **
processor	: 0
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2835
Revision	: c03111
Serial		: 10000000caed2093
Model		: Raspberry Pi 4 Model B Rev 1.1

--------------------------------------------------------------------------------
Configuration
--------------------------------------------------------------------------------
** cat /boot/cmdline.txt **
console=serial0,115200 console=tty1 root=PARTUUID=a8a362b8-02 rootfstype=ext4 fsck.repair=yes rootwait
** cat /boot/config.txt **
# For more options and information see
[... 856 lines cut ...]
** libcamera-still --version **
libcamera-apps build: 54a781dffdd1 01-03-2023 (17:53:36)
libcamera build: v0.0.4+22-923f5d70

--------------------------------------------------------------------------------
libcamera-still -t 1000 -o test.jpg
--------------------------------------------------------------------------------
Preview window unavailable
[0:03:36.754337721] [890]  INFO Camera camera_manager.cpp:299 libcamera v0.0.4+22-923f5d70
ERROR: *** no cameras available ***
```

**Additional context**
I've done the tests with a fresh install.

If i specify an overlay for my ov5647 :
```
pi@grab:~ $ grep -i 56 /boot/config.txt 
dtoverlay=ov5647
pi@grab:~ $ dmesg |grep -i OV5647
[    0.076910] platform fe801000.csi: Fixed dependency cycle(s) with /soc/i2c0mux/i2c@1/ov5647@36
[    6.995057] ov5647 10-0036: ID High expected 0x56 got 0
```

Any idea ?

6by9, 2023-10-04:
> ```[    6.995057] ov5647 10-0036: ID High expected 0x56 got 0```

Your sensor isn't replying appropriately, therefore it won't be detected.

At a guess something is shorting SDA to ground, therefore all transactions will return 0. Check with ```sudo i2cdetect -y 10``` (you'll need to have either tried loading the overlay, or add ```dtparam=i2c_vc=on``` to enable the relevant I2C bus, and have the ```i2c-dev``` module loaded).
If just SDA is shorted to ground, then it will run through quickly, but all addresses will be listed.
If SCL is also shorted to ground, then it will run slowly as each query takes 1 second to time out.

Check that the pinmuxing is correct with ```raspi-gpio get 0,1,44,45```. 44&45 should report as SDA0 and SCL0, and 0&1 should be inputs.

TiJof, 2023-10-05:
thank you for your quick response !
```
pi@grab:~ $ time sudo i2cdetect -y 10
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         08 09 0a 0b 0c 0d 0e 0f 
10: 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 
20: 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 
30: 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f 
40: 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f 
50: 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f 
60: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 
70: 70 71 72 73 74 75 76 77                         

real	0m0.101s
user	0m0.025s
sys	0m0.036s

pi@grab:~ $ raspi-gpio get 0,1,44,45
GPIO 0: level=1 fsel=0 func=INPUT pull=UP
GPIO 1: level=1 fsel=0 func=INPUT pull=UP
GPIO 44: level=0 fsel=5 alt=1 func=SDA0 pull=UP
GPIO 45: level=1 fsel=5 alt=1 func=SCL0 pull=UP
```

The i2cdetect run quickly, from what you say SDA seems shorted to ground... Pinmuxing seems good.
So this is a hardware problem ?

TiJof, 2023-10-05:
```
pi@grab:~ $ time sudo i2cdetect -y 10
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         

real	0m0.132s
user	0m0.012s
sys	0m0.053s
```

so, camera is defective... As I have test with 3 of them with same result, we can suppose that this bench of Waveshare camera are terrible ones.

Thank you for your response !

closed 2023-10-05 by TiJof