raspberrypi/rpicam-apps #575: [BUG] Error in 'timeout' parameter while compiling libcamera-hello.cpp manually

ManjunathaVG-MathWorks, 2023-10-04:
Hi,
I am trying to compile the libcamera-hello.cpp program using the below command. However, I am encountering an error related to a timeout. Could you please assist me in resolving this issue?

$ sudo apt-get update
$ sudo apt-get install libcamera-apps licamera-dev  libcamera0 libboost-dev -y
$ # Download libcamera-hello.cpp file from this repo and save it as test.cpp
$ g++ -std=c++17 -o test test.cpp -lcamera -lcamera_app  -I/usr/include/libcamera -I/usr/include/libcamera-apps

test.cpp: In function ‘void event_loop(LibcameraApp&)’:
test.cpp:44:65: error: request for member ‘value’ in ‘options->Options::timeout’, which is of non-class type ‘const uint64_t’ {aka ‘const long long unsigned int’}
   44 |   if (options->timeout && (now - start_time) > options->timeout.value)

Thanks,
Manjunatha

naushir, 2023-10-04:
I can't seem to reproduce this - what I did was go to the libcamera-apps/apps directory and run:
```
g++ -std=c++17 -o test test.cpp -lcamera -lcamera_app -I/usr/include/libcamera -I/usr/include/libcamera-apps
```
it seems to succeed correctly.  Can you try exactly this please?

ManjunathaVG-MathWorks, 2023-10-05:
Hi,
Thanks for replying.

My understanding is I don't have to download the source from this GitHub repo. All the necessary components for libcamera-apps should be available via the command $ sudo apt-get install libcamera-apps.

Therefore, I'm wondering what dependency I might be overlooking in the apt-get install process.

Also, I tried to download the source from the repository and navigated to libcamera-apps/app, but I still see the same issue. Here are the steps that I followed,

$ git clone https://github.com/raspberrypi/libcamera-apps.git
$ cd libcamera-apps/apps/
$ g++ -std=c++17 -o new libcamera_hello.cpp -lcamera -lcamera_app -I/usr/include/libcamera -I/usr/include/libcamera-apps

libcamera_hello.cpp: In function ‘void event_loop(LibcameraApp&)’:
libcamera_hello.cpp:44:65: error: request for member ‘value’ in ‘options->Option                                                                        s::timeout’, which is of non-class type ‘const uint64_t’ {aka ‘const long long u                                                                        nsigned int’}
   44 |   if (options->timeout && (now - start_time) > options->timeout.value)
      |  
	  
	  
However, this doesn't align with my use case, as I shouldn't need to download the source given that everything should be readily available via the apt-get install. Is there something that I am missing?

Here are some info about my setup,
$ uname -a
Linux raspberrypi 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux

$ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye

$ cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 1
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 2
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 3
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

Hardware        : BCM2835
Revision        : a21041
Serial          : 00000000e156716b
Model           : Raspberry Pi 2 Model B Rev 1.1

Thanks,
Manjunatha

naushir, 2023-10-05:
I think I understand the problem now.  You have installed the v1.2.1 release of libcamera-apps on your system.  However, since that release, we have made changes to the libcamera-apps code in this repo, and now your libraries/headers installed on the system are out of date with the ``libcamera_hello.cpp`` source code.

We are in the process of making a new release soon which should fix this issue.  In the meantime, please checkout the "v1.2.1" tag of this repo and use the ``libcamera_hello.cpp`` source file from that tag.

ManjunathaVG-MathWorks, 2023-10-05:
Using v1.2.1 works! 
Thanks for the info.

-Manjunatha

naushir, 2023-10-05:
Great thanks for confirming!

closed 2023-10-05 by naushir