raspberrypi/rpicam-apps #585: [BUG]

Yolnit, 2023-10-18:
**Describe the bug**
Hello everyone, I'm using a raspberry pi4 model b, it's the first time I'm using raspberry and I'm facing a problem. I created a sh file capable of recording videos as soon as raspberry was powered on. The problem is that I hadn't anticipated how to exit the program... After a sudo reboot of the card and without an sh file, when I launch the program from VS Code I get looping warnings in the terminal:

**Preview window unavailable
`[0:02:41.688285278]` [3480]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+52-a858d20b
[0:02:41.741497758] [3483]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:02:41.743067531] [3483]  WARN RPI vc4.cpp:333 Mismatch between Unicam and CamHelper for embedded data usage!
[0:02:41.743849121] [3483]  INFO RPI vc4.cpp:387 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media0 and ISP device /dev/media1
[0:02:41.744606657] [3480]  INFO Camera camera.cpp:1213 configuring streams: (0) 100x100-YUV420
[0:02:41.745096977] [3483]  INFO RPI vc4.cpp:549 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 640x480-SBGGR10_1X10 - Selected unicam format: 640x480-pBAA**
#0 (0.00 fps) exp 33289.00 ag 8.00 dg 1.00
terminate called after throwing an instance of 'std::runtime_error'
  what():  failed to open output file /media/raspyeaden/CLEF `COCO/videos_pfe/20231018-081114.h264`

The program can't find the video in the folder because it can't save it. Are the libcamera files corrupted? 

I use the raspberry pi v2 camera (the green one, not the black one).

Here are a few command lines that may help you too: 

`libcamera-hello --version
libcamera-apps build: 7e4d3d71867f 22-08-2023 (10:53:13)
libcamera build: `v0.1.0+52-a858d20b`

My config.txt file :

max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

Thank by advance, I'm sorry if the presentation of the issue is not correct, I'll try to do it better next time!

umutonuryasar, 2023-10-18:
The error message "failed to open output file /media/raspyeaden/CLEF/COCO/videos_pfe/20231018-081114.h264" indicates that the program is trying to open a file that does not exist. The libcamera files are not corrupted. The program is simply trying to save the video file to a directory that does not exist.

To fix the bug, you need to create the "/media/raspyeaden/CLEF/COCO/videos_pfe/" directory if it is not exist.

Also make sure that you are running the program with the correct permissions. You may need to run the program as root. You can check the file path where you're trying to save the video. Make sure the directory exists and is writable. Ensure that your program has the necessary permissions to write to the specified directory.

Yolnit, 2023-10-18:
Hello umutonuryasar, the folder exists, I've successfully recorded videos in it several times, the path is always the same from the start:

```
raspyeaden@raspberrypi:~ $ sudo -i
root@raspberrypi:~# cd /media/raspyeaden/CLEF\ COCO/videos_pfe/
root@raspberrypi:/media/raspyeaden/CLEF COCO/videos_pfe# ls -all
total 32
drwxr-xr-x 2 root root  4096 Oct 16 13:36 .
drwxr-xr-x 3 root root  4096 Oct 16 12:12 ..
-rw-r--r-- 1 root root 24576 Oct 16 13:36 20231016-133605.h264
```
However, I've just noticed that there's only one video visible with ls on the raspberry, but when I plug my usb key into my pc, there are 8 videos visible.

When I run libcamera-hello, something seems to record for 5 seconds but I don't see a black square (the video recorded by the camera) on my screen.

Here is some of my code, perhaps to make it easier for you to understand:

```
fileName = time.strftime("%Y%m%d-%H%M%S") + ".h264"
output_path = os.path.join(output_directory, fileName)
os.system('libcamera-vid --width 100 --height 100 -t 5000 -o "{}"'.format(output_path))

# Vérifier si le fichier a été correctement enregistré avant de le convertir
if os.path.exists(output_path):
    GPIO.output(pinLedG, False)
    GPIO.output(pinLedY, True)

    os.system('ffmpeg -i "{}" -c:v copy -an "{}.mp4" -y'.format(output_path, output_path))
    os.remove(output_path)  # Supprimer le fichier .h264 après la conversion
    #videos_cycle(free_space)
    recorded_vid += 1
    GPIO.output(pinLedY, False)
    
    if recorded_vid >= max_videos:
        print("Deux vidéos enregistrées.")
        
        # Attendre que le bouton soit pressé pour enregistrer à nouveau
        while GPIO.input(pinBtn) == GPIO.HIGH:
            pass

        # Attendre que le bouton d'arrêt soit pressé pour arrêter le programme
            while GPIO.input(pinBtnShutdown) == GPIO.HIGH:
                pass
                
            print("Bouton d'arrêt pressé. Arrêt du programme.")
            break  
            
        print("Bouton pressé. Enregistrement de nouvelles vidéos...")
        recorded_vid = 0  # Réinitialiser le compteur de vidéos enregistrées
        time.sleep(1)  # Attendre un court instant pour éviter les faux déclenchements
else:
    print("Erreur lors de l'enregistrement de la vidéo.")
```
`os.system('libcamera-vid --width 100 --height 100 -t 5000 -o "{}"'.format(output_path)) ` is working since the beginning and I don't understand..

Thank by advance!

Yolnit, 2023-10-18:
Ok I have some news ! 

I tried to find the 8 videos that were visible on my PC and not on my raspberry and guess what!

root@raspberrypi:~# cd /media/raspyeaden/
root@raspberrypi:/media/raspyeaden# ls
'CLEF COCO'  'CLEF COCO1'

I only have one plug-in key and yet there are 2 device names. The videos record well on CLE COCO1 but not on CLE COCO. I'll have to delete CLE COCO.

umutonuryasar, 2023-10-18:
It's good to hear! I'm glad you solved the issue.

Yolnit, 2023-10-18:
I would like to understand. I've always had just one folder associated with my CLEF COCO device. How can a second one appear? It's as if the raspberry detected a virtual device with a name to associate in kernel files. Am I mistaken in my reasoning or is this possible?

umutonuryasar, 2023-10-18:
I think your reasoning is correct. If the Raspberry Pi detected a virtual device with a name to associate in kernel files, it is possible for a second folder to appear for the CLEF COCO device. This is because the Raspberry Pi would be treating the virtual device as a separate device.

naushir, 2023-10-18:
I assume this is complete now, so resolving.  Please do re-open if you are still having issues.

closed 2023-10-18 by naushir