raspberrypi/rpicam-apps #591: [BUG] Raspberry Pi 4 not working with Pi Camera Module 3

HasanKhan-1, 2023-11-09:
I have been struggling with interfacing my Raspberry Pi 4 Model B (2019) with my Raspberry Pi Camera Module 3. I initially tried running the software and I am getting this reoccurring issue of:

Made X/EGL preview window
[0:08:37.968677583] [2192] INFO Camera camera_manager.cpp:284 libcamera v0.1.0+99-4a23664b
ERROR: *** no cameras available ***

when trying to run libcamera-hello. It also says that there are no cameras detected at all. I properly inserted the camera in the slot, and have made sure that power is working on it as I have an older camera that shows it is getting power. I tried following tutorials online and they follow steps that that no longer exist on my Raspberry PI OS. Here is the version of Raspberry Pi that I am using:

PRETTY_NAME="Raspbian GNU/Linux 12 (bookworm)"
NAME="Raspbian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

I referred to documentation such as:
[https://support.thepihut.com/hc/en-us/a ... -t-working](https://support.thepihut.com/hc/en-us/articles/360008782198-My-Camera-Module-isn-t-working)

For getting my issue fixed, but it is not working. Additionally, I did a sudo apt update and upgrade, and have tried re-flashing my SD card and rebooting multiple times. Does anyone have an idea of how I could fix this?

HasanKhan-1, 2023-11-09:
Thank you for clarifying this for me. Here is the generated txt  file:

--------------------------------------------------------------------------------
libcamera-apps Bug Report
--------------------------------------------------------------------------------
Date: 09-11-2023 (18:00:14)
Command: /usr/bin/camera-bug-report

--------------------------------------------------------------------------------
Hardware information
--------------------------------------------------------------------------------
** hostname **
raspberrypi

** cat /proc/cpuinfo **
processor	: 0
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Revision	: c03115
Serial		: 100000003803f24f
Model		: Raspberry Pi 4 Model B Rev 1.5

--------------------------------------------------------------------------------
Configuration
--------------------------------------------------------------------------------
** cat /boot/cmdline.txt **
console=serial0,115200 console=tty1 root=PARTUUID=e87f7e2b-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles cfg80211.ieee80211_regdom=CA
** cat /boot/config.txt **
# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

[... 999 lines cut ...]
Versions
--------------------------------------------------------------------------------
** uname -a **
Linux raspberrypi 6.1.61-v8+ #1696 SMP PREEMPT Thu Nov  2 16:44:46 GMT 2023 aarch64 GNU/Linux

** cat /etc/os-release **
PRETTY_NAME="Raspbian GNU/Linux 12 (bookworm)"
NAME="Raspbian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

** vcgencmd version **
Oct 17 2023 15:39:16 
Copyright (c) 2012 Broadcom
version 30f0c5e4d076da3ab4f341d88e7d505760b93ad7 (clean) (release) (start)

** libcamera-hello --version **
libcamera-apps build: f8b05c47894c 23-10-2023 (09:51:59)
libcamera build: v0.1.0+99-4a23664b

naushir, 2023-11-10:
The sensor is not being detected in the initial probe/boot process.  Have you had this particular camera board working in the past?  Can you remove and re-seat all cabling attached to the camera board and camera socket.

HasanKhan-1, 2023-11-10:
Yes, when I first booted up the system and connected the camera for the first time it worked intially. However, after restarting my Raspberry Pi it stopped working. Additionally, I have another camera and it indicates that it has power on the cable so I know that the cable is indeed working. I also repositioned my camera and changed the log file a bit after we had talked. What should I do from here? Could there be a hardware issue? Also, I will send a picture when I get back home. It'll probably be a bit later

naushir, 2023-11-10:
Having a photo of your setup would be useful to check, but it sounds like this camera board may be faulty.  If possibly I would suggest you try and arrange for a replacement from your supplier.

HasanKhan-1, 2023-11-10:
I got it from an official supplier. Do you think they could send me another one? I specifically got it from PiShop Canada.

naushir, 2023-11-10:
I can't say for sure, but definitively get in touch with them and let them know the product may be faulty and request an exchange.

naushir, 2023-11-13:
I'll close this issue down now, but if you have any further problems, please feel free to open a new issue.