raspberrypi/rpicam-apps #592: [BUG]"--tuning-file" parameter is invalidated

tom-ard, 2023-11-10:
"LIBCAMERA_RPI_TUNING_FILE" reads in the libcamera library is a vacant value.
As can be seen from the log, libcamera read "LIBCAMERA_RPI_TUNING_FILE" Before the apps settings.
[image]

Print information in the libcamera library
[image]

Print infomation in the libcamera-apps
[image]

naushir, 2023-11-10:
Thank you for the report.  You are right, a recent change has broken this functionality.  I will send a proposed fix shortly to get this working again.

naushir, 2023-11-10:
This should hopefully be fixed with https://github.com/raspberrypi/libcamera-apps/pull/593.  Would you be able to give this a try and report back?

tom-ard, 2023-11-10:
Okay, I'll start testing now

tom-ard, 2023-11-10:
The problem was solved perfectly, thank you.
[image]

naushir, 2023-11-10:
Thanks for confirming, the change has now been merged.

closed 2023-11-10 by naushir