raspberrypi/rpicam-apps #594: [BUG] libcamera-hello libcamera-hello: symbol lookup error: /lib/aarch64-linux-gnu/libQt5Widgets.so.5: undefined symbol: _ZTVN10__cxzabiv10__si_class_type_infoE, version CXXABI_1.3

cyborgdennett, 2023-11-13:
**Describe the bug**
I reinstalled libcamera-apps, libcamera0 after getting segmentation error.

Now I have the following error

```
$ libcamera-hello
libcamera-hello: symbol lookup error: /lib/aarch64-linux-gnu/libQt5Widgets.so.5: undefined symbol: _ZTVN10__cxzabiv10__si_class_type_infoE, version CXXABI_1.3
```

**Bug report**
Please use the ``camera-bug-report`` tool to create a bug report, and upload it here.

```
camera-bug-report -c libcamera-hello
```
Output: 
```
		-> "bcm2835-codec-image_fx-proc":0 [ENABLED,IMMUTABLE]

- entity 45: bcm2835-codec-image_fx-proc (2 pads, 2 links)
             type Node subtype Unknown flags 0
	pad0: Sink
		<- "bcm2835-codec-image_fx-source":0 [ENABLED,IMMUTABLE]
	pad1: Source
		-> "bcm2835-codec-image_fx-sink":0 [ENABLED,IMMUTABLE]

- entity 48: bcm2835-codec-image_fx-sink (1 pad, 1 link)
             type Node subtype V4L flags 0
             device node name /dev/video18
	pad0: Sink
		<- "bcm2835-codec-image_fx-proc":1 [ENABLED,IMMUTABLE]

- entity 57: bcm2835-codec-encode_image-sour (1 pad, 1 link)
             type Node subtype V4L flags 0
             device node name /dev/video31
	pad0: Source
		-> "bcm2835-codec-encode_image-proc":0 [ENABLED,IMMUTABLE]

- entity 59: bcm2835-codec-encode_image-proc (2 pads, 2 links)
             type Node subtype Unknown flags 0
	pad0: Sink
		<- "bcm2835-codec-encode_image-sour":0 [ENABLED,IMMUTABLE]
	pad1: Source
		-> "bcm2835-codec-encode_image-sink":0 [ENABLED,IMMUTABLE]

- entity 62: bcm2835-codec-encode_image-sink (1 pad, 1 link)
             type Node subtype V4L flags 0
             device node name /dev/video31
	pad0: Sink
		<- "bcm2835-codec-encode_image-proc":1 [ENABLED,IMMUTABLE]

** media-ctl -d 4 -p **
Failed to enumerate /dev/media4 (-2)

--------------------------------------------------------------------------------
Cameras
--------------------------------------------------------------------------------
** libcamera-hello --list-cameras **
libcamera-hello: symbol lookup error: /lib/aarch64-linux-gnu/libQt5Widgets.so.5: undefined symbol: _ZTVN10__cxzabiv10__si_class_type_infoE, version CXXABI_1.3

--------------------------------------------------------------------------------
Versions
--------------------------------------------------------------------------------
** uname -a **
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

** cat /etc/os-release **
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

** vcgencmd version **
Mar 17 2023 10:50:39 
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)

** libcamera-hello --version **
libcamera-hello: symbol lookup error: /lib/aarch64-linux-gnu/libQt5Widgets.so.5: undefined symbol: _ZTVN10__cxzabiv10__si_class_type_infoE, version CXXABI_1.3

--------------------------------------------------------------------------------
libcamera-hello
--------------------------------------------------------------------------------
libcamera-hello: symbol lookup error: /lib/aarch64-linux-gnu/libQt5Widgets.so.5: undefined symbol: _ZTVN10__cxzabiv10__si_class_type_infoE, version CXXABI_1.3
```

**Additional context**
Add any other context about the problem here.

naushir, 2023-11-13:
Can you start at the very beginning.  What caused the original "segmentation error" that caused you to reinstall?  Before reinstalling, did you run a local build of libcamera-apps, or was it a package install?  What commands did you run to reinstall libcamera-apps?

I suspect that your software environment is in a mixed up state because of the reinstall.

cyborgdennett, 2023-11-13:
I used sudo apt remove libcamera-* and sudo apt remove libcamera0. Then I did sudo apt install libcamera0 libcamera-apps

cyborgdennett, 2023-11-13:
I'm not sure what caused the segmentation error before the reinstall

naushir, 2023-11-13:
I would try doing the following:
```
sudo apt purge libcamera*
sudo apt install libcamera-apps -y
```

and see if that fixes it to start with.

naushir, 2023-11-13:
Can you post a full output of the command and the error please?

cyborgdennett, 2023-11-13:
```
0.000000] CPU features: detected: Spectre-v3a
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 1742098
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 983808
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  smsc95xx.macaddr=D8:3A:DD:37:7A:E8 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=b8bafdb5-02 rootfstype=ext4 fsck.repair=yes rootwait modules-load=dwc2,g_ether quiet splash plymouth.ignore-serial-consoles
[    0.000000] Unknown kernel command line parameters "splash modules-load=dwc2,g_ether", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x0000000034000000-0x0000000038000000] (64MB)
[    0.000000] Memory: 3305844K/3997696K available (11776K kernel code, 2106K rwdata, 3688K rodata, 4160K init, 1077K bss, 167564K reserved, 524288K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 39744 entries in 156 pages
[    0.000000] ftrace: allocated 156 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000001] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000325] Console: colour dummy device 80x25
[    0.000388] printk: console [tty1] enabled
[    0.000452] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000478] pid_max: default: 32768 minimum: 301
[    0.000601] LSM: Security Framework initializing
[    0.000807] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.000867] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.002169] cgroup: Disabling memory control group subsystem
[    0.004393] cblist_init_generic: Setting adjustable number of callback queues.
[    0.004411] cblist_init_generic: Setting shift to 2 and lim to 1.
[    0.004591] cblist_init_generic: Setting shift to 2 and lim to 1.
[    0.004768] cblist_init_generic: Setting shift to 2 and lim to 1.
[    0.005225] rcu: Hierarchical SRCU implementation.
[    0.005238] rcu: 	Max phase no-delay instances is 1000.
[    0.006567] EFI services will not be available.
[    0.007078] smp: Bringing up secondary CPUs ...
[    0.008239] Detected PIPT I-cache on CPU1
[    0.008389] CPU1: Booted secondary processor 0x0000000001 [0x410fd083]
[    0.009548] Detected PIPT I-cache on CPU2
[    0.009669] CPU2: Booted secondary processor 0x0000000002 [0x410fd083]
[    0.010804] Detected PIPT I-cache on CPU3
[    0.010929] CPU3: Booted secondary processor 0x0000000003 [0x410fd083]
[    0.011079] smp: Brought up 1 node, 4 CPUs
[    0.011101] SMP: Total of 4 processors activated.
[    0.011114] CPU features: detected: 32-bit EL0 Support
[    0.011124] CPU features: detected: 32-bit EL1 Support
[    0.011136] CPU features: detected: CRC32 instructions
[    0.011273] CPU: All CPU(s) started at EL2
[    0.011297] alternatives: applying system-wide alternatives
[    0.013116] devtmpfs: initialized
[    0.025549] Enabled cp15_barrier support
[    0.025590] Enabled setend support
[    0.025808] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[... 756 lines cut ...]

** cat /etc/os-release **
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

** vcgencmd version **
Mar 17 2023 10:50:39 
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)

** libcamera-hello --version **
Segmentation fault

--------------------------------------------------------------------------------
libcamera-hello
--------------------------------------------------------------------------------

```

cyborgdennett, 2023-11-13:
Command was:
`camera-bug-report -c libcamera-hello`

naushir, 2023-11-13:
Can you provide the output of
```
gdb -ex run --args libcamera-hello --list-cameras
```
This may help see where your segfault is occurring.

cyborgdennett, 2023-11-13:
```
gdb -ex run --args libcamera-hello --list-cameras
GNU gdb (Debian 10.1-1.7) 10.1.90.20210103-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "aarch64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from libcamera-hello...
(No debugging symbols found in libcamera-hello)
Starting program: /usr/bin/libcamera-hello --list-cameras

Program received signal SIGSEGV, Segmentation fault.
0x0000007ff7fd7d88 in elf_machine_rela_relative (reloc_addr_arg=0x80f5643000, reloc=0x7ff5082800, l_addr=549570957312) at ../sysdeps/aarch64/dl-machine.h:381
381	../sysdeps/aarch64/dl-machine.h: No such file or directory.
(gdb) 

```

naushir, 2023-11-13:
In the above ``(gdb)`` prompt, can you type ``bt`` and show the output?

cyborgdennett, 2023-11-13:
```bash
(gdb) info proc all
process 1393
cmdline = '/usr/bin/libcamera-hello --list-cameras'
cwd = '/home/klaas'
exe = '/usr/bin/libcamera-hello'
Mapped address spaces:

          Start Addr           End Addr       Size     Offset objfile
        0x5555550000       0x5555553000     0x3000        0x0 /usr/bin/libcamera-hello
        0x5555562000       0x5555564000     0x2000     0x2000 /usr/bin/libcamera-hello
        0x7fea10c000       0x7fea122000    0x16000        0x0 
        0x7fea122000       0x7fea14f000    0x2d000        0x0 /usr/lib/aarch64-linux-gnu/libtirpc.so.3.0.0
        0x7fea14f000       0x7fea15e000     0xf000    0x2d000 /usr/lib/aarch64-linux-gnu/libtirpc.so.3.0.0
        0x7fea15e000       0x7fea160000     0x2000    0x2c000 /usr/lib/aarch64-linux-gnu/libtirpc.so.3.0.0
        0x7fea160000       0x7fea161000     0x1000    0x2e000 /usr/lib/aarch64-linux-gnu/libtirpc.so.3.0.0
        0x7fea161000       0x7fea165000     0x4000        0x0 /usr/lib/aarch64-linux-gnu/libkeyutils.so.1.9
        0x7fea165000       0x7fea174000     0xf000     0x4000 /usr/lib/aarch64-linux-gnu/libkeyutils.so.1.9
        0x7fea174000       0x7fea175000     0x1000     0x3000 /usr/lib/aarch64-linux-gnu/libkeyutils.so.1.9
        0x7fea175000       0x7fea176000     0x1000     0x4000 /usr/lib/aarch64-linux-gnu/libkeyutils.so.1.9
        0x7fea176000       0x7fea178000     0x2000        0x0 
        0x7fea178000       0x7fea1aa000    0x32000        0x0 /usr/lib/aarch64-linux-gnu/libFLAC.so.8.3.0
        0x7fea1aa000       0x7fea1ba000    0x10000    0x32000 /usr/lib/aarch64-linux-gnu/libFLAC.so.8.3.0
        0x7fea1ba000       0x7fea1bb000     0x1000    0x32000 /usr/lib/aarch64-linux-gnu/libFLAC.so.8.3.0
        0x7fea1bb000       0x7fea1bc000     0x1000    0x33000 /usr/lib/aarch64-linux-gnu/libFLAC.so.8.3.0
        0x7fea1bc000       0x7fea1d1000    0x15000        0x0 /usr/lib/aarch64-linux-gnu/libnsl.so.2.0.1
        0x7fea1d1000       0x7fea1e0000     0xf000    0x15000 /usr/lib/aarch64-linux-gnu/libnsl.so.2.0.1
        0x7fea1e0000       0x7fea1e1000     0x1000    0x14000 /usr/lib/aarch64-linux-gnu/libnsl.so.2.0.1
        0x7fea1e1000       0x7fea1e2000     0x1000    0x15000 /usr/lib/aarch64-linux-gnu/libnsl.so.2.0.1
        0x7fea1e2000       0x7fea1e4000     0x2000        0x0 
        0x7fea1e4000       0x7fea201000    0x1d000        0x0 /usr/lib/aarch64-linux-gnu/liblz4.so.1.9.3
        0x7fea201000       0x7fea211000    0x10000    0x1d000 /usr/lib/aarch64-linux-gnu/liblz4.so.1.9.3
        0x7fea211000       0x7fea212000     0x1000    0x1d000 /usr/lib/aarch64-linux-gnu/liblz4.so.1.9.3
        0x7fea212000       0x7fea213000     0x1000    0x1e000 /usr/lib/aarch64-linux-gnu/liblz4.so.1.9.3
        0x7fea213000       0x7fea36b000   0x158000        0x0 /usr/lib/aarch64-linux-gnu/libgfortran.so.5.0.0
        0x7fea36b000       0x7fea37a000     0xf000   0x158000 /usr/lib/aarch64-linux-gnu/libgfortran.so.5.0.0
        0x7fea37a000       0x7fea37b000     0x1000   0x157000 /usr/lib/aarch64-linux-gnu/libgfortran.so.5.0.0
        0x7fea37b000       0x7fea37d000     0x2000   0x158000 /usr/lib/aarch64-linux-gnu/libgfortran.so.5.0.0
        0x7fea37d000       0x7fea37f000     0x2000        0x0 
        0x7fea37f000       0x7fea5f3000   0x274000        0x0 /usr/lib/aarch64-linux-gnu/libatlas.so.3.10.3
        0x7fea5f3000       0x7fea602000     0xf000   0x274000 /usr/lib/aarch64-linux-gnu/libatlas.so.3.10.3
        0x7fea602000       0x7fea604000     0x2000   0x273000 /usr/lib/aarch64-linux-gnu/libatlas.so.3.10.3
        0x7fea604000       0x7fea608000     0x4000   0x275000 /usr/lib/aarch64-linux-gnu/libatlas.so.3.10.3
        0x7fea608000       0x7fea628000    0x20000        0x0 /usr/lib/aarch64-linux-gnu/libbrotlicommon.so.1.0.9
        0x7fea628000       0x7fea637000     0xf000    0x20000 /usr/lib/aarch64-linux-gnu/libbrotlicommon.so.1.0.9
        0x7fea637000       0x7fea638000     0x1000    0x1f000 /usr/lib/aarch64-linux-gnu/libbrotlicommon.so.1.0.9
        0x7fea638000       0x7fea639000     0x1000    0x20000 /usr/lib/aarch64-linux-gnu/libbrotlicommon.so.1.0.9
        0x7fea639000       0x7fea640000     0x7000        0x0 /usr/lib/aarch64-linux-gnu/libdatrie.so.1.4.0
        0x7fea640000       0x7fea64f000     0xf000     0x7000 /usr/lib/aarch64-linux-gnu/libdatrie.so.1.4.0
        0x7fea64f000       0x7fea650000     0x1000     0x6000 /usr/lib/aarch64-linux-gnu/libdatrie.so.1.4.0
        0x7fea650000       0x7fea651000     0x1000     0x7000 /usr/lib/aarch64-linux-gnu/libdatrie.so.1.4.0
        0x7fea651000       0x7fea6d2000    0x81000        0x0 /usr/lib/aarch64-linux-gnu/libpcre2-8.so.0.10.1
        0x7fea6d2000       0x7fea6e1000     0xf000    0x81000 /usr/lib/aarch64-linux-gnu/libpcre2-8.so.0.10.1
        0x7fea6e1000       0x7fea6e2000     0x1000    0x80000 /usr/lib/aarch64-linux-gnu/libpcre2-8.so.0.10.1
        0x7fea6e2000       0x7fea6e3000     0x1000    0x81000 /usr/lib/aarch64-linux-gnu/libpcre2-8.so.0.10.1
        0x7fea6e3000       0x7fea732000    0x4f000        0x0 /usr/lib/aarch64-linux-gnu/libblkid.so.1.1.0
        0x7fea732000       0x7fea742000    0x10000    0x4f000 /usr/lib/aarch64-linux-gnu/libblkid.so.1.1.0
        0x7fea742000       0x7fea746000     0x4000    0x4f000 /usr/lib/aarch64-linux-gnu/libblkid.so.1.1.0
        0x7fea746000       0x7fea747000     0x1000    0x53000 /usr/lib/aarch64-linux-gnu/libblkid.so.1.1.0
        0x7fea747000       0x7fea749000     0x2000        0x0 
        0x7fea749000       0x7fea755000     0xc000        0x0 /usr/lib/aarch64-linux-gnu/libkrb5support.so.0.1
        0x7fea755000       0x7fea764000     0xf000     0xc000 /usr/lib/aarch64-linux-gnu/libkrb5support.so.0.1
```

naushir, 2023-11-13:
What about ``ldd $(which libcamera-hello)``?

cyborgdennett, 2023-11-13:
```bash
$ ldd $(which libcamera-hello)
	linux-vdso.so.1 (0x0000007f9ee3f000)
	libcamera_app.so.1.2.1 => /lib/aarch64-linux-gnu/libcamera_app.so.1.2.1 (0x0000007f9ed25000)
	libstdc++.so.6 => /lib/aarch64-linux-gnu/libstdc++.so.6 (0x0000007f9eb4d000)
	libgcc_s.so.1 => /lib/aarch64-linux-gnu/libgcc_s.so.1 (0x0000007f9eb29000)
	libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000007f9e9b5000)
	/lib/ld-linux-aarch64.so.1 (0x0000007f9ee0f000)
	libcamera.so.0.0 => /lib/aarch64-linux-gnu/libcamera.so.0.0 (0x0000007f9e840000)
	libcamera-base.so.0.0 => /lib/aarch64-linux-gnu/libcamera-base.so.0.0 (0x0000007f9e80b000)
	libboost_program_options.so.1.74.0 => /lib/aarch64-linux-gnu/libboost_program_options.so.1.74.0 (0x0000007f9e796000)
	libavcodec.so.58 => /lib/aarch64-linux-gnu/libavcodec.so.58 (0x0000007f9d411000)
	libavdevice.so.58 => /lib/aarch64-linux-gnu/libavdevice.so.58 (0x0000007f9d3cb000)
	libavformat.so.58 => /lib/aarch64-linux-gnu/libavformat.so.58 (0x0000007f9d13e000)
	libavutil.so.56 => /lib/aarch64-linux-gnu/libavutil.so.56 (0x0000007f9ce7e000)
	libswresample.so.3 => /lib/aarch64-linux-gnu/libswresample.so.3 (0x0000007f9ce56000)
	libexif.so.12 => /lib/aarch64-linux-gnu/libexif.so.12 (0x0000007f9cdfd000)
	libjpeg.so.62 => /lib/aarch64-linux-gnu/libjpeg.so.62 (0x0000007f9cdad000)
	libtiff.so.5 => /lib/aarch64-linux-gnu/libtiff.so.5 (0x0000007f9cd1d000)
	libpng16.so.16 => /lib/aarch64-linux-gnu/libpng16.so.16 (0x0000007f9ccd6000)
	libdrm.so.2 => /lib/aarch64-linux-gnu/libdrm.so.2 (0x0000007f9ccb3000)
	libX11.so.6 => /lib/aarch64-linux-gnu/libX11.so.6 (0x0000007f9cb64000)
	libepoxy.so.0 => /lib/aarch64-linux-gnu/libepoxy.so.0 (0x0000007f9ca0e000)
	libQt5Core.so.5 => /lib/aarch64-linux-gnu/libQt5Core.so.5 (0x0000007f9c4a6000)
	libQt5Widgets.so.5 => /lib/aarch64-linux-gnu/libQt5Widgets.so.5 (0x0000007f9bde4000)
	libQt5Gui.so.5 => /lib/aarch64-linux-gnu/libQt5Gui.so.5 (0x0000007f9b755000)
	libm.so.6 => /lib/aarch64-linux-gnu/libm.so.6 (0x0000007f9b6aa000)
	libpthread.so.0 => /lib/aarch64-linux-gnu/libpthread.so.0 (0x0000007f9b679000)
	libgnutls.so.30 => /lib/aarch64-linux-gnu/libgnutls.so.30 (0x0000007f9b45d000)
	libdl.so.2 => /lib/aarch64-linux-gnu/libdl.so.2 (0x0000007f9b449000)
	libudev.so.1 => /lib/aarch64-linux-gnu/libudev.so.1 (0x0000007f9b412000)
	libyaml-0.so.2 => /lib/aarch64-linux-gnu/libyaml-0.so.2 (0x0000007f9b3e6000)
	libdw.so.1 => /lib/aarch64-linux-gnu/libdw.so.1 (0x0000007f9b335000)
	libunwind.so.8 => /lib/aarch64-linux-gnu/libunwind.so.8 (0x0000007f9b2f6000)
	libvpx.so.6 => /lib/aarch64-linux-gnu/libvpx.so.6 (0x0000007f9b148000)
	libwebpmux.so.3 => /lib/aarch64-linux-gnu/libwebpmux.so.3 (0x0000007f9b12f000)
	libwebp.so.6 => /lib/aarch64-linux-gnu/libwebp.so.6 (0x0000007f9b0cd000)
	liblzma.so.5 => /lib/aarch64-linux-gnu/liblzma.so.5 (0x0000007f9b095000)
	libdav1d.so.4 => /lib/aarch64-linux-gnu/libdav1d.so.4 (0x0000007f9af9d000)
	librsvg-2.so.2 => /lib/aarch64-linux-gnu/librsvg-2.so.2 (0x0000007f9a6a2000)
	libgobject-2.0.so.0 => /lib/aarch64-linux-gnu/libgobject-2.0.so.0 (0x0000007f9a635000)
	libglib-2.0.so.0 => /lib/aarch64-linux-gnu/libglib-2.0.so.0 (0x0000007f9a4f2000)
	libcairo.so.2 => /lib/aarch64-linux-gnu/libcairo.so.2 (0x0000007f9a3cc000)
	libzvbi.so.0 => /lib/aarch64-linux-gnu/libzvbi.so.0 (0x0000007f9a330000)
	libz.so.1 => /lib/aarch64-linux-gnu/libz.so.1 (0x0000007f9a306000)
	libsnappy.so.1 => /lib/aarch64-linux-gnu/libsnappy.so.1 (0x0000007f9a2ed000)
	libaom.so.0 => /lib/aarch64-linux-gnu/libaom.so.0 (0x0000007f99fdf000)
	libcodec2.so.0.9 => /lib/aarch64-linux-gnu/libcodec2.so.0.9 (0x0000007f991f1000)
	libgsm.so.1 => /lib/aarch64-linux-gnu/libgsm.so.1 (0x0000007f991d6000)
	libmp3lame.so.0 => /lib/aarch64-linux-gnu/libmp3lame.so.0 (0x0000007f99154000)
	libopenjp2.so.7 => /lib/aarch64-linux-gnu/libopenjp2.so.7 (0x0000007f990ee000)
	libopus.so.0 => /lib/aarch64-linux-gnu/libopus.so.0 (0x0000007f99090000)
	libshine.so.3 => /lib/aarch64-linux-gnu/libshine.so.3 (0x0000007f99076000)
	libspeex.so.1 => /lib/aarch64-linux-gnu/libspeex.so.1 (0x0000007f9904f000)
	libtheoraenc.so.1 => /lib/aarch64-linux-gnu/libtheoraenc.so.1 (0x0000007f9900e000)
	libtheoradec.so.1 => /lib/aarch64-linux-gnu/libtheoradec.so.1 (0x0000007f98fe3000)
	libtwolame.so.0 => /lib/aarch64-linux-gnu/libtwolame.so.0 (0x0000007f98fb1000)
	libvorbis.so.0 => /lib/aarch64-linux-gnu/libvorbis.so.0 (0x0000007f98f78000)
	libvorbisenc.so.2 => /lib/aarch64-linux-gnu/libvorbisenc.so.2 (0x0000007f98ec8000)
	libwavpack.so.1 => /lib/aarch64-linux-gnu/libwavpack.so.1 (0x0000007f98e92000)
	libx264.so.160 => /lib/aarch64-linux-gnu/libx264.so.160 (0x0000007f98c42000)
	libx265.so.192 => /lib/aarch64-linux-gnu/libx265.so.192 (0x0000007f98970000)
	libxvidcore.so.4 => /lib/aarch64-linux-gnu/libxvidcore.so.4 (0x0000007f9887f000)
	libva.so.2 => /lib/aarch64-linux-gnu/libva.so.2 (0x0000007f98844000)
	libavfilter.so.7 => /lib/aarch64-linux-gnu/libavfilter.so.7 (0x0000007f98475000)
	libraw1394.so.11 => /lib/aarch64-linux-gnu/libraw1394.so.11 (0x0000007f98455000)
	libavc1394.so.0 => /lib/aarch64-linux-gnu/libavc1394.so.0 (0x0000007f9843e000)
	librom1394.so.0 => /lib/aarch64-linux-gnu/librom1394.so.0 (0x0000007f9842a000)
	libiec61883.so.0 => /lib/aarch64-linux-gnu/libiec61883.so.0 (0x0000007f9840d000)
	libjack.so.0 => /lib/aarch64-linux-gnu/libjack.so.0 (0x0000007f983a9000)
[... 133 lines cut ...]
	libfftw3.so.3 => /lib/aarch64-linux-gnu/libfftw3.so.3 (0x0000007f9197b000)
[... cut ...]
	libatlas.so.3 => /lib/aarch64-linux-gnu/libatlas.so.3 (0x0000007f911ae000)
	libgfortran.so.5 => /lib/aarch64-linux-gnu/libgfortran.so.5 (0x0000007f91042000)
	liblz4.so.1 => /lib/aarch64-linux-gnu/liblz4.so.1 (0x0000007f91013000)
	libnsl.so.2 => /lib/aarch64-linux-gnu/libnsl.so.2 (0x0000007f90feb000)
	libFLAC.so.8 => /lib/aarch64-linux-gnu/libFLAC.so.8 (0x0000007f90fa7000)
	libkeyutils.so.1 => /lib/aarch64-linux-gnu/libkeyutils.so.1 (0x0000007f90f90000)
	libtirpc.so.3 => /lib/aarch64-linux-gnu/libtirpc.so.3 (0x0000007f90f51000)
```

naushir, 2023-11-13:
Unfortunately I just cannot tell what is wrong here, and I'd only be guessing at this point.

I would suggest start with another fresh SDcard image and work your way back up to your configuration, ensuring the camera is operational with each step.

naushir, 2023-11-13:
Sorry about not being able to get a solution, these SW dependency breakages are horribly difficult to debug, particularly being remote.

cyborgdennett, 2023-11-13:
When I do `cheese` a small webcam app I can see the camera

cyborgdennett, 2023-11-13:
[image]

```bash
cheese
Xlib: sequence lost (0x100b0 > 0xb2) in reply type 0x0!
[0:27:24.943551536] [1454]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f
[0:27:24.987675128] [1462]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media4 and ISP device /dev/media1
[0:27:25.117794888] [1454]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f
[0:27:25.158513665] [1466]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media4 and ISP device /dev/media1
[0:27:25.225020795] [1477]  INFO Camera camera.cpp:1033 configuring streams: (0) 1024x768-NV21
[0:27:25.225800776] [1466]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
[0:27:25.389788443] [1477]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f
[0:27:25.427672202] [1479]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media4 and ISP device /dev/media1
[0:27:25.456733073] [1476]  INFO Camera camera.cpp:1033 configuring streams: (0) 1024x768-NV21
[0:27:25.457610610] [1479]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA

```

naushir, 2023-11-13:
This seems to suggest ``libcamera-apps`` is the culprit and not our libcamera packages.  You could try doing a custom compilation of libcamera-apps and see if that fixes your issue.  But doing so might also cause you to get more SW incompatibility.

naushir, 2023-11-16:
@cyborgdennett did you have any further updates on this issue?  Did a OS reinstall work for you?

cyborgdennett, 2023-11-16:
I used another way to record video since that worked.

naushir, 2023-11-16:
Thanks for the update!  I'll close this now, and if you have any similar issues in the future, feel free to open a new issue.

closed 2023-11-16 by naushir