raspberrypi/rpicam-apps #595: resolution vs. recorded area at imx477

LeifSec, 2023-11-14:
When using e.g. imx708 camera setting the resolution (e.g. via  `libcamera-vid --width 640 --height 480` or using the PiCamera2 interface and `configuration["main"]["size"] result in a video (stream) of full sensor area and the set resolution.

But when using camera imx477 results in a video of just an area of the set resolution.

naushir, 2023-11-14:
This is entirely dependent on how the sensor mode is configured in the kernel driver.  Could you provide exactly what you are trying to achieve, and what command you are using?

LeifSec, 2023-11-16:
about the camer amodule:
`libcamera-hello` 
```
[0:24:26.736960070] [2028]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+52-a858d20b
[0:24:26.898406297] [2031]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:24:26.908213868] [2031]  INFO RPI vc4.cpp:387 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media0 and ISP device /dev/media1
Stream configuration adjusted
[0:24:26.910858234] [2028]  INFO Camera camera.cpp:1213 configuring streams: (0) 2028x1520-YUV420
[0:24:26.911883127] [2031]  INFO RPI vc4.cpp:549 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
#0 (0.00 fps) exp 1911.00 ag 1.00 dg 1.00
#1 (30.01 fps) exp 1638.00 ag 1.00 dg 1.00
...
``` 
record a video 
`libcamera-vid --width 640 --height 480`
``` 
[0:30:07.943977722] [2067]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+52-a858d20b
[0:30:08.105135513] [2070]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:30:08.115136521] [2070]  INFO RPI vc4.cpp:387 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media0 and ISP device /dev/media1
[0:30:08.117394742] [2067]  INFO Camera camera.cpp:1213 configuring streams: (0) 640x480-YUV420
[0:30:08.118688853] [2070]  INFO RPI vc4.cpp:549 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
#0 (0.00 fps) exp 1820.00 ag 1.00 dg 1.00
#1 (30.01 fps) exp 1501.00 ag 1.00 dg 1.00
...
```
For  `imx708` this records a video of full sensors size with resolution 640x480. But for `imx477`  only a area of 640x480 is recorded.

naushir, 2023-11-16:
In both cases above, you are using the 2028x1520 full FoV sensor mode.  If you take a full resolution still capture with ``libcamera-still``, you ought to get the same field of view.  Can you confirm?

Could it be that the field of view difference you are seeing between the imx708 and imx477 is purely down to the module lenses you are using?

LeifSec, 2023-11-16:
`libcamera-still` is also for `imx477` generating pictures with full sensor area.
But `libcamera-vid`  (also `libcamera-vid --rawfull=1`) is generating videos with only one half (horizontal) of the sensor area.

naushir, 2023-11-17:
Can you share an image to show the difference between libcamera-vid and libcamera-still?  Please also include the command line used, and the console output generated.

LeifSec, 2023-11-17:
`libcamera-still --raw -o libcamera_still_raw_imx477.jpg`
```
Preview window unavailable
[0:03:51.751105794] [1470]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+52-a858d20b
[0:03:51.910695560] [1473]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:03:51.920999298] [1473]  INFO RPI vc4.cpp:387 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media0
Stream configuration adjusted
[0:03:51.924194659] [1470]  INFO Camera camera.cpp:1213 configuring streams: (0) 2028x1520-YUV420
[0:03:51.925615178] [1473]  INFO RPI vc4.cpp:549 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
#0 (0.00 fps) exp 29999.00 ag 3.52 dg 1.22
#1 (30.01 fps) exp 29999.00 ag 3.53 dg 1.21
#2 (30.01 fps) exp 29999.00 ag 3.52 dg 1.22
#3 (30.01 fps) exp 29999.00 ag 3.53 dg 1.21
#4 (30.01 fps) exp 32138.00 ag 4.00 dg 1.00
#5 (30.01 fps) exp 32168.00 ag 4.00 dg 1.00
#6 (30.01 fps) exp 32123.00 ag 4.00 dg 1.00
#7 (30.01 fps) exp 32168.00 ag 4.00 dg 1.00
#8 (30.01 fps) exp 32138.00 ag 4.00 dg 1.00
#9 (30.01 fps) exp 32183.00 ag 4.00 dg 1.00
#10 (30.01 fps) exp 32214.00 ag 4.00 dg 1.00
#11 (30.01 fps) exp 32214.00 ag 4.00 dg 1.00
#12 (30.01 fps) exp 32214.00 ag 4.00 dg 1.00
#13 (30.01 fps) exp 32229.00 ag 4.00 dg 1.00
#14 (30.01 fps) exp 32229.00 ag 4.00 dg 1.00
#15 (30.01 fps) exp 32229.00 ag 4.00 dg 1.00
#16 (30.01 fps) exp 32229.00 ag 4.00 dg 1.00
#17 (30.01 fps) exp 32229.00 ag 4.00 dg 1.00
#18 (30.01 fps) exp 32229.00 ag 4.00 dg 1.00
#19 (30.01 fps) exp 32229.00 ag 4.00 dg 1.00
#20 (30.01 fps) exp 32214.00 ag 4.00 dg 1.00
#21 (30.00 fps) exp 32214.00 ag 4.00 dg 1.00
#22 (30.01 fps) exp 32214.00 ag 4.00 dg 1.00
#23 (30.02 fps) exp 32214.00 ag 4.00 dg 1.00
#24 (30.01 fps) exp 32214.00 ag 4.00 dg 1.00
#25 (30.01 fps) exp 32229.00 ag 4.00 dg 1.00
#26 (30.01 fps) exp 32229.00 ag 4.00 dg 1.00
#27 (30.01 fps) exp 32229.00 ag 4.00 dg 1.00
#28 (30.01 fps) exp 32229.00 ag 4.00 dg 1.00
#29 (30.01 fps) exp 32229.00 ag 4.00 dg 1.00
#30 (29.99 fps) exp 32229.00 ag 4.00 dg 1.00
#31 (30.03 fps) exp 32229.00 ag 4.00 dg 1.00
#32 (30.01 fps) exp 32214.00 ag 4.00 dg 1.00
#33 (30.01 fps) exp 32214.00 ag 4.00 dg 1.00
#34 (30.01 fps) exp 32198.00 ag 4.00 dg 1.00
#35 (30.01 fps) exp 32198.00 ag 4.00 dg 1.00
#36 (30.01 fps) exp 32183.00 ag 4.00 dg 1.00
#37 (30.01 fps) exp 32183.00 ag 4.00 dg 1.00
#38 (30.01 fps) exp 32183.00 ag 4.00 dg 1.00
#39 (30.01 fps) exp 32168.00 ag 4.00 dg 1.00
#40 (30.01 fps) exp 32183.00 ag 4.00 dg 1.00
#41 (30.01 fps) exp 32183.00 ag 4.00 dg 1.00
#42 (30.01 fps) exp 32183.00 ag 4.00 dg 1.00
#43 (30.01 fps) exp 32183.00 ag 4.00 dg 1.00
#44 (30.01 fps) exp 32183.00 ag 4.00 dg 1.00
#45 (30.01 fps) exp 32183.00 ag 4.00 dg 1.00
#46 (30.01 fps) exp 32183.00 ag 4.00 dg 1.00
#47 (30.01 fps) exp 32183.00 ag 4.00 dg 1.00
#48 (30.01 fps) exp 32183.00 ag 4.00 dg 1.00
#49 (30.01 fps) exp 32183.00 ag 4.00 dg 1.00
#50 (30.01 fps) exp 32183.00 ag 4.00 dg 1.00
#51 (30.01 fps) exp 32183.00 ag 4.00 dg 1.00
#52 (30.01 fps) exp 32183.00 ag 4.00 dg 1.00
#53 (30.01 fps) exp 32198.00 ag 4.00 dg 1.00
#54 (30.01 fps) exp 32183.00 ag 4.00 dg 1.00
#55 (30.01 fps) exp 32198.00 ag 4.00 dg 1.00
#56 (30.01 fps) exp 32198.00 ag 4.00 dg 1.00
#57 (30.01 fps) exp 32198.00 ag 4.00 dg 1.00
#58 (30.01 fps) exp 32198.00 ag 4.00 dg 1.00
#59 (30.01 fps) exp 32198.00 ag 4.00 dg 1.00
#60 (30.01 fps) exp 32198.00 ag 4.00 dg 1.00
[... 395 lines cut ...]
#121 (30.01 fps) exp 29999.00 ag 3.81 dg 1.00
#122 (30.01 fps) exp 29999.00 ag 3.81 dg 1.00
#123 (30.01 fps) exp 29999.00 ag 3.81 dg 1.00
#124 (30.01 fps) exp 29999.00 ag 3.81 dg 1.00
#125 (30.01 fps) exp 29999.00 ag 3.81 dg 1.00
#126 (30.01 fps) exp 29999.00 ag 3.81 dg 1.00
#127 (30.01 fps) exp 29999.00 ag 3.81 dg 1.00
#128 (30.01 fps) exp 29999.00 ag 3.81 dg 1.00
#129 (30.01 fps) exp 29999.00 ag 3.81 dg 1.00
#130 (29.99 fps) exp 29999.00 ag 3.81 dg 1.00
#131 (30.04 fps) exp 29999.00 ag 3.81 dg 1.00
#132 (30.01 fps) exp 29999.00 ag 3.81 dg 1.00
#133 (30.01 fps) exp 29999.00 ag 3.81 dg 1.00
#134 (30.01 fps) exp 29999.00 ag 3.81 dg 1.00
#135 (30.01 fps) exp 29999.00 ag 3.81 dg 1.00
#136 (30.01 fps) exp 29999.00 ag 3.81 dg 1.00
#137 (30.01 fps) exp 29999.00 ag 3.81 dg 1.00
#138 (30.01 fps) exp 29999.00 ag 3.81 dg 1.00
#139 (30.01 fps) exp 29999.00 ag 3.81 dg 1.00
#140 (30.01 fps) exp 29999.00 ag 3.81 dg 1.00
#141 (30.01 fps) exp 29999.00 ag 3.81 dg 1.00
Halting: reached timeout of 5000 milliseconds.
```
Video resolution 640x480
[image]

naushir, 2023-11-17:
Thanks, I will look into this and see what I can find.

naushir, 2023-11-17:
One quick thought that occurred, this might be actually caused by a bug that was fixed with https://github.com/raspberrypi/linux/pull/5685.

Are you able to rebuild the up-to-date kernel and try this out again?  If not, you can do ``sudo rpi-update pulls/5685``.  However, please only do this if you are OK with running a different kernel temporarily and unrelated things may go wrong!

naushir, 2023-11-17:
Thanks for confirming the fix.  Closing now.

closed 2023-11-17 by naushir