raspberrypi/rpicam-apps #598: [BUG] Camera not loading in properly when using openCV

HasanKhan-1, 2023-11-16:
**Describe the bug**
When I try to run the following script:

import cv2
video_capture = cv2.VideoCapture(0)

cv2.namedWindow("Window")

while True:
    ret, frame = video_capture.read()
    cv2.imshow("Window", frame)

    #This breaks on 'q' key
    if cv2.waitKey(1) & 0xFF == ord('q'):
        break

video_capture.release()
cv2.destroyAllWindows()

I encounter this error:

/home/hasan_khan/project/test.py
[ WARN:0@10.199] global cap_v4l.cpp:1119 tryIoctl VIDEOIO(V4L2:/dev/video0): select() timeout.
Traceback (most recent call last):
  File "/home/hasan_khan/project/test.py", line 8, in <module>
    cv2.imshow("Window", frame)
cv2.error: OpenCV(4.8.1) /tmp/pip-install-5evo06li/opencv-python_61108a0d5bc541509a210d902bbc3879/opencv/modules/highgui/src/window.cpp:971: error: (-215:Assertion failed) size.width>0 && size.height>0 in function 'imshow'

I went on multiple forums online such as:

https://stackoverflow.com/questions/27953069/opencv-error-215size-width0-size-height0-in-function-imshow 

I have been trying to get this working since I last asked a question on this forum. After getting a new camera I don't understand why this issue is reoccurring. When running libcamera-hello the command works, and I get a screen. However, it does not work with OpenCV. How can I resolved this issue?

HasanKhan-1, 2023-11-16:
For reference here is my bug report:

**Bug report**
--------------------------------------------------------------------------------
libcamera-apps Bug Report
--------------------------------------------------------------------------------
Date: 15-11-2023 (22:04:03)
Command: /usr/bin/camera-bug-report

--------------------------------------------------------------------------------
Hardware information
--------------------------------------------------------------------------------
** hostname **
raspberrypi

** cat /proc/cpuinfo **
processor	: 0
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2835
Revision	: c03115
Serial		: 100000003803f24f
Model		: Raspberry Pi 4 Model B Rev 1.5

--------------------------------------------------------------------------------
Configuration
--------------------------------------------------------------------------------
** cat /boot/cmdline.txt **
console=serial0,115200 console=tty1 root=PARTUUID=ed6bfd88-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles cfg80211.ieee80211_regdom=CA
** cat /boot/config.txt **
# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
[... 2123 lines cut ...]
Versions
--------------------------------------------------------------------------------
** uname -a **
Linux raspberrypi 6.1.0-rpi6-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.58-1+rpt2 (2023-10-27) aarch64 GNU/Linux

** cat /etc/os-release **
PRETTY_NAME="Raspbian GNU/Linux 12 (bookworm)"
NAME="Raspbian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

** vcgencmd version **
Oct 17 2023 15:39:16 
Copyright (c) 2012 Broadcom
version 30f0c5e4d076da3ab4f341d88e7d505760b93ad7 (clean) (release) (start)

** libcamera-hello --version **
libcamera-apps build: f8b05c47894c 23-10-2023 (09:51:59)
libcamera build: v0.1.0+99-4a23664b

6by9, 2023-11-16:
OpenCV does not natively support libcamera - https://github.com/opencv/opencv/issues/21653

Please look at the picamera2 examples for using libcamera with OpenCV.

naushir, 2023-11-16:
Nothing more to add to @6by9's comment above.  You can find the picamera2 examples at https://github.com/raspberrypi/picamera2/tree/main/examples.

closed 2023-11-16 by naushir