raspberrypi/rpicam-apps #600: Libcamera not starting

Sdb-sign, 2023-11-17:
[Jump to bottom](https://github.com/libcamera-org/libcamera/issues/10#issue-comment-box)
Libcamera not starting #10
Closed
[Sdb-sign](https://github.com/Sdb-sign) opened this issue 2 hours ago · 1 comment
Closed
[Libcamera not starting](https://github.com/libcamera-org/libcamera/issues/10#top)
#10
[Sdb-sign](https://github.com/Sdb-sign) opened this issue 2 hours ago · 1 comment
Comments
[Sdb-sign](https://github.com/Sdb-sign) commented [2 hours ago](https://github.com/libcamera-org/libcamera/issues/10#issue-1999696932)
Hi,

I have tried to install this on my device, but when I test it I get the following message. I am using a Raspberry Pi 5, and get the same issue with the camera modules 2 and 3.

:~ $ libcamera-hello
WARNING: v3d support for hw version 71 is neither a complete nor a conformant OpenGL implementation. Testing use only.
Made X/EGL preview window
[0:01:19.660566483] [2039] INFO Camera camera_manager.cpp:284 libcamera v0.1.0+99-4a23664b
[0:01:19.678918830] [2043] INFO RPI pisp.cpp:652 libpisp version v1.0.0 ce5624ed8726 23-10-2023 (07:46:39)
[0:01:19.700389487] [2043] INFO RPI pisp.cpp:1106 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx219@10 to CFE device /dev/media0 and ISP device /dev/media1 using PiSP variant BCM2712_C0
Mode selection for 1640:1232:12:P
SRGGB10_CSI2P,640x480/0 - Score: 4504.81
SRGGB10_CSI2P,1640x1232/0 - Score: 1000
SRGGB10_CSI2P,1920x1080/0 - Score: 1541.48
SRGGB10_CSI2P,3280x2464/0 - Score: 1718
SRGGB8,640x480/0 - Score: 5504.81
SRGGB8,1640x1232/0 - Score: 2000
SRGGB8,1920x1080/0 - Score: 2541.48
SRGGB8,3280x2464/0 - Score: 2718
Stream configuration adjusted
[0:01:19.701359773] [2039] INFO Camera camera.cpp:1181 configuring streams: (0) 1640x1232-YUV420 (1) 1640x1232-BGGR16_PISP_COMP1
[0:01:19.701486304] [2043] INFO RPI pisp.cpp:1390 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected CFE format: 1640x1232-PC1B
[0:01:19.716245638] [2043] ERROR V4L2 v4l2_videodevice.cpp:1906 /dev/video4[20:cap]: Failed to start streaming: Invalid argument
ERROR: *** failed to start camera ***

suosuo6, 2023-11-18:
Hi, I have the same question, too,
raspi 5 bookworm ERROR V4L2 v4l2_videodevice.cpp:1906 /dev/video6[18:cap]: Failed to start streaming: Input/output error
ERROR: *** failed to start camera ***

naushir, 2023-11-20:
This looks to be caused by an incompatibility with your kernel and userland code.  Can you update your kernel with ``sudo rpi-update`` and try this again?  Note that you should only do this if you are ok with installing the latest kernel on your platform.  Please do not do this if you have critical services running on your board.

Kyle0122, 2023-11-21:
@naushir 
Hi. I encountered this issue on Ubuntu 23.10, Raspberry Pi 5. The libcamera and rpicam-apps are compiled today.

>rpicam-jpeg -o uu.jpg --tuning-file ~/Project/libcamera/src/ipa/rpi/pisp/data/imx219_noir.json -n
[0:08:49.803890642] [3378]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+117-99b177b3
[0:08:49.836920863] [3381]  INFO RPI pisp.cpp:653 libpisp version v1.0.1 5652744a74d7 21-11-2023 (18:34:46)
[0:08:49.879383025] [3381]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx290@1a to CFE device /dev/media0 and ISP device /dev/media1 using PiSP variant BCM2712_C0
Mode selection for 960:540:12:P
    SRGGB10_CSI2P,1280x720/0 - Score: 1125
    SRGGB10_CSI2P,1920x1080/0 - Score: 1375
    SRGGB12_CSI2P,1280x720/0 - Score: 125
    SRGGB12_CSI2P,1920x1080/0 - Score: 375
Stream configuration adjusted
[0:08:49.880555138] [3378]  INFO Camera camera.cpp:1183 configuring streams: (0) 960x540-YUV420 (1) 1280x720-RGGB16_PISP_COMP1
[0:08:49.880741148] [3381]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/imx290@1a - Selected sensor format: 1280x720-SRGGB12_1X12 - Selected CFE format: 1280x720-PC1R
[0:08:49.889847253] [3381] ERROR V4L2 v4l2_videodevice.cpp:1906 /dev/video4[16:cap]: Failed to start streaming: Invalid argument
ERROR: *** failed to start camera ***

naushir, 2023-11-21:
Have you tried running ``sudo rpi-update`` and tested again?

Kyle0122, 2023-11-21:
Well, there are no rpi-update on ubuntu.
I tried 'apt search raspberry' and there are three image listed: 
linux-image-6.5.0-1005-raspi linux-image-6.5.0-1006-raspi and linux-image-6.5.0-1007-raspi. I checked that I'm running version 1007.

naushir, 2023-11-21:
In that case, I suggest getting in touch with their support team and ensure they have our latest downstream kernel changes.  This will avoid a kernel/userland mismatch that is likely causing your issue.

suosuo6, 2023-11-21:
@naushir Hi ,I executed it sudo rpi-update and sudo apt full-upgrade,It still doesn't work on Raspberry Pi 5, it's too difficult. This camera works normally on my Zero 2w bullet. Here are the error prompts on Raspberry Pi 5
libcamera-jpeg -o test.jpg
Made X/EGL preview window
[0:00:53.374651940] [1827]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+99-4a23664b
[0:00:53.393222815] [1834]  INFO RPI pisp.cpp:652 libpisp version v1.0.0 ce5624ed8726 23-10-2023 (07:41:40)
[0:00:53.421229288] [1834]  INFO RPI pisp.cpp:1106 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx219@10 to CFE device /dev/media2 and ISP device /dev/media0 using PiSP variant BCM2712_C0
[0:00:53.422362465] [1827]  WARN V4L2 v4l2_pixelformat.cpp:336 Unsupported V4L2 pixel format RPBP
Mode selection for 1640:1232:12:P
    SRGGB10_CSI2P,640x480/0 - Score: 4504.81
    SRGGB10_CSI2P,1640x1232/0 - Score: 1000
    SRGGB10_CSI2P,1920x1080/0 - Score: 1541.48
    SRGGB10_CSI2P,3280x2464/0 - Score: 1718
    SRGGB8,640x480/0 - Score: 5504.81
    SRGGB8,1640x1232/0 - Score: 2000
    SRGGB8,1920x1080/0 - Score: 2541.48
    SRGGB8,3280x2464/0 - Score: 2718
Stream configuration adjusted
[0:00:53.422865055] [1827]  INFO Camera camera.cpp:1181 configuring streams: (0) 1640x1232-YUV420 (1) 1640x1232-BGGR16_PISP_COMP1
[0:00:53.423013637] [1834]  INFO RPI pisp.cpp:1390 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected CFE format: 1640x1232-PC1B
[0:00:53.468041213] [1834] ERROR V4L2 v4l2_videodevice.cpp:1906 /dev/video8[22:cap]: Failed to start streaming: Input/output error
ERROR: *** failed to start camera ***

dmesg
 rp1-cfe 1f00128000.csi: Using a link rate of 912 Mbps
[   53.760187] i2c_designware 1f00080000.i2c: timeout waiting for bus ready
[   53.760205] imx219 4-0010: Failed to write reg 0x30eb. error = -5
[   53.760208] imx219 4-0010: imx219_start_streaming failed to send mfg header
[   53.760231] rp1-cfe 1f00128000.csi: stream on failed in subdev
[   55.019358] i2c_designware 1f00080000.i2c: timeout in disabling adapter

naushir, 2023-11-21:
@suosuo6 I don't think your issue is related to this at all.  

```
dmesg
rp1-cfe 1f00128000.csi: Using a link rate of 912 Mbps
[ 53.760187] i2c_designware 1f00080000.i2c: timeout waiting for bus ready
[ 53.760205] imx219 4-0010: Failed to write reg 0x30eb. error = -5
[ 53.760208] imx219 4-0010: imx219_start_streaming failed to send mfg header
[ 53.760231] rp1-cfe 1f00128000.csi: stream on failed in subdev
[ 55.019358] i2c_designware 1f00080000.i2c: timeout in disabling adapter
```

This is an I2C transaction error, so the Pi has failed to talk to the sensor entirely.  This might be due to an incorrectly seated cable, or faulty hardware.

suosuo6, 2023-11-21:
It is difficult to run on bookworm, and using bookworm with zero2w will also result in the same error, which is not present in bulleye

Sdb-sign, 2023-11-21:
Hi,

I just managed to get it to work on an Rpi 5 running Raspian. I did an update, which didnt work, then sudo rpi-update and reboot, which seemed to work. I did get the following with a Warn.

sb@raspberrypi:~ $ libcamera-hello
Made X/EGL preview window
[0:00:20.324398765] [1816]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+99-4a23664b
[0:00:20.342378228] [1830]  INFO RPI pisp.cpp:652 libpisp version v1.0.0 ce5624ed8726 23-10-2023 (07:46:39)
[0:00:20.366963765] [1830]  INFO RPI pisp.cpp:1106 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx219@10 to CFE device /dev/media3 and ISP device /dev/media0 using PiSP variant BCM2712_C0
[0:00:20.367462172] [1816]  WARN V4L2 v4l2_pixelformat.cpp:336 Unsupported V4L2 pixel format RPBP
Mode selection for 1640:1232:12:P
    SRGGB10_CSI2P,640x480/0 - Score: 4504.81
    SRGGB10_CSI2P,1640x1232/0 - Score: 1000
    SRGGB10_CSI2P,1920x1080/0 - Score: 1541.48
    SRGGB10_CSI2P,3280x2464/0 - Score: 1718
    SRGGB8,640x480/0 - Score: 5504.81
    SRGGB8,1640x1232/0 - Score: 2000
    SRGGB8,1920x1080/0 - Score: 2541.48
    SRGGB8,3280x2464/0 - Score: 2718
Stream configuration adjusted
[0:00:20.367975135] [1816]  INFO Camera camera.cpp:1181 configuring streams: (0) 1640x1232-YUV420 (1) 1640x1232-BGGR16_PISP_COMP1
[0:00:20.368146042] [1830]  INFO RPI pisp.cpp:1390 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected CFE format: 1640x1232-PC1B

This is with a v2 camera NOIR, will try with the V3 module and update

naushir, 2023-11-21:
@sdb-sign, thanks for the update.  The WARN message is expected, and its been removed in the latest source files.

naushir, 2023-11-21:
> Is the latest v2 camera system not supported? If the CSI is plugged into both ports, it will prompt an error. I am sure I have plugged it in properly and this camera can function normally in zero 2w

Camera v2 is supported and working fine for me on Bookworm.  Please continue discussing your issue on your forum post or create a new issue.  This issue is not related to your problem.

naushir, 2023-11-21:
Closing this now as the original reported issue is resolved with the latest kernel.

closed 2023-11-21 by naushir