raspberrypi/rpicam-apps #601: [BUG] libcamera-still --latest link points to wrong folder

m-buchhorn, 2023-11-20:
**Describe the bug**

I'm using libcamera-still (rpicam* don't appear to be available yet?) for a timelapse capture on an RPi4
(_Linux rpi4 6.1.0-rpi4-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.54-1+rpt2 (2023-10-05) aarch64 GNU/Linux_). 

To make the latest image from the timelapse collection available via the web I use "libcamera-still --latest $HOME/public_html/view.jpg (+various timelapse args)" from the folder where libcamera-still is running and today's images are stored ($HOME/$DATE). 

BUG: The view.jpg symlink is created but it points to the same folder as the output symlink - which is not where the actual image is stored.

me@rpi4:~ $ ls -l public_html/
total 0
lrwxrwxrwx 1 me me 17 Nov  8 17:25 view.jpg -> image00002142.jpg

Should be view.jpg -> $HOME/$DATE/image00002142.jpg

Hmmm, and it's now not overwriting the symlink with each newest image, so it only creates the "latest" once. May be a second bug there? I updated a couple of days ago.

Or I'm doing something stupid :-)

bug-report is attached, generated while the timelapse capture is running in a separate process.
[bug.txt](https://github.com/raspberrypi/rpicam-apps/files/13405022/bug.txt)

naushir, 2023-11-21:
Would you be able to share the exact command you are using for this timelapse capture?

m-buchhorn, 2023-11-21:
Sure! The script I'm running starts with:

#!/bin/bash
DATE=`/bin/date +%Y%m%d`
mkdir -p $DATE
cd $DATE

/usr/bin/libcamera-still -n --latest /home/mbuch/public_html/17f.jpg -q 100 -t 52200000 --timelapse 20000 -o image%08d.jpg

(and is followed by post-processing to make a movie out of the images at the end of the run)

naushir, 2023-11-21:
I think this will only work if you fully qualify the output image with the path.  So something like
```
/usr/bin/libcamera-still -n --latest /home/mbuch/public_html/17f.jpg -q 100 -t 52200000 --timelapse 20000 -o $DATE/image%08d.jpg
```
ought to work.  Can you give that a try?

m-buchhorn, 2023-11-25:
Ok, tried that, and it's weird:

mbuch@rpi4:~ $ /usr/bin/libcamera-still -n --latest /home/mbuch/public_html/18f.jpg -q 100 -t 20000 --timelapse 2000 -o tmp/image%08d.jpg

yielded:

mbuch@rpi4:~ $ ls -l public_html/
total 0
lrwxrwxrwx 1 mbuch mbuch 21 Nov 25 21:09 18f.jpg -> tmp/image00000000.jpg

It's taken the '-o' path prefix as a literal prefix for the link, kind of what you suggested, but it's incorrect since there is no subdirectory ~/public_html/tmp. It's treated it as a relative path.

Note also, there are 9 images generated, but the --latest link is only created for the first image.

Trying again, with a full path specified:
mbuch@rpi4:~ $ /usr/bin/libcamera-still -n --latest /home/mbuch/public_html/18f.jpg -q 100 -t 20000 --timelapse 2000 -o /home/mbuch/tmp/image%08d.jpg

yielded:
mbuch@rpi4:~ $ ls -l public_html/
total 0
lrwxrwxrwx 1 mbuch mbuch 33 Nov 25 21:16 18f.jpg -> /home/mbuch/tmp/image00000008.jpg

Which, blow me down, worked on both fronts - it generated a functional link, and it also generated the right link for each image in the timelapse!?!? 

I also tried it with '-o ./image.jpg' but that failed the same way as the first (an incorrect relative link, and again only for the first image):
lrwxrwxrwx 1 mbuch mbuch 19 Nov 25 21:29 18f.jpg -> ./image00000000.jpg

It clearly needs the -o argument to start with '/'.

This is very different to the raspi-still command line, which worked as I had expected (it includes the $PWD of the output in the link, without having to be told). The libcamera-still tool I had understood to be largely a drop-in replacement, barring the loss of a couple of features (eg. -a and -dt). Ok, so I can use this, with a tweak of the script to include the full path, but it seems very counterintuitive to fix the --latest <img> link by modifying the -o <output> argument, isn't it? libcam knows where it is writing the output, so could link there...

At worst, perhaps the documentation could be clarified ;-)

Thanks!

naushir, 2023-11-26:
Thank you for the feedback! Glad you got it to work. I will look to rationalize this when I get the chance. Until then I'll update the docs to note that you have to use fully qualified paths in such cases.

closed 2023-11-26 by naushir