raspberrypi/rpicam-apps #603: [BUG] libcamera-still hangs with --immediate on Bookworm

aaronwmorris, 2023-11-20:
**Describe the bug**
libcamera-still hangs with the `--immediate` flag on Raspberry Pi OS 12 (Bookworm).  Tested with imx519 and imx378.  This appears to happen with all cameras.

**Bug report**
```
--------------------------------------------------------------------------------
libcamera-apps Bug Report
--------------------------------------------------------------------------------
Date: 20-11-2023 (12:50:18)
Command: /usr/bin/camera-bug-report -t 10 -o bug-report_imx378.txt -c libcamera-still --immediate --shutter 10000 --awbgains 1,1 --output /tmp/foo.jpg

--------------------------------------------------------------------------------
Hardware information
--------------------------------------------------------------------------------
** hostname **
allskypi4

** cat /proc/cpuinfo **
processor       : 0
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 1
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 2
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 3
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

Revision        : c03115
Serial          : 10000000e176bd86
Model           : Raspberry Pi 4 Model B Rev 1.5

--------------------------------------------------------------------------------
Configuration
--------------------------------------------------------------------------------
** cat /boot/cmdline.txt **
console=serial0,115200 console=tty1 root=PARTUUID=d37f5dc4-02 rootfstype=ext4 fsck.repair=yes rootwait

** cat /boot/config.txt **
# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
[... 915 lines cut ...]
Oct 17 2023 15:39:16
Copyright (c) 2012 Broadcom
version 30f0c5e4d076da3ab4f341d88e7d505760b93ad7 (clean) (release) (start)

** libcamera-still --version **
libcamera-apps build: f8b05c47894c 23-10-2023 (09:45:06)
libcamera build: v0.1.0+99-4a23664b

--------------------------------------------------------------------------------
libcamera-still --immediate --shutter 10000 --awbgains 1,1 --output /tmp/foo.jpg
--------------------------------------------------------------------------------
Preview window unavailable
[0:02:14.062368669] [1017]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+99-4a23664b
[0:02:14.103029053] [1020]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:02:14.105650642] [1020]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media4 and ISP device /dev/media1
[0:02:14.105715216] [1020]  INFO RPI pipeline_base.cpp:1142 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Mode selection for 4056:3040:12:P
    SRGGB10_CSI2P,1332x990/0 - Score: 10552.2
    SRGGB12_CSI2P,2028x1080/0 - Score: 8179.84
    SRGGB12_CSI2P,2028x1520/0 - Score: 7096
    SRGGB12_CSI2P,4056x3040/0 - Score: 0
[0:02:14.106919566] [1017]  INFO Camera camera.cpp:1181 configuring streams: (0) 4056x3040-YUV420 (1) 4056x3040-SBGGR12_CSI2P
[0:02:14.107350824] [1020]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 4056x3040-SBGGR12_1X12 - Selected unicam format: 4056x3040-pBCC
Error: ***** TIMEOUT *****
```

naushir, 2023-11-21:
I can reproduce this issue.  Unfortunately it is a regression caused by the zero shutter lag functionality we introduced a few weeks back.  The fix is quick and easy, i'll post an update shortly.

naushir, 2023-11-21:
This should be fixed with https://github.com/raspberrypi/rpicam-apps/pull/605.

naushir, 2023-11-21:
Change has been merged.  Resolving now.

closed 2023-11-21 by naushir