raspberrypi/rpicam-apps #607: pi 5 bookworm camera v2 fail

suosuo6, 2023-11-21:
When I install a clean bookworm system, I can take photos normally by running libcamera-jpeg - o test.jpg.
#140 (30.01 fps) exp 33251.00 ag 9.85 dg 1.02
#141 (30.01 fps) exp 33251.00 ag 9.85 dg 1.02
#142 (30.01 fps) exp 33251.00 ag 9.85 dg 1.02
[0:01:58.096818221] [2107]  WARN V4L2 v4l2_pixelformat.cpp:336 Unsupported V4L2 pixel format YM42
[0:01:58.097199388] [2107]  INFO Camera camera.cpp:1213 configuring streams: (0) 3280x2464-YUV420 (1) 3280x2464-BGGR16_PISP_COMP1
[0:01:58.103659536] [2110]  INFO RPI pisp.cpp:1428 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx219@10 - Selected sensor format: 3280x2464-SBGGR10_1X10 - Selected CFE format: 3280x2464-PC1B
Still capture image received

When I execute sudo update&&sudo upgrade, libcamera jpeg - o test.jpg is no longer available

libcamera-jpeg -o test.jpg
Preview window unavailable
[0:11:11.121336227] [35724]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+99-4a23664b
[0:11:11.139251438] [35727]  INFO RPI pisp.cpp:652 libpisp version v1.0.0 ce5624ed8726 23-10-2023 (07:41:40)
[0:11:11.157577352] [35727]  INFO RPI pisp.cpp:1106 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx219@10 to CFE device /dev/media3 and ISP device /dev/media0 using PiSP variant BCM2712_C0
Mode selection for 1640:1232:12:P
    SRGGB10_CSI2P,640x480/0 - Score: 4504.81
    SRGGB10_CSI2P,1640x1232/0 - Score: 1000
    SRGGB10_CSI2P,1920x1080/0 - Score: 1541.48
    SRGGB10_CSI2P,3280x2464/0 - Score: 1718
    SRGGB8,640x480/0 - Score: 5504.81
    SRGGB8,1640x1232/0 - Score: 2000
    SRGGB8,1920x1080/0 - Score: 2541.48
    SRGGB8,3280x2464/0 - Score: 2718
Stream configuration adjusted
[0:11:11.158796387] [35724]  INFO Camera camera.cpp:1181 configuring streams: (0) 1640x1232-YUV420 (1) 1640x1232-BGGR16_PISP_COMP1
[0:11:11.158914800] [35727]  INFO RPI pisp.cpp:1390 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected CFE format: 1640x1232-PC1B
[0:11:11.171041757] [35727] ERROR V4L2 v4l2_videodevice.cpp:1906 /dev/video4[16:cap]: Failed to start streaming: Invalid argument
ERROR: *** failed to start camera ***

dmesg

i2c_designware 1f00088000.i2c: timeout waiting for bus ready
[   28.986600] imx219 6-0010: Failed to write reg 0x30eb. error = -5
[   28.986604] imx219 6-0010: imx219_start_streaming failed to send mfg header
[   28.986618] rp1-cfe 1f00110000.csi: stream on failed in subdev
[   30.218623] i2c_designware 1f00088000.i2c: timeout in disabling adapter

I don't think there are any issues with the CSI cable and hardware. It was after updating the system software that the camera couldn't be opened

suosuo6, 2023-11-21:
use Raspberry pi os 64 2023.10.10 1.1GB

naushir, 2023-11-21:
I've just tired my Camera v2 on a Pi 5 with up-to-date software + kernel and it works fine.

Can you try the following if possible:

- Test this same camera on a Pi 4 board.
- Try attaching the same camera to the other CSI-2 port on the Pi 5
- If you have another official Raspberry Pi camera, try that on the Pi 5 board.

suosuo6, 2023-11-21:
This camera works normally on my zero 2w bulleye, but the same error occurs on the zreo 2w bookworm
I have also tried it on CSI 2
At present, I don't have any other official cameras in hand. I will consider buying another v3 to try
It's quite strange that not updating the system can be used normally

naushir, 2023-11-21:
This is indeed very strange.  Can you send me the output of ``uname -a``.  After doing a ``sudo rpi-update``, it should show
```
Linux pi5 6.1.62-v8+ #1697 SMP PREEMPT Fri Nov 17 17:36:01 GMT 2023 aarch64 GNU/Linux
```

suosuo6, 2023-11-21:
Linux suo 6.1.62-v8_16k+ #1697 SMP PREEMPT Fri Nov 17 17:38:54 GMT 2023 aarch64 GNU/Linux

naushir, 2023-11-21:
Nope, still works for me.  Can you dump the output of /boot/firmware/config.txt please?

suosuo6, 2023-11-21:
cat  /boot/firmware/config.txt
# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

naushir, 2023-11-22:
@suosuo6 I'm not really sure what more to suggest here.

When you ran a ``sudo rpi-update`` and your camera failed, do you still get this exact error message:
```
ERROR V4L2 v4l2_videodevice.cpp:1906 /dev/video8[18:cap]: Failed to start streaming: Input/output error
```
and the dmesg log shows i2c errors?

suosuo6, 2023-11-22:
yes ，After I went back, I was testing and reinstalling Bookworm

davidplowman, 2023-11-22:
I've had a go at this too. I have a Pi 4 4GB with an imx219. I flashed a new SD card image (the top one offered by rpi-imager), but didn't let it do any updates on first boot. It reported
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 6.1.0-rpi4-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.54-1+rpt2 (2023-10-05) aarch64 GNU/Linux
pi@raspberrypi:~ $ libcamera-hello --version
libcamera-apps build: 7e4d3d71867f 22-08-2023 (10:53:13)
libcamera build: v0.1.0+52-a858d20b
```
and `libcamera-hello` worked fine.

Then I did `sudo apt update` and `sudo apt upgrade`. Afterward a reboot it reported
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 6.1.0-rpi6-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.58-1+rpt2 (2023-10-27) aarch64 GNU/Linux
pi@raspberrypi:~ $ libcamera-hello --version
libcamera-apps build: f8b05c47894c 23-10-2023 (09:45:08)
libcamera build: v0.1.0+99-4a23664b
```
and `libcamera-hello` seems to work just the same. Not sure if that helps or just muddies the waters...

naushir, 2023-11-22:
The above matches my experience as well.  I'm out of ideas now.  @suosuo6 I suggest you try to get another official Raspberry Pi camera board and see if that still gives you problems.

suosuo6, 2023-11-22:
@naushir Thank you for your suggestion. I will try to purchase a camera or exchange for a PI 5

suosuo6, 2023-11-23:
@naushir  Hi ， I have noticed that the version of libcamera cannot be used after apt upgrade. I have noticed that the current version is 0.1.0+pi5~0-1. Is it possible to upgrade without supporting pi5？
sudo apt list --upgradable
libcamera-apps/stable 1.3.0-1 arm64 [upgradable from: 1.2.1-1+bookworm+1]
libcamera-ipa/stable 0.1.0+rpt20231020-1 arm64 [upgradable from: 0.1.0+pi5~0-1]
libcamera-tools/stable 0.1.0+rpt20231020-1 arm64 [upgradable from: 0.1.0+pi5~0-1]
libcamera0.1/stable 0.1.0+rpt20231020-1 arm64 [upgradable from: 0.1.0+pi5~0-1]

suosuo6, 2023-11-23:
@naushir  Hi, I noticed that the old version of libcarema comes with a PI5 version number, while the upgraded version does not. May this be the impact?
sudo apt-cache policy  libcamera0.1
libcamera0.1:
  Installed: 0.1.0+pi5~0-1
  Candidate: 0.1.0+rpt20231020-1
  Version table:
     0.1.0+rpt20231020-1 500
        500 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages
 *** 0.1.0+pi5~0-1 100
        100 /var/lib/dpkg/status

naushir, 2023-11-23:
I think that is because our old version of the library was created from a pre-release Pi 5 tree (hence the tag).  This tree then got merged into our main libcamera release tree so the tag is not present.

suosuo6, 2023-11-23:
Okay, tomorrow I'll have the v3 camera. Test it out

suosuo6, 2023-11-24:
@naushir After replacing the v3 camera, it can work normally, but there is a problem with the v2 camera

closed 2023-11-28 by naushir

jiapei100, 2024-02-22:
Meet the same error today.

```
➜  sudo rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 !!! Failed to download update for rpi-update!
 !!! Make sure you have ca-certificates installed and that the time is set correctly
```