raspberrypi/rpicam-apps #611: [BUG] pi5 bookworm, h264 profile error

Gordon999, 2023-11-25:
Pi5 bookworm 64bit, pi hq camera

gt64bw@raspberrypi:~ $ libcamera-vid --camera 0 -t 10000 -o /home/gt64bw/Videos/231125100852.h264 --framerate 25 --profile main --level 4 --brightness 0.0 --contrast 0.7 --width 1920 --height 1080 --exposure normal --gain 0 --awb auto --metering average --saturation 1.0 --sharpness 1.5 --denoise cdn_off -p 0,0,800,600
[0:51:14.927100860] [8897]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[0:51:14.950417154] [8902]  INFO RPI pisp.cpp:653 libpisp version v1.0.2 fa44a258644a 22-11-2023 (21:59:22)
[0:51:14.970286573] [8902]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media3 and ISP device /dev/media1 using PiSP variant BCM2712_C0
[0:51:14.970373443] [8902]  INFO RPI pisp.cpp:653 libpisp version v1.0.2 fa44a258644a 22-11-2023 (21:59:22)
[0:51:14.979376684] [8902]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx477@1a to CFE device /dev/media4 and ISP device /dev/media2 using PiSP variant BCM2712_C0
Made X/EGL preview window
Mode selection for 1920:1080:12:P(25)
    SRGGB10_CSI2P,1536x864/120.135 - Score: 2200
    SRGGB10_CSI2P,2304x1296/56.0255 - Score: 1150
    SRGGB10_CSI2P,4608x2592/14.3536 - Score: 23342.8
Stream configuration adjusted
[0:51:15.333786687] [8897]  INFO Camera camera.cpp:1183 configuring streams: (0) 1920x1080-YUV420 (1) 2304x1296-BGGR16_PISP_COMP1
[0:51:15.333916409] [8902]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected CFE format: 2304x1296-PC1B
ERROR: *** libav: no such profile main ***

Gordon999, 2023-11-26:
I took the card out of the Pi5 and put into a Pi4 and the commend works.

Pi5 not do h264 ? I don't see anything in https://www.raspberrypi.com/documentation/computers/camera_software.html#introduction

naushir, 2023-11-27:
I can reproduce this as well.  It seems to be something to do with how libav advertises (or does not in this case) codec profiles.  The libx264 codec seems to fail for this reason.  I've got a candidate fix at https://github.com/raspberrypi/rpicam-apps/pull/612 which uses a different mechanism to get the codec profiles that seems more robust.

naushir, 2023-11-28:
The fix has now been merged.

closed 2023-11-28 by naushir