raspberrypi/rpicam-apps #613: [BUG]Made X/EGL preview window [0:53:54.258772861] [3159]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f ERROR: *** no cameras available ***

moaz-azam, 2023-11-27:
i have v3 camera module it was working fine i install some libraries due to which vnc veiwer and putty stopped working have to flash card again since than camera not working
[report.txt](https://github.com/raspberrypi/rpicam-apps/files/13474373/report.txt)

naushir, 2023-11-28:
- Can you list exactly what libraries you installed?
- Have you tried running ``sudo apt update && sudo apt full-upgrade -y`` to see if that fixes your problem?

moaz-azam, 2023-11-28:
i dont remember what libraries i install i was working with webrtc frame work and libcamera command were working fine and rpicam commands not working command not found error tried this sudo apt update && sudo apt full-upgrade but didnot get any result . is my cam damaged ?

naushir, 2023-11-28:
It's very unlikely that your camera hardware is damaged from installing some additional libraries.  I would suggest you start with a fresh OS sdcard image and try again.

naushir, 2023-11-30:
Any updates on this?

moaz-azam, 2023-12-01:
camera connector got damaged ordered new pi zero w 2 all working fine. 

Now having some problem with picamera2 library 

Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.9/site-packages/numpy/core/__init__.py", line 24, in <module>
    from . import multiarray
  File "/home/pi/.local/lib/python3.9/site-packages/numpy/core/multiarray.py", line 10, in <module>
    from . import overrides
  File "/home/pi/.local/lib/python3.9/site-packages/numpy/core/overrides.py", line 8, in <module>
    from numpy.core._multiarray_umath import (
ImportError: libopenblas.so.0: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.9/site-packages/numpy/__init__.py", line 130, in <module>
    from numpy.__config__ import show as show_config
  File "/home/pi/.local/lib/python3.9/site-packages/numpy/__config__.py", line 4, in <module>
    from numpy.core._multiarray_umath import (
  File "/home/pi/.local/lib/python3.9/site-packages/numpy/core/__init__.py", line 50, in <module>
    raise ImportError(msg)
ImportError: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.9 from "/usr/bin/python"
  * The NumPy version is: "1.26.2"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: libopenblas.so.0: cannot open shared object file: No such file or directory

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/Desktop/New3.py", line 8, in <module>
    from picamera2 import Picamera2
  File "/usr/lib/python3/dist-packages/picamera2/__init__.py", line 7, in <module>
    from .converters import YUV420_to_RGB
  File "/usr/lib/python3/dist-packages/picamera2/converters.py", line 1, in <module>
    import numpy as np
  File "/home/pi/.local/lib/python3.9/site-packages/numpy/__init__.py", line 135, in <module>
    raise ImportError(msg) from e
ImportError: Error importing numpy: you should not try to import numpy from
        its source directory; please exit the numpy source tree, and relaunch
        your python interpreter from there.

------------------
(program exited with code: 1)
Press return to continue

naushir, 2023-12-01:
> Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

I think you are going to have to re-install numpy to resolve this.  Best to follow the instructions at https://numpy.org/install/.  I'll close this issue down now as the original problem is resolved.

closed 2023-12-01 by naushir