raspberrypi/rpicam-apps #614: [BUG] Sink camera gives ERROR IPARPI cam_helper.cpp:225 Embedded data buffer parsing failed

lindlou, 2023-11-29:
**Describe the bug**
I am following the [documentation](https://www.raspberrypi.com/documentation/accessories/camera.html#using-the-hq-camera) to capture two synchronized videos with two Pis and two HQ cameras.

To prevent the sink camera from timing out before it receives sync signals (see [this post](https://github.com/raspberrypi/picamera2/issues/748)) I added ``"camera_timeout_value_ms": 9000,`` to ``/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml``.

But now on the sink I get this error as soon as I start recording on the source (before the sink's timeout completes):
```ERROR IPARPI cam_helper.cpp:225 Embedded data buffer parsing failed```

If I start the sink when the source is already recording there is no bug and it works just fine.

**Bug reports**
[bug_sink.txt](https://github.com/raspberrypi/rpicam-apps/files/13501012/bug_sink.txt)
[bug_source.txt](https://github.com/raspberrypi/rpicam-apps/files/13501013/bug_source.txt)

**Additional information**
I upgraded the system with ``sudo apt update`` and ``sudo apt upgrade`` this morning, installed rpicam-vid manually with ``sudo apt install rpicam-apps`` (even after the upgrade I only had libcamera-vid) and tried rebooting a few times.

naushir, 2023-11-29:
Can you post the full error log please? The bug reports seem to have them missing.  Also, do you get the same problem if you swap source and sink devices?

lindlou, 2023-11-30:
***Full error log***
```$ rpicam-vid --frames 100 --nopreview -o sink.h264```
```
[0:09:04.102392864] [2062]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[0:09:04.156893530] [2065]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:09:04.159493580] [2065]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media0 and ISP device /dev/media2
[0:09:04.159575135] [2065]  INFO RPI pipeline_base.cpp:1142 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Mode selection for 640:480:12:P
    SRGGB10_CSI2P,1332x990/0 - Score: 1305.05
    SRGGB12_CSI2P,2028x1080/0 - Score: 701.167
    SRGGB12_CSI2P,2028x1520/0 - Score: 607.329
    SRGGB12_CSI2P,4056x3040/0 - Score: 1494.33
[0:09:04.161283879] [2062]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-YUV420 (1) 2028x1520-SBGGR12_CSI2P
[0:09:04.161830485] [2065]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
[0:09:07.702015664] [2071] ERROR IPARPI cam_helper.cpp:225 Embedded data buffer parsing failed
[0:09:07.715298802] [2071] ERROR IPARPI cam_helper.cpp:225 Embedded data buffer parsing failed
[0:09:16.716175115] [2065]  WARN V4L2 v4l2_videodevice.cpp:2007 /dev/video0[13:cap]: Dequeue timer of 9000000.00us has expired!
[0:09:16.716431094] [2065] ERROR RPI pipeline_base.cpp:1374 Camera frontend has timed out!
[0:09:16.716556834] [2065] ERROR RPI pipeline_base.cpp:1375 Please check that your camera sensor connector is attached securely.
[0:09:16.716674833] [2065] ERROR RPI pipeline_base.cpp:1376 Alternatively, try another cable and/or sensor.
ERROR: Device timeout detected, attempting a restart!!!
^Z
[1]+  Stopped                 rpicam-vid --frames 100 --nopreview -o sink.h264
```

***Full verbose error log***
```$ rpicam-vid --frames 100 --nopreview -o sink.h264 -v```
```
[0:10:43.967897114] [2100]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[0:10:44.023426176] [2103]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:10:44.026057523] [2103]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media0 and ISP device /dev/media2
[0:10:44.026140467] [2103]  INFO RPI pipeline_base.cpp:1142 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Options:
    verbose: 2
    info_text:
    timeout: 5000ms
    width: 640
    height: 480
    output: sink.h264
    post_process_file:
    preview: none
    qt-preview: 0
    transform: identity
    roi: all
    metering: centre
    exposure: normal
    ev: 0
    awb: auto
    flush: false
    wrap: 0
    brightness: 0
    contrast: 1
    saturation: 1
    sharpness: 1
    framerate: 30
    denoise: auto
    viewfinder-width: 0
    viewfinder-height: 0
    tuning-file: (libcamera)
    lores-width: 0
    lores-height: 0
    autofocus-range: normal
    autofocus-speed: normal
    autofocus-window: all
    hdr: off
    mode: unspecified
    viewfinder-mode: unspecified
    metadata:
    metadata-format: json
    bitrate: 0kbps
[... 84 lines cut ...]
```$ libcamera-vid --frames 100 -o sink.h264```
```
[0:35:47.102698974] [2124]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[0:35:47.156844345] [2127]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:35:47.159374904] [2127]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media4 and ISP device /dev/media1
[0:35:47.159502125] [2127]  INFO RPI pipeline_base.cpp:1142 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Preview window unavailable
Mode selection for 640:480:12:P
    SRGGB10_CSI2P,1332x990/0 - Score: 1305.05
    SRGGB12_CSI2P,2028x1080/0 - Score: 701.167
    SRGGB12_CSI2P,2028x1520/0 - Score: 607.329
    SRGGB12_CSI2P,4056x3040/0 - Score: 1494.33
[0:35:47.161626782] [2124]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-YUV420 (1) 2028x1520-SBGGR12_CSI2P
[0:35:47.162076499] [2127]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
[0:35:50.985146733] [2133] ERROR IPARPI cam_helper.cpp:225 Embedded data buffer parsing failed
[0:35:50.998709798] [2133] ERROR IPARPI cam_helper.cpp:225 Embedded data buffer parsing failed
[0:36:00.000167782] [2127]  WARN V4L2 v4l2_videodevice.cpp:2007 /dev/video0[13:cap]: Dequeue timer of 9000000.00us has expired!
[0:36:00.000405020] [2127] ERROR RPI pipeline_base.cpp:1374 Camera frontend has timed out!
[0:36:00.000526796] [2127] ERROR RPI pipeline_base.cpp:1375 Please check that your camera sensor connector is attached securely.
[0:36:00.000643221] [2127] ERROR RPI pipeline_base.cpp:1376 Alternatively, try another cable and/or sensor.
ERROR: Device timeout detected, attempting a restart!!!
^Z
[1]+  Stopped                 libcamera-vid --frames 100 -o sink.h264
```
Again,  ``ERROR IPARPI cam_helper.cpp:225 Embedded data buffer parsing failed`` happens when I start the source.

lindlou, 2023-11-30:
Yes I am using the standard Pi HQ camera v1.0
I set the timeout myself in ``/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml``, otherwise it would be 1s.
Here is the log
[error_log.txt](https://github.com/raspberrypi/rpicam-apps/files/13511498/error_log.txt)

naushir, 2023-12-01:
Adding @njhollinghurst who might be interested in this.

Unfortunately I won't be able to try this out until next week at the earliest.  @njhollinghurst did you ever try 2x IMX477 running in sync?

MitchFuchs, 2024-01-31:
Any information on this issue? I run into the same ERROR. 

Thanks, 
MItch

njhollinghurst, 2024-01-31:
Ah. It could be a documentation error. In a recent (well-meaning but wrong) documentation change, the recommended pull-up was changed from 1K:1K to 10K:10K, which may be too weak to keep XVS high when the source camera is inactive. It might be confusing the sink camera. A stop-gap solution is to replace the pull-up with a stronger one, e.g. 1K:1K.

Alternatively, try manually enabling the source camera and set its "always-on" parameter.
```
camera_auto_detect=0
dtoverlay=imx477,always-on
```

MitchFuchs, 2024-02-01:
@njhollinghurst: thanks for your reply. I changed the resistors to 1K:1K and tried 2 scenarios:

1) with the original config file on source and sink 
2) with the modified config file on source and original config on sink. 

Neither solution works, both return the ERROR 'Embedded data buffer parsing failed'

Questions: 
a) should I also modify the config file on sink RPI?
b) should I run 'sudo rpi-update pulls/5774' as suggested in https://forums.raspberrypi.com/viewtopic.php?t=360476?

Thanks for you help.

njhollinghurst, 2024-02-01:
a) You can, and it's more consistent that way, though theoretically not required.
b) Yes. I thought it might have been merged into the latest release by now, but perhaps it has not been.

A useful check might be to measure the quiescent voltage on XVS when both Raspberry Pis are powered but neither camera is streaming, and when the sink is streaming but the source is not started. In both cases it should be 1.65V-1.8V.

MitchFuchs, 2024-02-01:
@njhollinghurst Doing a) and b) solved the issue and leads to 2 HQ cameras in sync. Thanks! 

Out of scope:  can I keep the same approach to have more than 2 HQ cams in sync?

njhollinghurst, 2024-02-01:
Yes -- now all cameras need the "always-on" flag, and all Raspberry Pis must be powered up before starting streaming. The cameras have a weak internal pull-up (when powered on), so the external voltage divider should no longer be needed.

Now I had better (1) ensure the modification is publically released; (2) update the documentation!

MitchFuchs, 2024-02-01:
Thanks a lot! Can confirm I have 3 HQ cams in perfect sync (without external voltage divider).

naushir, 2024-02-02:
Resolving now...

closed 2024-02-02 by naushir

MitchFuchs, 2024-02-07:
@njhollinghurst quick question to make sure: do the multiple cameras still need to be grounded on RPi pin? Thanks!