raspberrypi/rpicam-apps #616: [BUG] Project unbuildable until new libcamera version released

jpds, 2023-12-05:
**Describe the bug**

Just a heads-up that this project is unbuildable (for at least downstream Linux distros) until a new version of libcamera is released as https://git.libcamera.org/libcamera/libcamera.git/tree/include/libcamera/orientation.h isn't in a versioned tarball.

```
meson: enabled parallel building
meson: enabled parallel installing
building
build flags: -j4
[1/42] Generating core/version.cpp with a custom command
[2/42] Compiling C++ object rpicam_app.so.1.4.1.p/meson-generated_.._core_version.cpp.o
[3/42] Compiling C++ object rpicam_app.so.1.4.1.p/core_rpicam_app.cpp.o
FAILED: rpicam_app.so.1.4.1.p/core_rpicam_app.cpp.o
g++ -Irpicam_app.so.1.4.1.p -I. -I.. -Icore -I/nix/store/1kfrd8rrcjqhrbz3s6f5hls4pizqy4kw-libcamera-0.0.5-dev/include/libcamera -I/nix/store/1kprdajc7d9s1lvcn8vddqf4gvbp8cxw-boost-1.79.0-dev/include -I/nix/store/4107dy5yv62mzfhk31l257wi7pdygs4m-ffmpeg-5.1.3-dev/include -I/nix/store/9j7fk9m7k9cdh8i4q60j8x2aiy0974pq-libexif-0.6.24/include -I/nix/store/2pkk6jcar40i32zhjicnqhhmly8q65hj-libjpeg-turbo-2.1.5.1-dev/include -I/nix/store/3gzq3f5kg927z7jhbmymj5hca33a9b7l-libtiff-4.5.1-dev/include -I/nix/store/86cvg8vkqkjls6cic26nb5rjjdjcjga3-libpng-apng-1.6.39-dev/include/libpng16 -I/nix/store/xii79a1if47i1jqfnn2d9i5v5pvz227f-libdrm-2.4.115-dev/include -I/nix/store/xii79a1if47i1jqfnn2d9i5v5pvz227f-libdrm-2.4.115-dev/include/libdrm -fdiagnostics-color=always -Wall -Winvalid-pch -Wextra -Wpedantic -Werror -std=c++17 -pedantic -Wno-unused-parameter -faligned-new -D_FILE_OFFSET_BITS=64 -Wno-psabi -DLIBAV_PRESENT=1 -DLIBDRM_PRESENT=1 -fPIC -pthread -DBOOST_ALL_NO_LIB -MD -MQ rpicam_app.so.1.4.1.p/core_rpicam_app.cpp.o -MF rpicam_app.so.1.4.1.p/core_rpicam_app.cpp.o.d -o rpicam_app.so.1.4.1.p/core_rpicam_app.cpp.o -c ../core/rpicam_app.cpp
../core/rpicam_app.cpp:25:10: fatal error: libcamera/orientation.h: No such file or directory
   25 | #include <libcamera/orientation.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[4/42] Compiling C++ object rpicam_app.so.1.4.1.p/core_buffer_sync.cpp.o
[5/42] Compiling C++ object rpicam_app.so.1.4.1.p/core_dma_heaps.cpp.o
[6/42] Compiling C++ object rpicam_app.so.1.4.1.p/core_options.cpp.o
FAILED: rpicam_app.so.1.4.1.p/core_options.cpp.o
g++ -Irpicam_app.so.1.4.1.p -I. -I.. -Icore -I/nix/store/1kfrd8rrcjqhrbz3s6f5hls4pizqy4kw-libcamera-0.0.5-dev/include/libcamera -I/nix/store/1kprdajc7d9s1lvcn8vddqf4gvbp8cxw-boost-1.79.0-dev/include -I/nix/store/4107dy5yv62mzfhk31l257wi7pdygs4m-ffmpeg-5.1.3-dev/include -I/nix/store/9j7fk9m7k9cdh8i4q60j8x2aiy0974pq-libexif-0.6.24/include -I/nix/store/2pkk6jcar40i32zhjicnqhhmly8q65hj-libjpeg-turbo-2.1.5.1-dev/include -I/nix/store/3gzq3f5kg927z7jhbmymj5hca33a9b7l-libtiff-4.5.1-dev/include -I/nix/store/86cvg8vkqkjls6cic26nb5rjjdjcjga3-libpng-apng-1.6.39-dev/include/libpng16 -I/nix/store/xii79a1if47i1jqfnn2d9i5v5pvz227f-libdrm-2.4.115-dev/include -I/nix/store/xii79a1if47i1jqfnn2d9i5v5pvz227f-libdrm-2.4.115-dev/include/libdrm -fdiagnostics-color=always -Wall -Winvalid-pch -Wextra -Wpedantic -Werror -std=c++17 -pedantic -Wno-unused-parameter -faligned-new -D_FILE_OFFSET_BITS=64 -Wno-psabi -DLIBAV_PRESENT=1 -DLIBDRM_PRESENT=1 -fPIC -pthread -DBOOST_ALL_NO_LIB -MD -MQ rpicam_app.so.1.4.1.p/core_options.cpp.o -MF rpicam_app.so.1.4.1.p/core_options.cpp.o.d -o rpicam_app.so.1.4.1.p/core_options.cpp.o -c ../core/options.cpp
../core/options.cpp: In member function 'virtual bool Options::Parse(int, char**)':
../core/options.cpp:289:57: error: 'class libcamera::CameraConfiguration' has no member named 'sensorConfig'
  289 |                                                 config->sensorConfig = libcamera::SensorConfiguration();
      |                                                         ^~~~~~~~~~~~
../core/options.cpp:289:83: error: 'SensorConfiguration' is not a member of 'libcamera'; did you mean 'StreamConfiguration'?
  289 |                                                 config->sensorConfig = libcamera::SensorConfiguration();
      |                                                                                   ^~~~~~~~~~~~~~~~~~~
      |                                                                                   StreamConfiguration
../core/options.cpp:290:57: error: 'class libcamera::CameraConfiguration' has no member named 'sensorConfig'
  290 |                                                 config->sensorConfig->outputSize = size;
      |                                                         ^~~~~~~~~~~~
../core/options.cpp:291:57: error: 'class libcamera::CameraConfiguration' has no member named 'sensorConfig'
  291 |                                                 config->sensorConfig->bitDepth = sensor_mode.depth();
      |                                                         ^~~~~~~~~~~~
ninja: build stopped: subcommand failed.
error: builder for '/nix/store/fcvpfhb4cni3l0zqjacgy5pnaw8hjbwk-rpicam-apps-1.4.1.drv' failed with exit code 1;
       last 10 log lines:
       >   289 |                                                 config->sensorConfig = libcamera::SensorConfiguration();
       >       |                                                                                   ^~~~~~~~~~~~~~~~~~~
       >       |                                                                                   StreamConfiguration
       > ../core/options.cpp:290:57: error: 'class libcamera::CameraConfiguration' has no member named 'sensorConfig'
       >   290 |                                                 config->sensorConfig->outputSize = size;
       >       |                                                         ^~~~~~~~~~~~
       > ../core/options.cpp:291:57: error: 'class libcamera::CameraConfiguration' has no member named 'sensorConfig'
       >   291 |                                                 config->sensorConfig->bitDepth = sensor_mode.depth();
       >       |                                                         ^~~~~~~~~~~~
       > ninja: build stopped: subcommand failed.
       For full logs, run 'nix log /nix/store/fcvpfhb4cni3l0zqjacgy5pnaw8hjbwk-rpicam-apps-1.4.1.drv'.
```

sausix, 2023-12-13:
After building and installing [https://github.com/raspberrypi/libcamera](https://github.com/raspberrypi/libcamera) rpicam-apps still fails to build.

And I just want to build for a Pi3. Any git tags or branches available which still build without getting the error?
```
../core/rpicam_app.cpp:25:10: fatal error: libcamera/orientation.h: No such file or directory
   25 | #include <libcamera/orientation.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
```

naushir, 2023-12-13:
> After building and installing https://github.com/raspberrypi/libcamera rpicam-apps still fails to build

This should definitely work. Are you sure meson is picking up your local libcamera buile during the configure stage? And you definitely did install your libcamera build?

naushir, 2023-12-13:
You can find the file in the repo here: https://github.com/raspberrypi/libcamera/blob/563cd78e1c9858769f7e4cc2628e2515836fd6e7/include/libcamera/orientation.h#L5C35-L5C35

sausix, 2023-12-13:
You may be correct. But I had already updated the SD card image from a legacy Raspbian to Raspi OS Lite.
But my terminal still remembers this ouput from the ssh session before:
```
  libcamera
    location             : /usr/local/lib/arm-linux-gnueabihf
    version              : 0.0.5
```
On the next compile I found and removed some preinstalled libcamera stuff before to be sure. But it's the same path!?
```
  libcamera
    location             : /usr/local/lib/arm-linux-gnueabihf
    version              : 0.1.0
```
[Official docs: Building libcamera](https://www.raspberrypi.com/documentation/computers/camera_software.html#building-libcamera)

Anyway, the compilation now has completed. If my camera will still not work is not on this issue.
Thank you for your hint!

closed 2023-12-14 by naushir

adriangalilea, 2024-02-27:
I had the same issue

```
adrian@zero:~/rpicam-apps $ meson compile -C build -j1
INFO: autodetecting backend as ninja
INFO: calculating backend command to run: /usr/bin/ninja -C /home/adrian/rpicam-apps/build -j 1
ninja: Entering directory `/home/adrian/rpicam-apps/build'
[5/41] Compiling C++ object rpicam_app.so.1.4.3.p/core_rpicam_app.cpp.o
FAILED: rpicam_app.so.1.4.3.p/core_rpicam_app.cpp.o 
c++ -Irpicam_app.so.1.4.3.p -I. -I.. -Icore -I/usr/include/libcamera -I/usr/include -I/usr/include/arm-linux-gnueabihf -I/usr/include/libpng16 -I/usr/include/libdrm -fdiagnostics-color=always -Wall -Winvalid-pch -Wextra -Wpedantic -Werror -std=c++17 -O3 -pedantic -Wno-unused-parameter -faligned-new -D_FILE_OFFSET_BITS=64 -Wno-psabi -DLIBDRM_PRESENT=1 -fPIC -pthread -DBOOST_ALL_NO_LIB -MD -MQ rpicam_app.so.1.4.3.p/core_rpicam_app.cpp.o -MF rpicam_app.so.1.4.3.p/core_rpicam_app.cpp.o.d -o rpicam_app.so.1.4.3.p/core_rpicam_app.cpp.o -c ../core/rpicam_app.cpp
../core/rpicam_app.cpp:25:10: fatal error: libcamera/orientation.h: No such file or directory
   25 | #include <libcamera/orientation.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
```