raspberrypi/rpicam-apps #619: no audio input is used

FredFeuerstein0815, 2023-12-12:
It does not even try to use audio input:
rpicam-vid -v --audio-device=default --audio-bitrate=96kbps --audio-codec=aac --audio-channels=1 --libav-audio=1 --audio-source=alsa --flush=1 --width 1280 --height 720 -t 10sec -n --hflip=1 --vflip=1 --brightness=0.23 --framerate=7.5 -b 30000000 --libav-video-codec=h264 --codec=h264 -o test.h264

camera-bug-report:
--------------------------------------------------------------------------------
rpicam-apps Bug Report
--------------------------------------------------------------------------------
Date: 12-12-2023 (09:39:33)
Command: /usr/bin/camera-bug-report -o bug.txt -c rpicam-vid -v --audio-device=default --audio-bitrate=96kbps --audio-codec=aac --audio-channels=1 --libav-audio=1 --audio-source=alsa --flush=1 --width 1280 --height 720 -t 10sec -n --hflip=1 --vflip=1 --brightness=0.23 --framerate=7.5 -b 30000000 --libav-video-codec=h264 --codec=h264 -o test.h264

Raspberry Pi ZeroW with Bookworm

dtoverlay=vc4-kms-v3d
max_framebuffers=2
disable_fw_kms_setup=1
disable_overscan=1
arm_boost=1
otg_mode=1
dtparam=act_led_trigger=actpwr
enable_uart=1
dtoverlay=ov5647
gpio=18,23,24,25,12,16,20,21=op,dh,np
gpio=26,19,13,6,5,22,27,17,4=ip,np
gpu_mem=64
framebuffer_width=1280
framebuffer_height=1024
hdmi_group=2
hdmi_drive=2
hdmi_mode=82
hdmi_force_hotplug=1
hdmi_safe=1

[    7.669977] usb 1-1.4: new full-speed USB device number 4 using dwc_otg
[    7.939034] usb 1-1.4: New USB device found, idVendor=0d8c, idProduct=0014, bcdDevice= 1.00
[    7.947891] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    7.956522] usb 1-1.4: Product: USB Audio Device
[    7.965202] usb 1-1.4: Manufacturer: C-Media Electronics Inc.
[    8.052179] input: C-Media Electronics Inc. USB Audio Device as /devices/platform/soc/20980000.usb/usb1/1-1/1-1.4/1-1.4:1.3/0003:0D8C:0014.0001/input/input0
[    8.160774] hid-generic 0003:0D8C:0014.0001: input,hidraw0: USB HID v1.00 Device [C-Media Electronics Inc. USB Audio Device] on usb-20980000.usb-1.4/input3

--------------------------------------------------------------------------------
Media Devices
--------------------------------------------------------------------------------
** media-ctl -d 0 -p **
Media controller API version 6.1.63

Media device information
------------------------
driver          bcm2835-isp
model           bcm2835-isp
serial          
bus info        platform:bcm2835-isp
hw revision     0x0
driver version  6.1.63

Device topology
- entity 1: bcm2835_isp0 (4 pads, 4 links)
            type Node subtype Unknown flags 0
	pad0: Sink
		<- "bcm2835-isp0-output0":0 [ENABLED,IMMUTABLE]
	pad1: Source
		-> "bcm2835-isp0-capture1":0 [ENABLED,IMMUTABLE]
	pad2: Source
		-> "bcm2835-isp0-capture2":0 [ENABLED,IMMUTABLE]
	pad3: Source
		-> "bcm2835-isp0-capture3":0 [ENABLED,IMMUTABLE]

- entity 6: bcm2835-isp0-output0 (1 pad, 1 link)
            type Node subtype V4L flags 0
            device node name /dev/video13
[... 317 lines cut ...]
    AeMeteringMode : [0..3]
Buffers allocated and mapped
Video setup complete
Opened H264Encoder on /dev/video11 as fd 31
Got 6 output buffers
Got 12 capture buffers
Codec streaming started
Requests created
Camera started!
Viewfinder frame 0
FileOutput: opened output file test.h264
FileOutput: output buffer 0xa313b000 size 153050
Viewfinder frame 1
FileOutput: output buffer 0xa30bb000 size 6362
Viewfinder frame 2
.....
FileOutput: output buffer 0xa2dbb000 size 5868
Viewfinder frame 68
Halting: reached timeout of 10000 milliseconds.
Camera stopped!
H264Encoder closed
Closing RPiCam application(frames displayed 68, dropped 0)
Camera stopped!
Tearing down requests, buffers and configuration
Camera closed

FredFeuerstein0815, 2023-12-12:
arecord -L
null
    Discard all samples (playback) or generate zero samples (capture)
default
    Default Audio Device
sysdefault
    Default Audio Device
hw:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct hardware device without any conversions
plughw:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Hardware device with all software conversions
default:CARD=Device
    USB Audio Device, USB Audio
    Default Audio Device
sysdefault:CARD=Device
    USB Audio Device, USB Audio
    Default Audio Device
front:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Front output / input
dsnoop:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct sample snooping device

naushir, 2023-12-12:
Can you try running just the following command and inspect the log output:
```
libcamera-vid -t 10s -o test.mkv --libav-audio --bitrate 10mbps --qt-preview
```
This is the output I get:
```
[24:19:02.912267410] [3661]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0
[24:19:02.929054946] [3664]  INFO RPI pisp.cpp:662 libpisp version 000000000000-invalid 21-11-2023 (16:06:35)
[24:19:03.091108056] [3664]  INFO RPI pisp.cpp:1119 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx708@1a to CFE device /dev/media0 and ISP device /dev/media1 using PiSP variant BCM2712_C0
Made QT preview window
Mode selection for 640:480:12:P
    SRGGB10_CSI2P,1536x864/0 - Score: 1486.67
    SRGGB10_CSI2P,2304x1296/0 - Score: 1786.67
    SRGGB10_CSI2P,4608x2592/0 - Score: 2686.67
Stream configuration adjusted
[24:19:03.465999013] [3661]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-YUV420 (1) 1536x864-BGGR16_PISP_COMP1
[24:19:03.466658568] [3664]  INFO RPI pisp.cpp:1403 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected CFE format: 1536x864-PC1B
[libx264 @ 0x558e676f00] using cpu capabilities: ARMv8 NEON
[libx264 @ 0x558e676f00] profile High, level 3.2, 4:2:0, 8-bit
[libx264 @ 0x558e676f00] 264 - core 164 r3095 baee400 - H.264/MPEG-4 AVC codec - Copyleft 2003-2022 - http://www.videolan.org/x264.html - options: cabac=1 ref=1 deblock=1:0:0 analyse=0x3:0x3 me=dia subme=0 psy=1 psy_rd=1.00:0.00 mixed_ref=0 me_range=16 chroma_me=0 trellis=0 8x8dct=1 cqm=0 deadzone=21,11 fast_pskip=1 chroma_qp_offset=0 threads=6 lookahead_threads=1 sliced_threads=0 slices=1 nr=0 decimate=1 interlaced=0 bluray_compat=0 constrained_intra=0 bframes=1 b_pyramid=0 b_adapt=1 b_bias=0 direct=1 weightb=0 open_gop=0 weightp=0 keyint=30 keyint_min=3 scenecut=0 intra_refresh=0 rc=abr mbtree=0 bitrate=10000 ratetol=1.0 qcomp=0.60 qpmin=0 qpmax=69 qpstep=4 ip_ratio=1.40 pb_ratio=1.30 aq=1:1.00
Input #0, pulse, from 'default':
  Duration: N/A, start: 1702372127.620628, bitrate: 1536 kb/s
  Stream #0:0: Audio: pcm_s16le, 48000 Hz, 2 channels, s16, 1536 kb/s
Output #0, matroska, to 'test.mkv':
  Stream #0:0: Video: h264, yuv420p(tv, smpte170m/smpte170m/bt709), 640x480, q=2-31, 10000 kb/s, 30 fps, 30 tbr, 1000k tbn
  Stream #0:1: Audio: aac (LC), 48000 Hz, stereo, fltp, 32 kb/s
Halting: reached timeout of 10000 milliseconds.
[libx264 @ 0x558e676f00] frame I:10    Avg QP: 9.43  size:100070
[libx264 @ 0x558e676f00] frame P:147   Avg QP:11.29  size: 55479
[libx264 @ 0x558e676f00] frame B:136   Avg QP:13.46  size: 24558
[libx264 @ 0x558e676f00] consecutive B-frames:  7.2% 92.8%
[libx264 @ 0x558e676f00] mb I  I16..4:  6.6% 20.9% 72.5%
[libx264 @ 0x558e676f00] mb P  I16..4:  4.1% 10.3%  7.5%  P16..4: 77.5%  0.0%  0.0%  0.0%  0.0%    skip: 0.6%
[libx264 @ 0x558e676f00] mb B  I16..4:  1.6%  1.7%  1.0%  B16..8: 26.0%  0.0%  0.0%  direct:59.4%  skip:10.3%  L0:21.7% L1:44.6% BI:33.7%
[libx264 @ 0x558e676f00] final ratefactor: 12.40
[libx264 @ 0x558e676f00] 8x8 transform intra:40.7% inter:34.6%
[libx264 @ 0x558e676f00] coded y,uvDC,uvAC intra: 99.6% 99.5% 98.9% inter: 77.9% 63.6% 39.5%
[libx264 @ 0x558e676f00] i16 v,h,dc,p:  3%  8% 66% 23%
[libx264 @ 0x558e676f00] i8 v,h,dc,ddl,ddr,vr,hd,vl,hu: 11% 24% 37%  3%  6%  3%  6%  3%  7%
[libx264 @ 0x558e676f00] i4 v,h,dc,ddl,ddr,vr,hd,vl,hu:  9% 36% 12%  3%  9%  5% 11%  3% 11%
[libx264 @ 0x558e676f00] i8c dc,h,v,p: 40% 31% 14% 14%
[libx264 @ 0x558e676f00] kb/s:10240.26
[aac @ 0x558e9377f0] Qavg: 299.678
```

Do you see log message from the audio codec?

naushir, 2023-12-12:
I think I see your problem.  You are using `` --codec=h264``, whereas for audo encoding, you must use ``--codec=libav`` as per the documentation [here](https://www.raspberrypi.com/documentation/computers/camera_software.html#libav-integration-with-rpicam-vid).

FredFeuerstein0815, 2023-12-12:
No log about audio codec or device. Its headless with Bookworm Lite, so no qt-preview.

libcamera-vid -t 10s -o test.mkv --libav-audio --bitrate 10mbps --codec=libav
[25:10:57.071062051] [15448] ERROR IPAModule ipa_module.cpp:172 Symbol ipaModuleInfo not found
[25:10:57.074203016] [15448] ERROR IPAModule ipa_module.cpp:292 v4l2-compat.so: IPA module has no valid info
[25:10:57.077501979] [15448]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[25:10:57.486198396] [15451]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[25:10:57.511339114] [15451]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media0
[25:10:57.514169083] [15451]  INFO RPI pipeline_base.cpp:1142 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Preview window unavailable
Mode selection for 640:480:12:P
    SGBRG10_CSI2P,640x480/0 - Score: 1000
    SGBRG10_CSI2P,1296x972/0 - Score: 1287
    SGBRG10_CSI2P,1920x1080/0 - Score: 1636.67
    SGBRG10_CSI2P,2592x1944/0 - Score: 1854
Stream configuration adjusted
[25:10:57.552192656] [15448]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-YUV420 (1) 640x480-SGBRG10_CSI2P
[25:10:57.554995625] [15451]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
[h264_v4l2m2m @ 0x1829cf0]  <<< v4l2_encode_init: fmt=179/0
[h264_v4l2m2m @ 0x1829cf0] Using device /dev/video11
[h264_v4l2m2m @ 0x1829cf0] driver 'bcm2835-codec' on card 'bcm2835-codec-encode' in mplane mode
[h264_v4l2m2m @ 0x1829cf0] requesting formats: output=YU12 capture=H264
ERROR: *** libav: cannot open pulse input device default ***

naushir, 2023-12-12:
```
ERROR: *** libav: cannot open pulse input device default ***
```
This is almost certainly relevant.  I'm not sure if all the libav libraries are installed by default on the Raspberry Pi OS Lite.  Are you able to try this with the full Raspberry Pi OS image?

FredFeuerstein0815, 2023-12-12:
On Pi4 with Bullseye:

libcamera-vid -t 10 -o test.mkv --libav-audio --bitrate 10000 --codec=libav
Preview window unavailable
[1179:43:59.696237910] [1027099]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f
[1179:43:59.746382421] [1027100]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media4 and ISP device /dev/media0
[1179:43:59.746470049] [1027100]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[1179:43:59.815866209] [1027099]  INFO Camera camera.cpp:1033 configuring streams: (0) 640x480-YUV420
[1179:43:59.816846689] [1027100]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
[h264_v4l2m2m @ 0x558da5dff0]  <<< v4l2_encode_init: fmt=181/0
[h264_v4l2m2m @ 0x558da5dff0] Using device /dev/video11
[h264_v4l2m2m @ 0x558da5dff0] driver 'bcm2835-codec' on card 'bcm2835-codec-encode' in mplane mode
[h264_v4l2m2m @ 0x558da5dff0] requesting formats: output=YU12 capture=H264
Input #0, pulse, from 'default':
  Duration: N/A, start: 1702373404.734846, bitrate: 1536 kb/s
    Stream #0:0: Audio: pcm_s16le, 48000 Hz, 2 channels, s16, 1536 kb/s
Output #0, matroska, to 'test.mkv':
    Stream #0:0: Video: h264 (h264_v4l2m2m), drm_prime(tv, smpte170m/smpte170m/bt709), 640x480, q=-1--1, 10 kb/s, 30 fps, 30 tbr, 1000k tbn
    Stream #0:1: Audio: aac (LC), 48000 Hz, stereo, fltp, 32 kb/s
Halting: reached timeout of 10 milliseconds.
Speicherzugriffsfehler

naushir, 2023-12-12:
This seems to have worked - in so far as you are at least getting an audio stream encoded in your mkv file.  Does the audio work when you play back the file?

FredFeuerstein0815, 2023-12-12:
Yes, i got sound with:
libcamera-vid -t 10000 -o test.mkv --libav-audio --bitrate 10000 --codec=libav
So i have to install Raspbian OS with Desktop ?

FredFeuerstein0815, 2023-12-12:
Or can i install needed software ?
i yes, which do i have to install ?

naushir, 2023-12-12:
```
So i have to install Raspbian OS with Desktop ?
```
Unfortunately, for now the answer is yes.  The problem is that ffmpeg/libav pull in loads of dependencies that we don't want installing in the lite OS image.  When I get some time, I will see if I can somehow reduce the list of dependencies, then we can possibly include it as well.

FredFeuerstein0815, 2023-12-12:
Ok, thanks.

closed 2023-12-12 by naushir

FredFeuerstein0815, 2023-12-12:
It does not work on ZeroW with Bookworm (with Desktop):
It hangs until i stop it:
[0:53:40.717802662] [2119]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[0:53:41.094203636] [2122]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:53:41.117476387] [2122]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media0
[0:53:41.120697352] [2122]  INFO RPI pipeline_base.cpp:1142 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Preview window unavailable
Mode selection for 640:480:12:P
    SGBRG10_CSI2P,640x480/0 - Score: 1000
    SGBRG10_CSI2P,1296x972/0 - Score: 1287
    SGBRG10_CSI2P,1920x1080/0 - Score: 1636.67
    SGBRG10_CSI2P,2592x1944/0 - Score: 1854
Stream configuration adjusted
[0:53:41.161502916] [2119]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-YUV420 (1) 640x480-SGBRG10_CSI2P
[0:53:41.164999879] [2122]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
[h264_v4l2m2m @ 0xd35db0]  &lt;&lt;&lt; v4l2_encode_init: fmt=179/0
[h264_v4l2m2m @ 0xd35db0] Using device /dev/video11
[h264_v4l2m2m @ 0xd35db0] driver 'bcm2835-codec' on card 'bcm2835-codec-encode' in mplane mode
[h264_v4l2m2m @ 0xd35db0] requesting formats: output=YU12 capture=H264