raspberrypi/rpicam-apps #623: Framerate Issues RPi 5 + Camera Module 3

GuyPerets106, 2023-12-20:
Hi , I'm using Raspberry Pi 5 with 8GB of RAM + Camera Module 3 NoIR Wide. 
When running the command `rpicam-vid --mode 1536:864:10:P --width 640 --height 400 --hflip 1 --vflip 1 --framerate 120 --frames 1200 -n --hdr off --level 4.2 --lens-position 6 --denoise off -o test.h264` and view the h264 file output I see the video in slow motion. It still 120fps (Checked) but slowed. When trying to convert to mp4 with ffmpeg I get a fast-moving video , also the  specifying the framerate in the conversion. 
Something I noticed is when recording I see: 
`Output #0, h264, to 'test.h264':
  Stream #0:0: Video: h264, yuv420p(tv, smpte170m/smpte170m/bt709), 640x400, q=2-31, 120 fps, 120 tbr, 1000k tbn`
but when running `ffprobe -i test.h264` I see:
`Input #0, h264, from 'test.h264':
  Duration: N/A, bitrate: N/A
  Stream #0:0: Video: h264 (High), yuv420p(tv, smpte170m/smpte170m/bt709, progressive), 640x400, 1000k tbr, 1200k tbn`

The differences in tbr (which I assume should represent fps) goes from 120 in the rpicam-vid text output to 1000k (or 1000000) in ffprobe. 

There is something I'm doing wrong? 
Thanks!

naushir, 2023-12-20:
H.264 elementary streams do not have any timestamping information in them.  So when you play them back in any video application, it assumes a rate of 30fps.  You need to use a container such as mkv or mp4 when saving the encoded file so that timestamps are embedded correctly.

In your command above, replace ``-o test.h264`` with ``-o test.mp4`` and that should then work correctly.

GuyPerets106, 2023-12-20:
> H.264 elementary streams do not have any timestamping information in them. So when you play them back in any video application, it assumes a rate of 30fps. You need to use a container such as mkv or [...]

Oh ok. I actually used this command with my RPi 4 (when it was libcamera-vid) and then converted to mp4 with ffmpeg and it worked fine. So I thought it was a RPi 5 problem. 
There's any performance differences with using `-o <filename>.mp4` compared to `-o <filename>.h264` ? 
I'm really trying to push the limit of stable framerate with this device. Should I see an increase in stable framerate with the same sensor as going from RPi 4 to 5?

naushir, 2023-12-20:
>There's any performance differences with using `-o <filename>.mp4` compared to `-o <filename>.h264` ?

There is a very minimal overhead to using a mp4 container.  With everything else running in the system during video encode, you likely won't even be able to measure the difference.

>I'm really trying to push the limit of stable framerate with this device. Should I see an increase in stable framerate with the same sensor as going from RPi 4 to 5?

Yes you should see better encode performance on the Pi 5.  You also have the ability to dial up/down the encoder CPU usage to match your application if you want.

I'll mark this as complete now.  If there is anything else you need to query, feel free to reopen.

closed 2023-12-20 by naushir

GuyPerets106, 2023-12-20:
> There is a very minimal overhead to using a mp4 container. With everything else running in the system during video encode, you likely won't even be able to measure the difference.
>
>
> Yes you should [...]

Thanks , How can I dial up the encoder CPU usage?

naushir, 2023-12-20:
> Thanks , How can I dial up the encoder CPU usage?

You can use the ``--libav-video-codec-opts`` command line arguments to set various ffmpeg codec options to set encode features.  However, I would leave this as-is unless you really need to.

GuyPerets106, 2023-12-20:
> > Thanks , How can I dial up the encoder CPU usage?
> 
> You can use the `--libav-video-codec-opts` command line arguments to set various ffmpeg codec options to set encode features. However, I would leave this as-is unless you really need to.

Ok , thanks. I will try different options. What does the 'preset=ultrafast' do? and how can I incorporate i in a rpicam-vid command? 
My ultimate goal is to get 120fps recording in 2 cameras simultaneously , hopefully using the non-cropped mode (2304x1296) , I currently use the cropped 1536x864 mode with 100fps in each camera. You think my goal is feasible?

naushir, 2023-12-20:
You can find details on the presets here: https://dev.beandog.org/x264_preset_reference.html.

Dual camera encode at high framerate is going to be quite restrictive I'm afraid.  You should easily be able to get something like 800x600 120fps working, but I'm not sure you are going to get much higher in resolution.  Perhaps MJPEG might be better, it's less CPU intensive, but it does take up approx. 10x more disk space.

GuyPerets106, 2023-12-20:
> You can find details on the presets here: https://dev.beandog.org/x264_preset_reference.html.
>
> Dual camera encode at high framerate is going to be quite restrictive I'm afraid. You should easily be [...]

Oh I didn't mention - I'm only using this resolutions as the camera modes that control the crop , but I specify `--width 640 --height 400` for my videos.

naushir, 2023-12-20:
In which case, the default encoder settings ought to just work for you for dual 640x400 120fps encode, including the full FoV sensor mode.

GuyPerets106, 2023-12-21:
What could be the reason for which I specify `--mode 2304:1296` and still get `Selected sensor format: 1536x864-SBGGR10_1X10 - Selected CFE format: 1536x864-PC1B`?

naushir, 2023-12-21:
I think you want to use ``--mode 2304:1296:10:P``?

GuyPerets106, 2023-12-21:
> I think you want to use `--mode 2304:1296:10:P`?

Yes of course. Same problem. 
Full command is `rpicam-vid --mode 2304:1296:10:P --width 640 --height 400 --camera 0 --framerate 120 --frames 600 -n --hflip 1 --vflip 1 --level 4.2 --hdr off --denoise off -o test.mp4` if I did a  mistake somewhere else.

naushir, 2023-12-21:
Oh, ok sorry I misunderstood what you were trying to do.  Since you are specifying ``--framerate 120`` on your command line, it is overriding the ``--mode 2304:1296:10:P`` option and selecting the 1536x864 mode.  This is the only mode that can run at 120fps since the camera is still only operating with 2 CSI-2 lanes.  If you use the full FoV 2304:x1296 mode, you can only get a maximum of 56 fps unfortunately.  This is down to the sensor and 2-lane operating limitations.

Also on Pi 5, you no longer need to specify ``--denoise off`` for fast farmerate operation.

GuyPerets106, 2023-12-21:
> Oh, ok sorry I misunderstood what you were trying to do. Since you are specifying `--framerate 120` on your command line, it is overriding the `--mode 2304:1296:10:P` option and selecting the [...]

Oh wow , that's unfortunate to hear. Since the Module 3's are relativity new , I thought it could really take advantage of the new hardware of the RPi 5 and not have the same limitations. :/ 

Thanks anyway for all the clarifications!

GuyPerets106, 2023-12-24:
> H.264 elementary streams do not have any timestamping information in them. So when you play them back in any video application, it assumes a rate of 30fps. You need to use a container such as mkv or [...]

Hi @naushir , when using `.mp4` as output file format , I noticed I can't stop the recording mid-way using simple SIGTERM. For example , If I pre-determined around 30 seconds when executing `rpicam-vid` command and wanted to terminate the video at around 15 seconds , the video file comes out corrupted , probably due to mp4 headers not written to the file in the last truncated frame. When using `.h264` file output , I didn't had that problem. How do you recommend stopping an ongoing recording without corrupt the mp4 file? There is a safe tool for that in the library? 
Thanks