raspberrypi/rpicam-apps #625: Raspberry Pi camera issue

obioheme, 2024-01-02:


naushir, 2024-01-03:
Which camera are you using.  You do not need to use ``sudo`` to run the camera.  Also, please ignore ``sudo vcgencmd get_camera``, it is only supported on the legacy camera stack.

naushir, 2024-01-03:
It would also be worth looking at the following post and and working through the instructions:
https://forums.raspberrypi.com/viewtopic.php?t=362707

naushir, 2024-01-04:
Any update on this?  If not, I'll close it down shortly.

Jai-GAY, 2024-01-10:
I like to know why cheese can detect and preview the legacy camera but not rpicam or libcamera?

[Ubuntu 22.04 64-bit](https://ubuntu.com/download/raspberry-pi) on Pi 4B.

$ libcamera-hello --list-cameras
Could not open any dmaHeap device
ERROR: rpicam-apps currently only supports the Raspberry Pi platforms.
Contributions for other platforms are welcome at https://github.com/raspberrypi/rpicam-apps.
$ sudo libcamera-hello --list-cameras
[sudo] password for pi:
ERROR: rpicam-apps currently only supports the Raspberry Pi platforms.
Contributions for other platforms are welcome at https://github.com/raspberrypi/rpicam-apps.

#config.txt
[pi4]
dtoverlay=vc4-fkms-v3d
#dtoverlay=vc4-kms-v3d
#camera_auto_detect=1
start_x=1
gpu_mem=128

$ sudo v4l2-ctl --list-devices
bcm2835-codec-decode (platform:bcm2835-codec):
        /dev/video10
        /dev/video11
        /dev/video12
        /dev/video18
        /dev/video31
        /dev/media1

bcm2835-isp (platform:bcm2835-isp):
        /dev/video13
        /dev/video14
        /dev/video15
        /dev/video16
        /dev/video20
        /dev/video21
        /dev/video22
        /dev/video23
        /dev/media0
        /dev/media2

mmal service 16.1 (platform:bcm2835-v4l2-0):
        /dev/video0

$ vcgencmd get_camera
VCHI initialization failed
$ sudo vcgencmd get_camera
supported=1 detected=1, libcamera interfaces=0

Installing symlink pointing to libcamera-base.so.0.1.0 to /usr/local/lib/aarch64-linux-gnu/libcamera-base.so.0.1
Installing symlink pointing to libcamera-base.so.0.1 to /usr/local/lib/aarch64-linux-gnu/libcamera-base.so
Installing symlink pointing to libpisp.so.1.0 to /usr/local/lib/aarch64-linux-gnu/libpisp.so
Installing symlink pointing to libcamera.so.0.1.0 to /usr/local/lib/aarch64-linux-gnu/libcamera.so.0.1
Installing symlink pointing to libcamera.so.0.1 to /usr/local/lib/aarch64-linux-gnu/libcamera.so
Running custom install script '/home/pi/libcamera/src/ipa/ipa-sign-install.sh /home/pi/libcamera/build/src/ipa-priv-key.pem lib/aarch64-linux-gnu/libcamera/ipa_rpi_vc4.so lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so'

Installing rpicam_app.so.1.4.1 to /usr/local/lib/aarch64-linux-gnu
Installing apps/rpicam-still to /usr/local/bin
Installing apps/rpicam-vid to /usr/local/bin
Installing apps/rpicam-hello to /usr/local/bin
Installing apps/rpicam-raw to /usr/local/bin
Installing apps/rpicam-jpeg to /usr/local/bin

**sudo** cheese # can get the preview
[image]

Jai-GAY, 2024-01-10:
> #config.txt
#adding
dtoverlay=ov5647

$ libcamera-hello --list-cameras
Could not open any dmaHeap device
No cameras available!
$ sudo libcamera-hello --list-cameras
[sudo] password for pi: 
Available cameras
-----------------
0 : ov5647 [2592x1944 10-bit GBRG] (/base/soc/i2c0mux/i2c@1/ov5647@36)
    Modes: 'SGBRG10_CSI2P' : 640x480 [58.92 fps - (16, 0)/2560x1920 crop]
                             1296x972 [43.25 fps - (0, 0)/2592x1944 crop]
                             1920x1080 [30.62 fps - (348, 434)/1928x1080 crop]
                             2592x1944 [15.63 fps - (0, 0)/2592x1944 crop]

$ sudo v4l2-ctl --list-devices
bcm2835-codec-decode (platform:bcm2835-codec):
	/dev/video10
	/dev/video11
	/dev/video12
	/dev/video18
	/dev/video31
	/dev/media1

bcm2835-isp (platform:bcm2835-isp):
	/dev/video13
	/dev/video14
	/dev/video15
	/dev/video16
	/dev/video20
	/dev/video21
	/dev/video22
	/dev/video23
	/dev/media2
	/dev/media3

unicam (platform:fe801000.csi):
	/dev/video0
	/dev/media0

$ v4l2-ctl --list-devices
bcm2835-codec-decode (platform:bcm2835-codec):
	/dev/video10
	/dev/video11
	/dev/video12
	/dev/video18
	/dev/video31
	/dev/media1

bcm2835-isp (platform:bcm2835-isp):
	/dev/video13
	/dev/video14
	/dev/video15
	/dev/video16
	/dev/video20
	/dev/video21
	/dev/video22
	/dev/video23
	/dev/media2
	/dev/media3

unicam (platform:fe801000.csi):
	/dev/video0
	/dev/media0

$ vcgencmd get_camera
VCHI initialization failed
$ sudo vcgencmd get_camera
supported=1 **detected=0**, libcamera interfaces=1

$ sudo rpicam-hello
[0:06:23.954700378] [4545]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[0:06:24.026313253] [4546]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:06:24.033455383] [4546]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media0 and ISP device /dev/media2
[0:06:24.034031918] [4546]  INFO RPI pipeline_base.cpp:1142 Using configuration file '/usr/local/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
libEGL warning: DRI2: failed to authenticate
Made X/EGL preview window
Mode selection for 1296:972:12:P
    SGBRG10_CSI2P,640x480/0 - Score: 3296
    SGBRG10_CSI2P,1296x972/0 - Score: 1000
    SGBRG10_CSI2P,1920x1080/0 - Score: 1349.67
    SGBRG10_CSI2P,2592x1944/0 - Score: 1567
Stream configuration adjusted
[0:06:24.420523761] [4545]  INFO Camera camera.cpp:1183 configuring streams: (0) 1296x972-YUV420 (1) 1296x972-SGBRG10_CSI2P
[0:06:24.422196973] [4546]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
terminate called after throwing an instance of 'std::runtime_error'
  what():  failed to import fd 20

$ rpicam-hello
Could not open any dmaHeap device
[0:06:45.846151361] [4565]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[0:06:45.886873695] [4567] ERROR RPI dma_heaps.cpp:53 Could not open any dmaHeap device
[0:06:45.920450497] [4567]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:06:45.925181738] [4567] ERROR RPI vc4.cpp:216 Failed to register camera ov5647 10-0036: -12
libEGL warning: DRI2: failed to authenticate
Made X/EGL preview window
ERROR: *** no cameras available ***

$ sudo cheese
(cheese:4757): GLib-GObject-CRITICAL **: 16:23:44.167: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:4757): GLib-GObject-CRITICAL **: 16:23:44.168: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:4757): GLib-GObject-CRITICAL **: 16:23:44.168: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:4757): GLib-GObject-CRITICAL **: 16:23:44.168: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:4757): GLib-GObject-CRITICAL **: 16:23:44.168: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:4757): GLib-GObject-CRITICAL **: 16:23:44.577: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:4757): GLib-GObject-CRITICAL **: 16:23:44.582: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:4757): GLib-GObject-CRITICAL **: 16:23:44.587: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:4757): GLib-GObject-CRITICAL **: 16:23:44.652: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:4757): GLib-GObject-CRITICAL **: 16:23:44.657: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:4757): cheese-WARNING **: 16:23:54.214: Failed to allocate required memory.: ../sys/v4l2/gstv4l2src.c(759): gst_v4l2src_decide_allocation (): /GstCameraBin:camerabin/GstWrapperCameraBinSrc:camera_source/GstBin:bin19/GstV4l2Src:v4l2src2:
Buffer pool activation failed

naushir, 2024-01-10:
@Jai-GAY please open a new issue for your problem.  It is not related to this issue at all.

naushir, 2024-01-10:
Given no activity from the original author, I'm closing this down now.

closed 2024-01-10 by naushir

Jai-GAY, 2024-01-10:
> Given no activity from the original author, I'm closing this down now.

[okay. done and thank you.](https://github.com/raspberrypi/rpicam-apps/issues/628)