raspberrypi/rpicam-apps #626: [BUG] libav: cannot allocate output context using stdout

Lillifee, 2024-01-03:
Hey,

I'm currently using the rpicam apps in my web application and encountered a bug reported on GitHub involving a Raspberry Pi 5 with a Camera Module 3. The issue can be found at: https://github.com/Lillifee/raspiCam/issues/67

I utilize the `--output -` option to stream the stdout to the website. However, when an output file is explicitly defined using `--output test.h264`, everything appears to work fine.

```
libcamera-vid --nopreview --width 1280 --height 720 --framerate 25 --bitrate 6000000 --codec H264 --timeout 0 --inline --output -

[0:06:16.583445990] [1264]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[0:06:16.597824541] [1271]  INFO RPI pisp.cpp:653 libpisp version v1.0.2 fa44a258644a 22-11-2023 (21:59:22)
[0:06:16.615570848] [1271]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media3 and ISP device /dev/media0 using PiSP variant BCM2712_C0
[0:06:16.616787786] [1264]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format Y16
[0:06:16.616813361] [1264]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format RGB6
[0:06:16.616820879] [1264]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format BGR6
[0:06:16.616829213] [1264]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format PC1M
Mode selection for 1280:720:12:P(25)
    SRGGB10_CSI2P,1536x864/120.135 - Score: 1100
    SRGGB10_CSI2P,2304x1296/56.0255 - Score: 1400
    SRGGB10_CSI2P,4608x2592/14.3536 - Score: 23592.8
Stream configuration adjusted
[0:06:16.617063585] [1264]  INFO Camera camera.cpp:1183 configuring streams: (0) 1280x720-YUV420 (1) 1536x864-BGGR16_PISP_COMP1
[0:06:16.617130975] [1271]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected CFE format: 1536x864-PC1B
[NULL @ 0x555675704d20] Unable to find a suitable output format for '-'
ERROR: *** libav: cannot allocate output context ***
```

Unfortunately, I am unable to test this myself as I do not have access to a Raspberry Pi 5. However, I came across a few sources that describe a similar problem:
https://forums.raspberrypi.com/viewtopic.php?t=360020
https://forums.raspberrypi.com/viewtopic.php?t=361176

This issue might be connected to my initial problem when I tried to set the profile to baseline, resulting in the error message: "ERROR: *** libav: no such profile baseline ***." Subsequently, I removed the profile, and I am confident that I can undo this modification. https://github.com/raspberrypi/rpicam-apps/issues/611

I would greatly appreciate any assistance since I am currently stuck without access to a Raspberry Pi 5 for testing. :(
You can find all information i got on https://github.com/Lillifee/raspiCam/issues/67

Thank you!

naushir, 2024-01-03:
You should add ``--libav-format h264`` to your command line and everything should work as expected.

Lillifee, 2024-01-03:
Hey @naushir,

thanks for the quick reply! I'll make a new release and close the issue if it works!

Does that mean libav is always used, even if I set the codec to h264?
Is the raspberrypi documentation still correct? 

This can select how the video frames are encoded. Valid options are:
- h264 - use H.264 encoder (the default)
- mjpeg - use MJPEG encoder
- yuv420 - output uncompressed YUV420 frames.
- libav - use the libav backend to encode audio and video

 Do you suggest using libav exclusively for all PI versions?

naushir, 2024-01-03:
On Pi 5 yes, libav is the only available H.264 codec.  The libav codec is significantly more featureful then the old codec framework, but it might struggle to work well on the lower powered Pi devices (Zero, 1/2/3).  So I would probably not use libav on them unless your use case is not too taxing.  Note that on the Pi 1/2/3/4 libav still uses the hardware codec, only Pi 5 uses the libav x264 software codec.

Lillifee, 2024-01-03:
Thanks for the explanation! I still had a problem with the profile that is not supported on pi5 but works on pi4 with libav codec...

However, thanks for the fast help!!

closed 2024-01-03 by Lillifee

ben-xD, 2024-10-19:
@naushir, would you know how to install libav on Raspberry Pi 5 (debian 12 / bookworm)? I tried:
- Installing ffmpeg with `sudo apt install -y ffmpeg`
- `sudo apt install libav-tools`, but this errors with `Package 'libav-tools' has no installation candidate`
- `sudo apt install libavcodec-dev`, but this still causes my error: Unable to find an appropriate H.264 codec

I'm not sure why we use libav, since ["Libav is dead"](https://superuser.com/a/507473/712630). I'm not sure ow to debug this "Unable to find an appropriate H.264 codec".

## More details
```bash
$ rpicam-vid -t 0 -n --inline
[0:12:40.093604312] [1527]  INFO Camera camera_manager.cpp:325 libcamera v0.3.2+27-7330f29b
[0:12:40.104777755] [1530]  INFO RPI pisp.cpp:695 libpisp version v1.0.7 28196ed6edcf 29-08-2024 (16:33:32)
[0:12:40.119539541] [1530]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/ov5647@36 to CFE device /dev/media2 and ISP device /dev/media0 using PiSP variant BCM2712_C0
Mode selection for 640:480:12:P
    SGBRG10_CSI2P,640x480/0 - Score: 1000
    SGBRG10_CSI2P,1296x972/0 - Score: 1287
    SGBRG10_CSI2P,1920x1080/0 - Score: 1636.67
    SGBRG10_CSI2P,2592x1944/0 - Score: 1854
Stream configuration adjusted
[0:12:40.120397400] [1527]  INFO Camera camera.cpp:1197 configuring streams: (0) 640x480-YUV420 (1) 640x480-GBRG_PISP_COMP1
[0:12:40.120530438] [1530]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected CFE format: 640x480-PC1g
ERROR: *** Unable to find an appropriate H.264 codec ***
```

naushir, 2024-10-21:
Try the following:
```
sudo apt install libavcodec-dev libavdevice-dev libavformat-dev libswresample-dev
```