raspberrypi/rpicam-apps #627: compiling error

hmne, 2024-01-03:
hi there 

i have rasp Zero W 32 bit kernel and user space bullseye 

kernel number 6.1.21+

libcamera compiled ok 

info for libcamera build

libcamera 0.1.0

  Versions
    Sources                  : 0.1.0+145-577e0c6b

  Paths
    LIBCAMERA_DATA_DIR       : "/usr/local/share/libcamera"
    LIBCAMERA_SYSCONF_DIR    : "/usr/local/etc/libcamera"
    IPA_PROXY_DIR            : "/usr/local/libexec/libcamera"
    IPA_CONFIG_DIR           : "/usr/local/etc/libcamera/ipa:/usr/local/share/libcamera/ipa"
    IPA_MODULE_DIR           : "/usr/local/lib/arm-linux-gnueabihf/libcamera"

  Configuration
    IPA modules signed with  : libcrypto
    Enabled pipelines        : rpi/vc4
                               vimc
    Enabled IPA modules      : rpi/vc4
                               vimc
    Controls files           : control_ids_draft.yaml
                               control_ids_core.yaml
    Properties files         : property_ids_draft.yaml
                               property_ids_core.yaml
    Hotplug support          : YES
    Tracing support          : YES
    Android support          : NO
    GStreamer support        : YES
    Python bindings          : YES
    V4L2 emulation support   : YES
    cam application          : YES
    qcam application         : NO
    lc-compliance application: YES
    Unit tests               : YES

  User defined options
    buildtype                : release
    cam                      : enabled
    documentation            : disabled
    ipas                     : rpi/vc4
    lc-compliance            : enabled
    pipelines                : rpi/vc4
    pycamera                 : enabled
    test                     : true
    v4l2                     : true

Found ninja-1.10.1 at /usr/bin/ninja

—————————————————————

info of rpicam error of compiling 

pi@raspi:~/libcamera-apps $ sudo ninja -C build install
ninja: Entering directory `build'
[14/44] Compiling C++ object rpicam_app.so.1.4.1.p/image_dng.cpp.o
FAILED: rpicam_app.so.1.4.1.p/image_dng.cpp.o 
c++ -Irpicam_app.so.1.4.1.p -I. -I.. -Icore -I/usr/local/include/libcamera -I/usr/include -I/usr/include/arm-linux-gnueabihf -I/usr/include/libpng16 -I/usr/include/libdrm -fdiagnostics-color=always -Wall -Winvalid-pch -Wextra -Wpedantic -Werror -std=c++17 -O3 -pedantic -Wno-unused-parameter -faligned-new -D_FILE_OFFSET_BITS=64 -Wno-psabi -DLIBAV_PRESENT=1 -DLIBDRM_PRESENT=1 -DLIBEGL_PRESENT=1 -fPIC -pthread -DBOOST_ALL_NO_LIB -MD -MQ rpicam_app.so.1.4.1.p/image_dng.cpp.o -MF rpicam_app.so.1.4.1.p/image_dng.cpp.o.d -o rpicam_app.so.1.4.1.p/image_dng.cpp.o -c ../image/dng.cpp
../image/dng.cpp:73:13: error: ‘RGGB16_PISP_COMP1’ is not a member of ‘libcamera::formats’
   73 |  { formats::RGGB16_PISP_COMP1, { "RGGB-16-PISP", 16, TIFF_RGGB, false, true } },
      |             ^~~~~~~~~~~~~~~~~
../image/dng.cpp:74:13: error: ‘GRBG16_PISP_COMP1’ is not a member of ‘libcamera::formats’
   74 |  { formats::GRBG16_PISP_COMP1, { "GRBG-16-PISP", 16, TIFF_GRBG, false, true } },
      |             ^~~~~~~~~~~~~~~~~
../image/dng.cpp:75:13: error: ‘GBRG16_PISP_COMP1’ is not a member of ‘libcamera::formats’
   75 |  { formats::GBRG16_PISP_COMP1, { "GBRG-16-PISP", 16, TIFF_GBRG, false, true } },
      |             ^~~~~~~~~~~~~~~~~
../image/dng.cpp:76:13: error: ‘BGGR16_PISP_COMP1’ is not a member of ‘libcamera::formats’
   76 |  { formats::BGGR16_PISP_COMP1, { "BGGR-16-PISP", 16, TIFF_BGGR, false, true } },
      |             ^~~~~~~~~~~~~~~~~
../image/dng.cpp:77:1: error: could not convert ‘{{libcamera::formats::SRGGB10_CSI2P, {"RGGB-10", 10, TIFF_RGGB, true, false}}, {libcamera::formats::SGRBG10_CSI2P, {"GRBG-10", 10, TIFF_GRBG, true, false}}, {libcamera::formats::SBGGR10_CSI2P, {"BGGR-10", 10, TIFF_BGGR, true, false}}, {libcamera::formats::SGBRG10_CSI2P, {"GBRG-10", 10, TIFF_GBRG, true, false}}, {libcamera::formats::SRGGB10, {"RGGB-10", 10, TIFF_RGGB, false, false}}, {libcamera::formats::SGRBG10, {"GRBG-10", 10, TIFF_GRBG, false, false}}, {libcamera::formats::SBGGR10, {"BGGR-10", 10, TIFF_BGGR, false, false}}, {libcamera::formats::SGBRG10, {"GBRG-10", 10, TIFF_GBRG, false, false}}, {libcamera::formats::SRGGB12_CSI2P, {"RGGB-12", 12, TIFF_RGGB, true, false}}, {libcamera::formats::SGRBG12_CSI2P, {"GRBG-12", 12, TIFF_GRBG, true, false}}, {libcamera::formats::SBGGR12_CSI2P, {"BGGR-12", 12, TIFF_BGGR, true, false}}, {libcamera::formats::SGBRG12_CSI2P, {"GBRG-12", 12, TIFF_GBRG, true, false}}, {libcamera::formats::SRGGB12, {"RGGB-12", 12, TIFF_RGGB, false, false}}, {libcamera::formats::SGRBG12, {"GRBG-12", 12, TIFF_GRBG, false, false}}, {libcamera::formats::SBGGR12, {"BGGR-12", 12, TIFF_BGGR, false, false}}, {libcamera::formats::SGBRG12, {"GBRG-12", 12, TIFF_GBRG, false, false}}, {libcamera::formats::SRGGB16, {"RGGB-16", 16, TIFF_RGGB, false, false}}, {libcamera::formats::SGRBG16, {"GRBG-16", 16, TIFF_GRBG, false, false}}, {libcamera::formats::SBGGR16, {"BGGR-16", 16, TIFF_BGGR, false, false}}, {libcamera::formats::SGBRG16, {"GBRG-16", 16, TIFF_GBRG, false, false}}, {libcamera::formats::R10_CSI2P, {"BGGR-10", 10, TIFF_BGGR, true, false}}, {libcamera::formats::R10, {"BGGR-10", 10, TIFF_BGGR, false, false}}, {libcamera::formats::R12, {"BGGR-12", 12, TIFF_BGGR, false, false}}, {<expression error>, {"RGGB-16-PISP", 16, TIFF_RGGB, false, true}}, {<expression error>, {"GRBG-16-PISP", 16, TIFF_GRBG, false, true}}, {<expression error>, {"GBRG-16-PISP", 16, TIFF_GBRG, false, true}}, {<expression error>, {"BGGR-16-PISP", 16, TIFF_BGGR, false, true}}}’ from ‘<brace-enclosed initializer list>’ to ‘const std::map<libcamera::PixelFormat, BayerFormat>’
   77 | };
      | ^
      | |
      | <brace-enclosed initializer list>
[15/44] Compiling C++ object rpicam_app.so.1.4.1.p/image_jpeg.cpp.o
ninja: build stopped: subcommand failed.

naushir, 2024-01-03:
You need to make sure you have first compiled libcamera with the tree from here:
https://github.com/raspberrypi/libcamera

Is there any specific reason you need to recompile rpicam-apps?

hmne, 2024-01-03:
i did compiled from the link above 

i need the newest available libcamera and rpicam-apps

if i can

hmne, 2024-01-03:
hi i did changes to dng.cpp 

/* PiSP compressed formats. */
	{ formats::RGGB16_PISP_COMP1, { "RGGB-16-PISP", 16, TIFF_RGGB, false, true } },
	{ formats::GRBG16_PISP_COMP1, { "GRBG-16-PISP", 16, TIFF_GRBG, false, true } },
	{ formats::GBRG16_PISP_COMP1, { "GBRG-16-PISP", 16, TIFF_GBRG, false, true } },
	{ formats::BGGR16_PISP_COMP1, { "BGGR-16-PISP", 16, TIFF_BGGR, false, true } },

change it to false, false and it’s compiling with no errors 

i hope doesn’t do any errors in the future if I are on the program 

i have imx708 cam

hmne, 2024-01-03:
i did change 

/image/meson.build

removed 

rpicam_app_src += files([
    'bmp.cpp',
    'dng.cpp', <------ i deleted 
    'jpeg.cpp',
    'png.cpp',
    'yuv.cpp',
])

I hope that there will be no problems with the program in the future

Because I expect that I will not need to shoot in DNG format i need only jpg format 

Note: I use an imx708

hmne, 2024-01-04:
thanks 🙏🏼 

i did something stupid and i know  i will regret that

i did delete this section 

/* PiSP compressed formats. */
	{ formats::RGGB16_PISP_COMP1, { "RGGB-16-PISP", 16, TIFF_RGGB, false, true } },
	{ formats::GRBG16_PISP_COMP1, { "GRBG-16-PISP", 16, TIFF_GRBG, false, true } },
	{ formats::GBRG16_PISP_COMP1, { "GBRG-16-PISP", 16, TIFF_GBRG, false, true } },
	{ formats::BGGR16_PISP_COMP1, { "BGGR-16-PISP", 16, TIFF_BGGR, false, true } },
};

from dng.cpp

and it’s compiled normally the weird thing is i did 
tested libcamera-still -r and it’s working 

i don’t need dng and I will compile it again 

better to be safe than sorry

thank you for your help