raspberrypi/rpicam-apps #628: Raspberry Pi 4B camera issue on Ubuntu 22.04 64-bit

Jai-GAY, 2024-01-10:
like to know why sudo cheese (sudo is a must) can detect and preview the legacy camera but not rpicam or libcamera?

[Ubuntu 22.04 64-bit](https://ubuntu.com/download/raspberry-pi) on Pi 4B.

$ libcamera-hello --list-cameras
Could not open any dmaHeap device
ERROR: rpicam-apps currently only supports the Raspberry Pi platforms.
Contributions for other platforms are welcome at https://github.com/raspberrypi/rpicam-apps.
$ sudo libcamera-hello --list-cameras
[sudo] password for pi:
ERROR: rpicam-apps currently only supports the Raspberry Pi platforms.
Contributions for other platforms are welcome at https://github.com/raspberrypi/rpicam-apps.

#config.txt
[pi4]
dtoverlay=vc4-fkms-v3d
#dtoverlay=vc4-kms-v3d
#camera_auto_detect=1
start_x=1
gpu_mem=128

$ sudo v4l2-ctl --list-devices
bcm2835-codec-decode (platform:bcm2835-codec):
        /dev/video10
        /dev/video11
        /dev/video12
        /dev/video18
        /dev/video31
        /dev/media1

bcm2835-isp (platform:bcm2835-isp):
        /dev/video13
        /dev/video14
        /dev/video15
        /dev/video16
        /dev/video20
        /dev/video21
        /dev/video22
        /dev/video23
        /dev/media0
        /dev/media2

mmal service 16.1 (platform:bcm2835-v4l2-0):
        /dev/video0

$ vcgencmd get_camera
VCHI initialization failed
$ **sudo** vcgencmd get_camera
supported=1 detected=1, libcamera interfaces=0

libcamera 0.1.0
  Versions
    Sources                  : 0.1.0+118-563cd78e

  Paths
    LIBCAMERA_DATA_DIR       : "/usr/local/share/libcamera"
    LIBCAMERA_SYSCONF_DIR    : "/usr/local/etc/libcamera"
    IPA_PROXY_DIR            : "/usr/local/libexec/libcamera"
    IPA_CONFIG_DIR           : "/usr/local/etc/libcamera/ipa:/usr/local/share/libcamera/ipa"
    IPA_MODULE_DIR           : "/usr/local/lib/aarch64-linux-gnu/libcamera"

  Configuration
    IPA modules signed with  : gnutls
    Enabled pipelines        : rpi/vc4
                               rpi/pisp
    Enabled IPA modules      : rpi/vc4
                               rpi/pisp
    Hotplug support          : YES
    Tracing support          : NO
    Android support          : NO
    GStreamer support        : YES
    Python bindings          : YES
    V4L2 emulation support   : YES
    cam application          : NO
    qcam application         : NO
    lc-compliance application: NO
    Unit tests               : NO

  Subprojects
    libpisp                  : YES
    nlohmann_json            : YES

  User defined options
    buildtype                : release
    cam                      : disabled
    documentation            : disabled
    gstreamer                : enabled
    ipas                     : rpi/vc4,rpi/pisp
    lc-compliance            : disabled
    pipelines                : rpi/vc4,rpi/pisp
    pycamera                 : enabled
    qcam                     : disabled
    test                     : false
    v4l2                     : true

[Installing symlink pointing to libcamera-base.so.0.1.0](https://www.raspberrypi.com/documentation/computers/camera_software.html#building-libcamera-and-rpicam-apps) to /usr/local/lib/aarch64-linux-gnu/libcamera-base.so.0.1
Installing symlink pointing to libcamera-base.so.0.1 to /usr/local/lib/aarch64-linux-gnu/libcamera-base.so
Installing symlink pointing to libpisp.so.1.0 to /usr/local/lib/aarch64-linux-gnu/libpisp.so
Installing symlink pointing to libcamera.so.0.1.0 to /usr/local/lib/aarch64-linux-gnu/libcamera.so.0.1
Installing symlink pointing to libcamera.so.0.1 to /usr/local/lib/aarch64-linux-gnu/libcamera.so
Running custom install script '/home/pi/libcamera/src/ipa/ipa-sign-install.sh /home/pi/libcamera/build/src/ipa-priv-key.pem lib/aarch64-linux-gnu/libcamera/ipa_rpi_vc4.so lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so'

rpicam-apps 1.4.1
  libcamera
    location             : /usr/local/lib/aarch64-linux-gnu
    version              : 0.1.0

  Build configuration
    libav encoder        : YES
    drm preview          : YES
    egl preview          : YES
    qt preview           : YES
    OpenCV postprocessing: YES
    TFLite postprocessing: NO

  User defined options
    enable_drm           : true
    enable_egl           : true
    enable_libav         : true
    enable_opencv        : true
    enable_qt            : true
    enable_tflite        : false
[Installing rpicam_app.so.1.4.1](https://www.raspberrypi.com/documentation/computers/camera_software.html#building-rpicam-apps) to /usr/local/lib/aarch64-linux-gnu
Installing apps/rpicam-still to /usr/local/bin
Installing apps/rpicam-vid to /usr/local/bin
Installing apps/rpicam-hello to /usr/local/bin
Installing apps/rpicam-raw to /usr/local/bin
Installing apps/rpicam-jpeg to /usr/local/bin

$ sudo chmod 666 /dev/video0
[sudo] password for pi: 
**$ cheese #not working**
(cheese:3910): GStreamer-Video-CRITICAL **: 15:10:09.944: gst_video_format_to_string: assertion 'format != GST_VIDEO_FORMAT_UNKNOWN' failed
(cheese:3910): cheese-WARNING **: 15:11:10.307: Internal data stream error.: ../libs/gst/base/gstbasesrc.c(3127): gst_base_src_loop (): /GstCameraBin:camerabin/GstWrapperCameraBinSrc:camera_source/GstBin:bin18/GstPipeWireSrc:pipewiresrc1:
streaming stopped, reason not-negotiated (-4)

**sudo** cheese # can get the preview
[image]

Jai-GAY, 2024-01-10:
> #config.txt

#adding
dtoverlay=ov5647

$ libcamera-hello --list-cameras
Could not open any dmaHeap device
No cameras available!
$ sudo libcamera-hello --list-cameras
[sudo] password for pi: 
Available cameras
'-----------------
0 : ov5647 [2592x1944 10-bit GBRG] (/base/soc/i2c0mux/i2c@1/ov5647@36)
    Modes: 'SGBRG10_CSI2P' : 640x480 [58.92 fps - (16, 0)/2560x1920 crop]
                             1296x972 [43.25 fps - (0, 0)/2592x1944 crop]
                             1920x1080 [30.62 fps - (348, 434)/1928x1080 crop]
                             2592x1944 [15.63 fps - (0, 0)/2592x1944 crop]

$ sudo v4l2-ctl --list-devices
bcm2835-codec-decode (platform:bcm2835-codec):
	/dev/video10
	/dev/video11
	/dev/video12
	/dev/video18
	/dev/video31
	/dev/media1

bcm2835-isp (platform:bcm2835-isp):
	/dev/video13
	/dev/video14
	/dev/video15
	/dev/video16
	/dev/video20
	/dev/video21
	/dev/video22
	/dev/video23
	/dev/media2
	/dev/media3

unicam (platform:fe801000.csi):
	/dev/video0
	/dev/media0

$ v4l2-ctl --list-devices
bcm2835-codec-decode (platform:bcm2835-codec):
	/dev/video10
	/dev/video11
	/dev/video12
	/dev/video18
	/dev/video31
	/dev/media1

bcm2835-isp (platform:bcm2835-isp):
	/dev/video13
	/dev/video14
	/dev/video15
	/dev/video16
	/dev/video20
	/dev/video21
	/dev/video22
	/dev/video23
	/dev/media2
	/dev/media3

unicam (platform:fe801000.csi):
	/dev/video0
	/dev/media0

$ vcgencmd get_camera
VCHI initialization failed
$ sudo vcgencmd get_camera
supported=1 **detected=0**, libcamera interfaces=1

$ sudo rpicam-hello
[0:06:23.954700378] [4545]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[0:06:24.026313253] [4546]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:06:24.033455383] [4546]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media0 and ISP device /dev/media2
[0:06:24.034031918] [4546]  INFO RPI pipeline_base.cpp:1142 Using configuration file '/usr/local/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
libEGL warning: DRI2: failed to authenticate
Made X/EGL preview window
Mode selection for 1296:972:12:P
    SGBRG10_CSI2P,640x480/0 - Score: 3296
    SGBRG10_CSI2P,1296x972/0 - Score: 1000
    SGBRG10_CSI2P,1920x1080/0 - Score: 1349.67
    SGBRG10_CSI2P,2592x1944/0 - Score: 1567
Stream configuration adjusted
[0:06:24.420523761] [4545]  INFO Camera camera.cpp:1183 configuring streams: (0) 1296x972-YUV420 (1) 1296x972-SGBRG10_CSI2P
[0:06:24.422196973] [4546]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
terminate called after throwing an instance of 'std::runtime_error'
  what():  failed to import fd 20

$ rpicam-hello
Could not open any dmaHeap device
[0:06:45.846151361] [4565]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[0:06:45.886873695] [4567] ERROR RPI dma_heaps.cpp:53 Could not open any dmaHeap device
[0:06:45.920450497] [4567]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:06:45.925181738] [4567] ERROR RPI vc4.cpp:216 Failed to register camera ov5647 10-0036: -12
libEGL warning: DRI2: failed to authenticate
Made X/EGL preview window
ERROR: *** no cameras available ***

$ sudo cheese #no preview
(cheese:4757): GLib-GObject-CRITICAL **: 16:23:44.167: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:4757): GLib-GObject-CRITICAL **: 16:23:44.168: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:4757): GLib-GObject-CRITICAL **: 16:23:44.168: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:4757): GLib-GObject-CRITICAL **: 16:23:44.168: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:4757): GLib-GObject-CRITICAL **: 16:23:44.168: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:4757): GLib-GObject-CRITICAL **: 16:23:44.577: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:4757): GLib-GObject-CRITICAL **: 16:23:44.582: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:4757): GLib-GObject-CRITICAL **: 16:23:44.587: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:4757): GLib-GObject-CRITICAL **: 16:23:44.652: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:4757): GLib-GObject-CRITICAL **: 16:23:44.657: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:4757): cheese-WARNING **: 16:23:54.214: Failed to allocate required memory.: ../sys/v4l2/gstv4l2src.c(759): gst_v4l2src_decide_allocation (): /GstCameraBin:camerabin/GstWrapperCameraBinSrc:camera_source/GstBin:bin19/GstV4l2Src:v4l2src2:
Buffer pool activation failed

naushir, 2024-01-10:
The cheese application uses direct v4l2 calls to drive the camera.  This used to work in the legacy camera stack because we had a v4l2 driver that used to emulate a full camera pipeline.  To use cheese with libcamera, you need to call it with the libcamera v4l2 emulation layer:

```bash
libcamerify cheese
```

However, you do have another problem:
> $ rpicam-hello
Could not open any dmaHeap device
[0:06:45.846151361] [4565] INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[0:06:45.886873695] [4567] ERROR RPI dma_heaps.cpp:53 Could not open any dmaHeap device
[0:06:45.920450497] [4567] WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:06:45.925181738] [4567] ERROR RPI vc4.cpp:216 Failed to register camera ov5647 10-0036: -12
libEGL warning: DRI2: failed to authenticate
Made X/EGL preview window
ERROR: *** no cameras available ***

You seem to be using an Ubuntu OS distribution.  I don't know how packages are built on there, and it seems something is wrong with the libcamera build/install.  I suggest you stick to an official Raspberry Pi OS image to start with.

Jai-GAY, 2024-01-10:
> I suggest you stick to an official Raspberry Pi OS image to start with.

I cannot, because [ROS 2 Humble installation is only available on Ubuntu 22.04](https://www.ros.org/reps/rep-2000.html#humble-hawksbill-may-2022-may-2027).

> ```shell
> libcamerify cheese
> ```

okay, it flashes with a 1 or 0.5-sec preview and then a black and error message.

$ sudo libcamerify cheese
[sudo] password for pi: 
[0:33:47.548958809] [5013]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[0:33:47.626417229] [5034]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:33:47.631865992] [5034]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media0 and ISP device /dev/media2
(cheese:5013): GLib-GObject-CRITICAL **: 16:44:27.291: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:5013): GLib-GObject-CRITICAL **: 16:44:27.306: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:5013): GLib-GObject-CRITICAL **: 16:44:27.323: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:5013): GLib-GObject-CRITICAL **: 16:44:27.337: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:5013): GLib-GObject-CRITICAL **: 16:44:27.338: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:5013): GLib-GObject-CRITICAL **: 16:44:27.338: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
[0:33:48.034356146] [5041]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-YUYV
[0:33:48.035184399] [5034]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
[0:33:48.037995902] [5041]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-YUYV
[0:33:48.038727246] [5034]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
(cheese:5013): GLib-GObject-CRITICAL **: 16:44:27.792: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
[0:33:48.267455012] [5044]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-YUYV
[0:33:48.268193874] [5034]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
[0:33:48.270300848] [5044]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-YUYV
[0:33:48.271631571] [5034]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
(cheese:5013): GLib-GObject-CRITICAL **: 16:44:27.833: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:5013): GLib-GObject-CRITICAL **: 16:44:27.893: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:5013): GLib-GObject-CRITICAL **: 16:44:27.973: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:5013): GLib-GObject-CRITICAL **: 16:44:27.998: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
(cheese:5013): GLib-GObject-CRITICAL **: 16:44:28.020: g_value_dup_string: assertion 'G_VALUE_HOLDS_STRING (value)' failed
[0:33:49.814324687] [5048] ERROR Request request.cpp:472 FrameBuffer already set for stream
[0:33:49.814429781] [5048] ERROR V4L2Compat v4l2_camera.cpp:259 Can't set buffer for request
[0:33:49.929011843] [5048] ERROR Request request.cpp:472 FrameBuffer already set for stream
[0:33:49.929086381] [5048] ERROR V4L2Compat v4l2_camera.cpp:259 Can't set buffer for request
[0:33:49.929937190] [5048] ERROR Request request.cpp:472 FrameBuffer already set for stream
[0:33:49.930472716] [5048] ERROR V4L2Compat v4l2_camera.cpp:259 Can't set buffer for request
[0:33:49.958286965] [5048] ERROR Request request.cpp:472 FrameBuffer already set for stream
[0:33:49.958504264] [5048] ERROR V4L2Compat v4l2_camera.cpp:259 Can't set buffer for request
(cheese:5013): cheese-WARNING **: 16:44:29.594: Failed to allocate a buffer: ../sys/v4l2/gstv4l2src.c(1146): gst_v4l2src_create (): /GstCameraBin:camerabin/GstWrapperCameraBinSrc:camera_source/GstBin:bin18/GstV4l2Src:v4l2src1

naushir, 2024-01-10:
Before you get ``libcamerify cheese`` working, you need to make sure your libcamera install is working correctly with a basic application.  What happens if you try to run ``rpicam-hello``?

Jai-GAY, 2024-01-10:
> What happens if you try to run `rpicam-hello`?

It flashes with a black background and closes within 1 sec.

$ rpicam-hello
Could not open any dmaHeap device
[1:18:36.870025332] [5233]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[1:18:36.916841839] [5235] ERROR RPI dma_heaps.cpp:53 Could not open any dmaHeap device
[1:18:36.946063743] [5235]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[1:18:36.951910401] [5235] ERROR RPI vc4.cpp:216 Failed to register camera ov5647 10-0036: -12
libEGL warning: DRI2: failed to authenticate
Made X/EGL preview window
ERROR: *** no cameras available ***

$ sudo rpicam-hello
[sudo] password for pi: 
[1:18:48.031136815] [5250]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[1:18:48.101454712] [5251]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[1:18:48.108380883] [5251]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media0 and ISP device /dev/media2
[1:18:48.109013521] [5251]  INFO RPI pipeline_base.cpp:1142 Using configuration file '/usr/local/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
libEGL warning: DRI2: failed to authenticate
Made X/EGL preview window
Mode selection for 1296:972:12:P
    SGBRG10_CSI2P,640x480/0 - Score: 3296
    SGBRG10_CSI2P,1296x972/0 - Score: 1000
    SGBRG10_CSI2P,1920x1080/0 - Score: 1349.67
    SGBRG10_CSI2P,2592x1944/0 - Score: 1567
Stream configuration adjusted
[1:18:48.411607856] [5250]  INFO Camera camera.cpp:1183 configuring streams: (0) 1296x972-YUV420 (1) 1296x972-SGBRG10_CSI2P
[1:18:48.413883087] [5251]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
terminate called after throwing an instance of 'std::runtime_error'
  what():  failed to import fd 20

naushir, 2024-01-10:
Try ``sudo rpicam-hello --qt-preview``, I think that might work for you.  Although, I'm not sure why you need sudo.

Jai-GAY, 2024-01-10:
> Try `sudo rpicam-hello --qt-preview`, I think that might work for you. Although, I'm not sure why you need sudo.

okay, with qt-preview works. do you think I will face an issue with openCV 2 (python3)?
[image]

naushir, 2024-01-10:
Unfortunately I can't really help much as I am unfamiliar with cheese and/or the gstreamer pipe it is setting up and using.  However, perhaps you don't need to use cheese, and use one of the rpicam-apps instead?  What are you trying to do in your use case?

Jai-GAY, 2024-01-10:
> What are you trying to do in your use case?

Mavros 2 + openCV + camera + arcou map + Arducopter

Jai-GAY, 2024-01-11:
> Can you replace `cheese` with `rpicam-*` given the latter works as expected?

okay, maybe can for now. thanks

Jai-GAY, 2024-01-11:
> Although, I'm not sure why you need sudo.

oh ya, do you know why?

naushir, 2024-01-11:
Its likely due to user permissions not having access to /dev/video* file nodes.

Jai-GAY, 2024-01-12:
> $ sudo chmod 666 /dev/video0

> permissions not having access to /dev/video*

I already tested it before, I will retry it again to see if that is the issue.

Another question, do you know what are the differences between [python3-libcamera](https://packages.debian.org/sid/python3-libcamera) and [libcamera](https://github.com/raspberrypi/libcamera)? Can I remove libcamera and use python3-libcamera to build [rpicam-apps](https://www.raspberrypi.com/documentation/computers/camera_software.html#building-rpicam-apps-without-rebuilding-libcamera) 

while searching for an issue and [found this](https://www.waveshare.net/wiki/RPi_Camera_V2#.E8.BD.AF.E4.BB.B6.E9.85.8D.E7.BD.AE) and I like to share it with more people. Unfortunately is in Chinese.

naushir, 2024-01-12:
python3-libcamera is the python bindings for the libcamera API.  If you intend on using our picamera2 python library, this will be needed.

Regarding the permissions, it would be worth checking that your username has access to the video and plugdev groups as well.  If this still does not work, I suggest asking the Ubuntu support team for further help.  This is not something we directly support.

Jai-GAY, 2024-01-12:
> python3-libcamera is the python bindings for the libcamera API.

I cannot use pip3 install libcamera then pip3 install picamera2. I need sudo apt-get install -y python3-picamera2.

$ pip3 show picamera2
Name: picamera2
Version: 0.3.16
Summary: The libcamera-based Python interface to Raspberry Pi cameras, based on the original Picamera library
Home-page: https://github.com/RaspberryPi/picamera2
Author: Raspberry Pi & Raspberry Pi Foundation
Author-email: picamera2@raspberrypi.com
License: BSD 2-Clause License
Location: /home/pi/.local/lib/python3.10/site-packages
Requires: av, numpy, PiDNG, piexif, pillow, python-prctl, simplejpeg, v4l2-python3
Required-by:

raceback (most recent call last):
  File "/home/pi/backup/aruco_pose/cam_cal.py", line 5, in <module>
    from picamera2 import Picamera2
ModuleNotFoundError: No module named 'picamera2'

naushir, 2024-01-12:
As mentioned earlier, please speak to the Ubuntu support team for further help.  We don't directly support packages on any operating system besides Raspberry Pi OS.

closed 2024-01-12 by naushir

BlackLight, 2024-02-04:
I can confirm that the permission issues also happens to me on a RPi OS image freshly upgraded to bookworm.

Camera commands without sudo:

```
libcamera-still -o test.jpg
libEGL warning: DRI2: failed to authenticate
Made X/EGL preview window
[0:07:11.686624111] [2241]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f
[0:07:11.870833721] [2251]  WARN RPI vc4.cpp:383 Mismatch between Unicam and CamHelper for embedded data usage!
[0:07:11.873099295] [2251]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
[0:07:11.873332368] [2251]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[0:07:11.875800546] [2241]  INFO Camera camera.cpp:1033 configuring streams: (0) 1640x1232-YUV420
[0:07:11.877159349] [2251]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected unicam format: 1640x1232-pBAA
[0:07:12.058702082] [2251]  INFO V4L2 v4l2_videodevice.cpp:1820 /dev/video0[13:cap]: Zero sequence expected for first frame (got 10)
terminate called after throwing an instance of 'std::runtime_error'
  what():  failed to import fd 19
Aborted (core dumped)
```

When using sudo, everything works.

Permissions:

```
$ ls -l /dev/video*
crw-rw----+ 1 root video 81, 13 Feb  4 12:45 /dev/video0
crw-rw----+ 1 root video 81, 14 Feb  4 12:45 /dev/video1
crw-rw----+ 1 root video 81,  2 Feb  4 12:45 /dev/video10
crw-rw----+ 1 root video 81,  5 Feb  4 12:45 /dev/video11
crw-rw----+ 1 root video 81,  6 Feb  4 12:45 /dev/video12
crw-rw----+ 1 root video 81,  0 Feb  4 12:45 /dev/video13
crw-rw----+ 1 root video 81,  1 Feb  4 12:45 /dev/video14
crw-rw----+ 1 root video 81,  3 Feb  4 12:45 /dev/video15
crw-rw----+ 1 root video 81,  4 Feb  4 12:45 /dev/video16
crw-rw----+ 1 root video 81, 10 Feb  4 12:45 /dev/video18
crw-rw----+ 1 root video 81,  7 Feb  4 12:45 /dev/video20
crw-rw----+ 1 root video 81,  8 Feb  4 12:45 /dev/video21
crw-rw----+ 1 root video 81,  9 Feb  4 12:45 /dev/video22
crw-rw----+ 1 root video 81, 11 Feb  4 12:45 /dev/video23
crw-rw----+ 1 root video 81, 12 Feb  4 12:45 /dev/video31

$ ls -l /dev/media*
crw-rw----+ 1 root video 241, 0 Feb  4 12:45 /dev/media0
crw-rw----+ 1 root video 241, 1 Feb  4 12:45 /dev/media1
crw-rw----+ 1 root video 241, 2 Feb  4 12:45 /dev/media2
crw-rw----+ 1 root video 241, 3 Feb  4 12:45 /dev/media3
```

User groups:

```
$ groups
pi adm dialout cdrom sudo audio video plugdev games users input render netdev lpadmin gpio i2c spi
```

alikasundara, 2024-12-19:
@blacklight check the ownership of `/dev/dma_heap/system`. I had to add an udev rule to change the group to `video` as it was `root:root` by default, as per [this answer](https://raspberrypi.stackexchange.com/questions/141106/how-to-fix-the-libcamera-error-could-not-open-any-dmaheap-device)

pranavk-2003, 2025-02-03:
> I like to know why sudo cheese (sudo is a must) can detect and preview the legacy camera but not rpicam or libcamera?
> 
> [Ubuntu 22.04 64-bit](https://ubuntu.com/download/raspberry-pi) on Pi 4B.
> 
> $ libcamera-hello --list-cameras Could not open any dmaHeap device ERROR: rpicam-apps currently only supports the Raspberry Pi platforms. Contributions for other platforms are welcome at https://github.com/raspberrypi/rpicam-apps. $ sudo libcamera-hello --list-cameras [sudo] password for pi: ERROR: rpicam-apps currently only supports the Raspberry Pi platforms. Contributions for other platforms are welcome at https://github.com/raspberrypi/rpicam-apps.
> 
> #config.txt [pi4] dtoverlay=vc4-fkms-v3d #dtoverlay=vc4-kms-v3d #camera_auto_detect=1 start_x=1 gpu_mem=128
> 
> $ sudo v4l2-ctl --list-devices bcm2835-codec-decode (platform:bcm2835-codec): /dev/video10 /dev/video11 /dev/video12 /dev/video18 /dev/video31 /dev/media1
> 
> bcm2835-isp (platform:bcm2835-isp): /dev/video13 /dev/video14 /dev/video15 /dev/video16 /dev/video20 /dev/video21 /dev/video22 /dev/video23 /dev/media0 /dev/media2
> 
> mmal service 16.1 (platform:bcm2835-v4l2-0): /dev/video0
> 
> $ vcgencmd get_camera VCHI initialization failed $ **sudo** vcgencmd get_camera supported=1 detected=1, libcamera interfaces=0
> 
> libcamera 0.1.0 Versions Sources : 0.1.0+118-563cd78e
> 
> Paths LIBCAMERA_DATA_DIR : "/usr/local/share/libcamera" LIBCAMERA_SYSCONF_DIR : "/usr/local/etc/libcamera" IPA_PROXY_DIR : "/usr/local/libexec/libcamera" IPA_CONFIG_DIR : "/usr/local/etc/libcamera/ipa:/usr/local/share/libcamera/ipa" IPA_MODULE_DIR : "/usr/local/lib/aarch64-linux-gnu/libcamera"
> 
> Configuration IPA modules signed with : gnutls Enabled pipelines : rpi/vc4 rpi/pisp Enabled IPA modules : rpi/vc4 rpi/pisp Hotplug support : YES Tracing support : NO Android support : NO GStreamer support : YES Python bindings : YES V4L2 emulation support : YES cam application : NO qcam application : NO lc-compliance application: NO Unit tests : NO
> 
> Subprojects libpisp : YES nlohmann_json : YES
> 
> User defined options buildtype : release cam : disabled documentation : disabled gstreamer : enabled ipas : rpi/vc4,rpi/pisp lc-compliance : disabled pipelines : rpi/vc4,rpi/pisp pycamera : enabled qcam : disabled test : false v4l2 : true
> 
> [Installing symlink pointing to libcamera-base.so.0.1.0](https://www.raspberrypi.com/documentation/computers/camera_software.html#building-libcamera-and-rpicam-apps) to /usr/local/lib/aarch64-linux-gnu/libcamera-base.so.0.1 Installing symlink pointing to libcamera-base.so.0.1 to /usr/local/lib/aarch64-linux-gnu/libcamera-base.so Installing symlink pointing to libpisp.so.1.0 to /usr/local/lib/aarch64-linux-gnu/libpisp.so Installing symlink pointing to libcamera.so.0.1.0 to /usr/local/lib/aarch64-linux-gnu/libcamera.so.0.1 Installing symlink pointing to libcamera.so.0.1 to /usr/local/lib/aarch64-linux-gnu/libcamera.so Running custom install script '/home/pi/libcamera/src/ipa/ipa-sign-install.sh /home/pi/libcamera/build/src/ipa-priv-key.pem lib/aarch64-linux-gnu/libcamera/ipa_rpi_vc4.so lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so'
> 
> rpicam-apps 1.4.1 libcamera location : /usr/local/lib/aarch64-linux-gnu version : 0.1.0
> 
> Build configuration libav encoder : YES drm preview : YES egl preview : YES qt preview : YES OpenCV postprocessing: YES TFLite postprocessing: NO
> 
> User defined options enable_drm : true enable_egl : true enable_libav : true enable_opencv : true enable_qt : true enable_tflite : false [Installing rpicam_app.so.1.4.1](https://www.raspberrypi.com/documentation/computers/camera_software.html#building-rpicam-apps) to /usr/local/lib/aarch64-linux-gnu Installing apps/rpicam-still to /usr/local/bin Installing apps/rpicam-vid to /usr/local/bin Installing apps/rpicam-hello to /usr/local/bin Installing apps/rpicam-raw to /usr/local/bin Installing apps/rpicam-jpeg to /usr/local/bin
> 
> $ sudo chmod 666 /dev/video0 [sudo] password for pi: **$ cheese #not working** (cheese:3910): GStreamer-Video-CRITICAL **: 15:10:09.944: gst_video_format_to_string: assertion 'format != GST_VIDEO_FORMAT_UNKNOWN' failed (cheese:3910): cheese-WARNING **: 15:11:10.307: Internal data stream error.: ../libs/gst/base/gstbasesrc.c(3127): gst_base_src_loop (): /GstCameraBin:camerabin/GstWrapperCameraBinSrc:camera_source/GstBin:bin18/GstPipeWireSrc:pipewiresrc1: streaming stopped, reason not-negotiated (-4)
> 
> **sudo** cheese # can get the preview [image]

i am currently facing the same issue with ov5647 camera, it worked once and stopped working again i dont understand. https://forums.raspberrypi.com/viewtopic.php?t=352011 This topic here discusses the exact issue i am facing but im blind about what shall i change. if anyone can i would appreciate any help

6by9, 2025-02-03:
> i am currently facing the same issue with ov5647 camera, it worked once and stopped working again i dont understand. https://forums.raspberrypi.com/viewtopic.php?t=352011 This topic here discusses [...]

Please stick to your forum thread - https://forums.raspberrypi.com/viewtopic.php?t=383473
Commenting on closed issues is a good way for the discussion to get lost.