raspberrypi/rpicam-apps #630: recorded area at imx708 for video too small

LeifSec, 2024-01-12:
Similar  to #595 now for imx708 (no zoom lens etc.) the video area is much smaller than the picture area.

I have already done `rpi_update` and reboot / power off /on.

` libcamera-still -o libcamera_still.jpg`

[image]

`libcamera-vid  -t 5000 -o libcamera_vid.h264 `
[image]

naushir, 2024-01-15:
What resolution are you trying to record?  But default ``rpicam-vid`` records 640x480, which will crop the field of view.  Try adding ``--width 1920 --height 1080``, that should give you the full FoV.

naushir, 2024-01-15:
> Does this mean that if I I want to record the full sensor area I always have to record with ` --width 1920 --height 1080`and then do a down sampling of the resolution by myself

No, you can ask the ISP to do the downsampling.  You can do this with the following command:
```
rpicam-vid  -t 5000 -o libcamera_vid.h264  --width <width> --height <height> --mode 2304:1296:10:P
```

This will request the full FoV mode from the sensor, and scale down to any resolution you requset in the ISP.  However, you need to match the 16:9 aspect ratio to ensure the ISP does not do any further crop of the FoV.

LeifSec, 2024-01-15:
Ok `rpicam-vid  -t 5000 -o libcamera_vid_mod_800.h264  --width 800 --height 450 --mode 2304:1296:10:P` produced full sensor area.

Thanks a lot.

P.S.:
Now I have to figure out how this is handles in `pycamera2` ...